JvanKatwijk/qt-dab #101: Unable to compile on RPi3

NomisGninnac, 2018-04-14:
gcc -c -pipe -flto -ffast-math -O2 -Wall -W -D_REENTRANT -DGITHASH=\"aef9006\" -D__THREADED_BACKEND -D__THREADED_DECODING -DMSC_DATA__ -DPRESET_NAME -DTII_GUESSING -DHAVE_SPECTRUM -D__QUALITY -DHAVE_RTLSDR -DHAVE_RTL_TCP -DTRY_EPG -DIMPULSE_RESPONSE -DSSE_AVAILABLE -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -I. -I. -Isrc -Iincludes -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/viterbi_768 -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/various -Idevices -Idevices/rawfiles -Idevices/wavfiles -I/usr/local/include -I/usr/local/include -I/usr/include/qt4/qwt -I/usr/include/qt5/qwt -I/usr/include/qt4/qwt -I/usr/include/qwt -I/usr/local/qwt-6.1.4-svn/ -Iincludes/scopes-qwt6 -Iincludes/scopes-qwt6 -Ioptional-scope -Idevices/rtlsdr-handler -Idevices/rtl_tcp -Iincludes/backend/data/epg -Ioptional-ir -I. -I. -o spiral-sse.o src/backend/viterbi_768/spiral-sse.c
src/backend/viterbi_768/spiral-sse.c:23:23: fatal error: emmintrin.h: No such file or directory
 #include <emmintrin.h>
                       ^
compilation terminated.
Makefile:1384: recipe for target 'spiral-sse.o' failed
make: *** [spiral-sse.o] Error 1

JvanKatwijk, 2018-04-14:
specify NO-SSE in the ".pro" file

NomisGninnac, 2018-04-14:
Produces the following error

/usr/lib/arm-linux-gnueabihf/qt4/bin/qmake -o Makefile qt-dab.pro
Project MESSAGE: Current git hash = aef9006
g++ -flto -Wl,-O1 -o linux-bin/qt-dab-1.0-alpha main.o radio.o dab-processor.o sample-reader.o ofdm-decoder.o phasereference.o phasetable.o freq-interleaver.o tii_table.o tii_detector.o viterbi-768.o fic-handler.o msc-handler.o protection.o eep-protection.o uep-protection.o fib-processor.o galois.o reed-solomon.o rscodec.o charsets.o firecode-checker.o frame-processor.o protTables.o virtual-backend.o audio-backend.o data-backend.o mp2processor.o mp4processor.o faad-decoder.o pad-handler.o data-processor.o virtual-datahandler.o tdc-datahandler.o ip-datahandler.o mot-databuilder.o mot-data.o journaline-datahandler.o crc_8_16.o log.o newssvcdec_impl.o Splitter.o dabdgdec_impl.o newsobject.o NML.o audio-base.o newconverter.o audiosink.o fft-handler.o Xtan2.o dab-params.o band-handler.o text-mapper.o ensemble-printer.o virtual-input.o rawfiles.o wavfiles.o iqdisplay.o spectrum-handler.o rtlsdr-handler.o rtl-dongleselect.o rtl_tcp_client.o epgdec.o impulse-viewer.o moc_radio.o moc_dab-processor.o moc_sample-reader.o moc_ofdm-decoder.o moc_phasereference.o moc_tii_table.o moc_fic-handler.o moc_fib-processor.o moc_data-backend.o moc_mp2processor.o moc_mp4processor.o moc_faad-decoder.o moc_data-processor.o moc_pad-handler.o moc_virtual-datahandler.o moc_tdc-datahandler.o moc_ip-datahandler.o moc_mot-data.o moc_audio-base.o moc_iqdisplay.o moc_spectrum-handler.o moc_rtlsdr-handler.o moc_rtl-dongleselect.o moc_rtl_tcp_client.o moc_impulse-viewer.o qrc_resources.o    -L/usr/lib/arm-linux-gnueabihf -lfftw3f -lusb-1.0 -ldl -lportaudio -lz -lsndfile -lsamplerate -lfaad /usr/lib/libqwt.so -lQtXml -lQtGui -lQtNetwork -lQtCore -lpthread 
/tmp/ccEIsPVF.ltrans7.ltrans.o: In function `viterbi_768::deconvolve(short*, unsigned char*)':
<artificial>:(.text+0xccc): undefined reference to `FULL_SPIRAL_sse'
collect2: error: ld returned 1 exit status
Makefile:279: recipe for target 'linux-bin/qt-dab-1.0-alpha' failed
make: *** [linux-bin/qt-dab-1.0-alpha] Error 1

JvanKatwijk, 2018-04-14:
My mistake, it should be CONFIG += NO_SSE.
If you are compiling on an x64 based machine, you could choose for CONFIG += SSE,
if you are compiling on an RPI then CONFIG += NEON might work.

The latter two cause the executing machine to to SSE resp. simd instructions, especially aiming
at the viterbi decoder (which consumes a lot of resources)