JvanKatwijk/qt-dab #102: Build error with qt 5.10 on a raspberry pi 3

jbouwh, 2018-04-19:
/usr/lib/arm-linux-gnueabihf/libQt5OpenGL.so.5: undefined reference to `qTriangulate(QVectorPath const&, QTransform const&, double)@Qt_5_PRIVATE_API'
collect2: error: ld returned 1 exit status
Makefile:556: recept voor doel 'linux-bin/qt-dab-1.0-alpha' is mislukt
make: *** [linux-bin/qt-dab-1.0-alpha] Fout 1

At link state the qTriangulate methos is not present in lib QtOpenGL.so the compilation succeeds

JvanKatwijk, 2018-04-19:
Thanks for the report. Can you be slightly more precise on the command
sequence applied to reach this result?
I.e. qmake/cmake, qt-dab.pro or CMakeLists.txt file).
I just ran a quick test on am RPI3 (Stretch, using qmake), which ran fine

JvanKatwijk, 2018-04-19:
> I used qmake following a make - j4. I built Qt 5.10 from source.

Yes,
Can you send me the ".pro" file, then I can do a run as you did with the
same parameters

jbouwh, 2018-04-21:
I have discovered that opengl qt 5.7.1. library was installed as dependency on qwt-qt5. It seems I have to compile this from source against 5.10.1. Any idea how I can do this?

JvanKatwijk, 2018-04-22:
Sorry,no I only use precompiled Qt packages. If you switch off spectrum
(and impulse response) in the ".pro" file, you won't need qwt (as is the
case
with dabradio, no qwt)

jbouwh, 2018-04-23:
I am missing qwt in stretch, or are you using http://www.qtrpi.com/home?

JvanKatwijk, 2018-04-23:
afaik stretch supports qwt, compiled against Qt5

JvanKatwijk, 2018-04-23:
> so you use the 5.7.1 qt with stretch as a base.

I always use the libraries on the distribution. When I used jessie (up to
last summer) I used Qt4 since the qwt distributed on jessie was compiled
against qt4,
the qwt on Stretch is compiled against qt5, so  qt5 is used.

The x64 appImage is built in a Trusty environment, so there I am using
(have to use) Qt4 size qwt is compiled against Qt4

JvanKatwijk, 2018-04-26:
> Now I am compiling againt qt 5.7.1. If have qwt 6.1.4-svn installed with
> the qt5-qwt package from debain stretch.
>
> I am getting an internal compiler error. Before every ran fine at this
> point. Any idea?
>
> /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I.
> -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o dab-processor.o
> dab-processor.cpp
> g++ -c -pipe -std=c++11 -flto -ffast-math -isystem
> /usr/include/arm-linux-gnueabihf/qt5 -O2 -Wall -W -D_REENTRANT -fPIC
> -DGITHASH=\"50a18c9\" -D__THREADED_BACKEND -DMSC_DATA__ -DPRESET_NAME
> -DTII_GUESSING -D__QUALITY -DHAVE_RTLSDR -DHAVE_SDRPLAY -DHAVE_AIRSPY
> -DHAVE_RTL_TCP -DTRY_EPG -DIMPULSE_RESPONSE -DQT_NO_DEBUG -DQT_WIDGETS_LIB
> -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_CORE_LIB -I. -I. -I. -Isrc
> -Iincludes -Iincludes/ofdm -Iincludes/backend
> -Iincludes/backend/viterbi_768 -Iincludes/backend/audio
> -Iincludes/backend/data -Iincludes/backend/data/journaline
> -Iincludes/output -Iincludes/various -Idevices -Idevices/rawfiles
> -Idevices/wavfiles -isystem /usr/local/include -isystem /usr/local/include
> -isystem /usr/include/qt4/qwt -isystem /usr/include/qt5/qwt -isystem
> /usr/include/qt4/qwt -isystem /usr/include/qwt -I/usr/local/qwt-6.1.4-svn
> -Iincludes/scopes-qwt6 -Iincludes/scopes-qwt6 -Ioptional-scope
> -Idevices/rtlsdr-handler -Idevices/sdrplay-handler
> -Idevices/airspy-handler -Idevices/airspy-handler/libairspy
> [... 29 lines cut ...]
> /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem
> /usr/include/arm-linux-gnueabihf/qt5/QtXml -isystem
> /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I.
> -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o ofdm-decoder.o
> src/ofdm/ofdm-decoder.cpp

The only thing I can imagine is that you have -DIMPULSE_RESPONSE on and not
-DHAVE_SPECTRUM=On

jbouwh, 2018-04-28:
I removed -DIMPULSE_RESPONSE -DHAVE_SPECTRUM=On was already on.
qt-dab is compiling and running now.

JvanKatwijk, 2018-04-28:
Good to hear.

If I understand you, then you have -DHAVE_SPECTRUM=ON, and not
-DIMPULSE_RESPONSE and the error(s)
occurred with IMPULSE_RESPONSE=ON
in which case the code for impulseresponse should have caused the error

jbouwh, 2018-04-28:
that is correct

JvanKatwijk, 2018-04-28:
Interesting

I compiled qt-dab with option spectrum on, with option impulseresponse on
and off, and with spectrum off (and impulseresponse off) on both an RPI2
and 3
(both running stretch), but all without problem. So, I am not able to
(re)construct the error and I am unable to tell whether it is an error in
the compilation
environment or in the qt-dab software

kaits68, 2018-11-27:
pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# apt-get install -y vim qt5-default qt5-qmake libusb-1.0-0 libusb-1.0-0-dev \
> portaudio19-dev faad libfaad-dev libsndfile1-dev \
> sndfile-tools librtlsdr0 librtlsdr-dev libfftw3-3 libfftw3-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libusb-1.0-0 is already the newest version (2:1.0.21-1).
libusb-1.0-0 set to manually installed.
The following packages were automatically installed and are no longer required:
  erlang-base erlang-crypto erlang-syntax-tools fonts-lato
  libboost-thread1.62.0 libqt5scintilla2-12v5 libqt5scintilla2-l10n
  libqwt-qt5-6 libruby2.3 libscsynth1 libsctp1 libyaml-0-2 rake ruby
  ruby-did-you-mean ruby-minitest ruby-net-telnet ruby-power-assert
  ruby-test-unit ruby2.3 rubygems-integration
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  jackd1 libasound2-dev libdrm-dev libdrm-exynos1 libdrm-omap1 libdrm-tegra0
  libegl1-mesa-dev libfftw3-bin libflac-dev libgles2-mesa-dev libjack-dev
  libjack0 libogg-dev libportaudiocpp0 libpthread-stubs0-dev libqt5opengl5-dev
  libqt5test5 libusb-1.0-doc libvorbis-dev libwayland-bin libwayland-dev
  libx11-dev libx11-doc libx11-xcb-dev libxau-dev libxcb-dri2-0-dev
  libxcb-dri3-dev libxcb-glx0-dev libxcb-present-dev libxcb-randr0-dev
  libxcb-render0-dev libxcb-shape0-dev libxcb-sync-dev libxcb-xfixes0-dev
  libxcb1-dev libxdamage-dev libxdmcp-dev libxext-dev libxfixes-dev
  libxshmfence-dev libxxf86vm-dev libzita-alsa-pcmi0 libzita-resampler1
  qtbase5-dev qtbase5-dev-tools qtchooser uuid-dev vim-runtime
  x11proto-core-dev x11proto-damage-dev x11proto-dri2-dev x11proto-fixes-dev
  x11proto-gl-dev x11proto-input-dev x11proto-kb-dev x11proto-xext-dev
  x11proto-xf86vidmode-dev xorg-sgml-doctools xtrans-dev
Suggested packages:
  jack-tools meterbridge libasound2-doc libfftw3-doc libxcb-doc libxext-doc
  portaudio19-doc default-libmysqlclient-dev firebird-dev libgl1-mesa-dev
  libpq-dev libsqlite3-dev unixodbc-dev ctags vim-doc vim-scripts
Recommended packages:
  qjackctl
The following packages will be REMOVED:
  jackd jackd2 libjack-jackd2-0 qjackctl sonic-pi supercollider-server
The following NEW packages will be installed:
  faad jackd1 libasound2-dev libdrm-dev libdrm-exynos1 libdrm-omap1
  libdrm-tegra0 libegl1-mesa-dev libfaad-dev libfftw3-3 libfftw3-bin
  libfftw3-dev libflac-dev libgles2-mesa-dev libjack-dev libjack0 libogg-dev
  libportaudiocpp0 libpthread-stubs0-dev libqt5opengl5-dev libqt5test5
  librtlsdr-dev librtlsdr0 libsndfile1-dev libusb-1.0-0-dev libusb-1.0-doc
  libvorbis-dev libwayland-bin libwayland-dev libx11-dev libx11-doc
  libx11-xcb-dev libxau-dev libxcb-dri2-0-dev libxcb-dri3-dev libxcb-glx0-dev
  libxcb-present-dev libxcb-randr0-dev libxcb-render0-dev libxcb-shape0-dev
  libxcb-sync-dev libxcb-xfixes0-dev libxcb1-dev libxdamage-dev libxdmcp-dev
  libxext-dev libxfixes-dev libxshmfence-dev libxxf86vm-dev libzita-alsa-pcmi0
  libzita-resampler1 portaudio19-dev qt5-default qt5-qmake qtbase5-dev
  qtbase5-dev-tools qtchooser sndfile-tools uuid-dev vim vim-runtime
  x11proto-core-dev x11proto-damage-dev x11proto-dri2-dev x11proto-fixes-dev
  x11proto-gl-dev x11proto-input-dev x11proto-kb-dev x11proto-xext-dev
  x11proto-xf86vidmode-dev xorg-sgml-doctools xtrans-dev
0 upgraded, 72 newly installed, 6 to remove and 0 not upgraded.
Need to get 19.2 MB of archives.
After this operation, 82.9 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf libasound2-dev armhf 1.1.3-5+rpi3 [262 kB]
Get:3 http://archive.raspberrypi.org/debian stretch/main armhf libegl1-mesa-dev armhf 13.0.6-1+rpi2 [55.3 kB]
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf libgles2-mesa-dev armhf 13.0.6-1+rpi2 [75.4 kB]
Get:5 http://archive.raspberrypi.org/debian stretch/main armhf librtlsdr0 armhf 0.5.3-11+rpt1 [25.1 kB]
Get:6 http://archive.raspberrypi.org/debian stretch/main armhf librtlsdr-dev armhf 0.5.3-11+rpt1 [28.4 kB]
Get:2 http://raspbian.trivini.no/raspbian stretch/main armhf libjack0 armhf 1:0.125.0-2 [50.4 kB]
Get:7 http://raspbian.trivini.no/raspbian stretch/main armhf faad armhf 2.8.0~cvs20161113-1+deb9u1 [34.1 kB]
Get:8 http://raspbian.trivini.no/raspbian stretch/main armhf libzita-alsa-pcmi0 armhf 0.2.0-4 [11.7 kB]
Get:9 http://raspbian.trivini.no/raspbian stretch/main armhf libzita-resampler1 armhf 1.6.0-2 [10.3 kB]
Get:10 http://raspbian.trivini.no/raspbian stretch/main armhf jackd1 armhf 1:0.125.0-2 [237 kB]
Get:11 http://raspbian.trivini.no/raspbian stretch/main armhf libdrm-omap1 armhf 2.4.74-1 [15.1 kB]
Get:12 http://raspbian.trivini.no/raspbian stretch/main armhf libdrm-exynos1 armhf 2.4.74-1 [18.1 kB]
Get:13 http://raspbian.trivini.no/raspbian stretch/main armhf libdrm-tegra0 armhf 2.4.74-1 [14.5 kB]
[... 453 lines cut ...]
/usr/lib/arm-linux-gnueabihf/qt5/bin/uic devices/rtlsdr-handler/rtlsdr-widget.ui -o ui_rtlsdr-widget.h
/usr/lib/arm-linux-gnueabihf/qt5/bin/uic devices/sdrplay-handler/sdrplay-widget.ui -o ui_sdrplay-widget.h
/usr/lib/arm-linux-gnueabihf/qt5/bin/uic devices/hackrf-handler/hackrf-widget.ui -o ui_hackrf-widget.h
/usr/lib/arm-linux-gnueabihf/qt5/bin/uic devices/airspy-handler/airspy-widget.ui -o ui_airspy-widget.h
[... cut ...]
 #include <samplerate.h>
                        ^
compilation terminated.
Makefile:1106: recipe for target 'radio.o' failed
make: *** [radio.o] Error 1
root@raspberrypi:/home/pi/qt-dab# cd linux-bin
root@raspberrypi:/home/pi/qt-dab/linux-bin# ./dab-rpi-0.997
bash: ./dab-rpi-0.997: No such file or directory

Hei. I use first time raspberry and can not go more
or have you new lessons how set up dab radio. i use raspb 3b+ and Raspbian Stretch with desktop and recommended software
raspbian jenny no work for me :(

JvanKatwijk, 2018-11-28:
did you install libsamplerate?

kaits68, 2018-11-28:
how can install libsamplerate?

kaits68, 2018-11-28:
and can use this stich to dab+ radio with raspberry?https://www.dx.com/p/rtl2832u-r820t-mini-dvb-t-dab-fm-usb-digital-tv-dongle-black-2021120#.W_7WqTGNyUk

can use with windows7-yes

JvanKatwijk, 2018-11-28:
It looks like the one I have

kaits68, 2018-11-28:
I used all this command but ...
sudo su -
raspi-config
sudo su -
apt-get update && apt-get dist-upgrade -y

apt-get install -y vim qt5-default qt5-qmake libusb-1.0-0 libusb-1.0-0-dev \
portaudio19-dev faad libfaad-dev libsndfile1-dev \
sndfile-tools librtlsdr0 librtlsdr-dev libfftw3-3 libfftw3-dev

git clone https://github.com/JvanKatwijk/qt-dab
cd qt-dab

vim dab-rpi.pro

qmake
make

cd linux-bin
./dab-rpi-0.997- her is wrong :( or much more??

JvanKatwijk, 2018-11-28:
dab-rpi-0.xxx is at least one year old. You better look at
http://github.com/JvanKatwijk/qt-dab, and read the "README"

kaits68, 2018-11-29:
Whats system is ok-NOOBS or Raspian strech

JvanKatwijk, 2018-11-29:
I run stretch, that is full distro. It is derived from debian - as is
ubuntu - so the description in the README file is reasonably accurate for
stretch

kaits68, 2018-11-29:
I try too strech but... Whats folder i need take this Readme file?

JvanKatwijk, 2018-11-29:
The readme file is part of the source tree. If you downloaded the software,
it is named README.md

kaits68, 2018-11-30:
root@raspberrypi:/home/pi/rtl-sdr/build# sudo ldconfig
root@raspberrypi:/home/pi/rtl-sdr/build# cd ..
root@raspberrypi:/home/pi/rtl-sdr# rm -rf build
root@raspberrypi:/home/pi/rtl-sdr# cd ..
root@raspberrypi:/home/pi# cd host-master/build
bash: cd: host-master/build: No such file or directory   After last command coming this fail

kaits68, 2018-11-30:
first i havening problem with this command:
cmake ../ -DINSTALL_UDEV_RULES=ON -DDETACH_KERNEL_DRIVER=ON 
cannot find cmake folder or something havening
and i install this cmake
sudo apt-get install build-essential cmake cmake-curses-gui pkg-config
this is OK?

JvanKatwijk, 2018-11-30:
The command relates to building the ".so" file for the RTLSDR based devices
did you
a. download the sourcetree
b. cd into this sourcetree

kaits68, 2018-12-02:
Now i get error if i try begin with qt-dab--sdrplay:no library or device.
and error report libmir_sdr.so:cannot open shared object file: No such file or dir.

JvanKatwijk, 2018-12-03:
well, the sdrplay is a device that can be in or out the configuration.
If you have the device, then you should install the library, if not it is
best
to remove it from the configuration

If you are using qmake then you should comment out the line CONFIG +=
sdrplay in the file qt-dab.pro

the lines are in the unix section and read:

# comment or uncomment for the devices you want to have support for
# (you obviously have libraries installed for the selected ones)
CONFIG          += dabstick
CONFIG          += sdrplay
#CONFIG         += rtl_tcp
CONFIG          += airspy
CONFIG          += hackrf
#CONFIG         += elad_s1      # does not work yet

JvanKatwijk, 2018-12-03:
Text from the README file

Options in the configuration are:

  a) select or unselect devices

  b) select the output: soundcard or TCP connection

  c) select or unselect basic MSC data handling

Adding or removing from the configuration is in all cases by commenting or
uncommenting a line in the configuration file.

Comment the lines out by prefixing the line with a `#` in the `qt-dab.pro`
file (section "unix") for the device(s) you want to exclude in the
configuration. In the example below, rtl_tcp (i.e. the connection to the
rtlsdr server) won't be used.
```
CONFIG          += dabstick
CONFIG          += sdrplay
#CONFIG          += rtl_tcp
CONFIG          += airspy

Op ma 3 dec. 2018 om 00:42 schreef kaits68 <notifications@github.com>:

> Now i get error if i try begin with qt-dab--sdrplay:no library or device.
> and error report libmir_sdr.so:cannot open shared object file: No such
> file or dir.
>
> —
> You are receiving this because you commented.
> Reply to this email directly, view it on GitHub
> <https://github.com/JvanKatwijk/qt-dab/issues/102#issuecomment-443553507>,
> or mute the thread
> <https://github.com/notifications/unsubscribe-auth/AITzwGM67bv818BRvBtu63Eo3kfb9BCOks5u1GVfgaJpZM4TcLbn>
> .
>

-- 
Jan van Katwijk

+31 (0)15 3698980
+31 (0) 628260355

kaits68, 2018-12-03:
Maybe problem her- Bus 001 Device 006: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

JvanKatwijk, 2018-12-03:
What is the problem you are referring to

kaits68, 2018-12-03:
RTL2838, This is OK?

JvanKatwijk, 2018-12-03:
Why not?

kaits68, 2018-12-03:
a) the path /usr/local/lib is in the load library paths (i.e. it is named in one of the .conf files in the etc/ld.so.conf directory

b) a file exists in the /etc/udev/rules.d directory describing the device, allowing "ordinary" users to access the device.
Maybe need her change device data or not? 
After i disable Sdr play .

JvanKatwijk, 2018-12-03:
I do not understand the issue/question

I presume you are talking about applying an RT2832 based device.
You did not mention what the problem is (if running, what is the output)

kaits68, 2018-12-04:
Problem is this at No work

JvanKatwijk, 2018-12-04:
Yes, buit you should provide some information so that I can deduce what is
going on.
Pls send me the output of the make process as well as the output of the
program when starting it
then I might be able to see what is going right and wrong

JvanKatwijk, 2018-12-10:
Progress?

kaits68, 2018-12-10:
Nothing😥😥. Maybe can check en time more tomorrow

JvanKatwijk, 2018-12-10:
If you are just interested in running a dab program on an RPI rather than
having to compile and build it yourself,
the "releases: section of the github repo contains an "appimage" for use on
the rpi2/3, an image with everything on/in it
(apart from a driver program of the device)

kaits68, 2018-12-11:
Hei,yesteday is this problem/
root@raspberrypi:/home/pi# cd qt-dab
root@raspberrypi:/home/pi/qt-dab# qmake qt-dab.pro
Project MESSAGE: Current git hash = 4382693
root@raspberrypi:/home/pi/qt-dab# make
gcc -c -pipe -flto -ffast-math -O2 -Wall -W -D_REENTRANT -fPIC -DGITHASH=\"4382693\" -DMSC_DATA__ -DPRESET_NAME -D__THREADED_BACKEND -DHAVE_RTLSDR -DHAVE_SDRPLAY -DHAVE_HACKRF -DHAVE_AIRSPY -D_SEND_DATAGRAM_ -DTRY_EPG -DNEON_AVAILABLE -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_CORE_LIB -I. -I. -I. -Isrc -Iincludes -Iservice-description -Iincludes/protection -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/mot -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/support -Iincludes/support/viterbi_768 -Idevices -Idevices/rawfiles -Idevices/wavfiles -Iincludes/scopes-qwt6 -Ispectrum-viewer -Iimpulse-viewer -Itii-viewer -isystem /usr/local/include -isystem /usr/local/include -isystem /usr/include/qt4/qwt -isystem /usr/include/qt5/qwt -isystem /usr/include/qt4/qwt -isystem /usr/include/qwt -I/usr/local/qwt-6.1.4-svn/ -Idevices/rtlsdr-handler -Idevices/sdrplay-handler -Idevices/hackrf-handler -Idevices/airspy-handler -Idevices/airspy-handler/libairspy -Iincludes/backend/data/epg -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtXml -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o spiral-neon.o src/support/viterbi_768/spiral-neon.c
In file included from src/support/viterbi_768/spiral-neon.c:27:0:
src/support/viterbi_768/SSE2NEON.h: In function ‘_mm_clflush’:
src/support/viterbi_768/SSE2NEON.h:1678:42: warning: unused parameter ‘p’ [-Wunused-parameter]
 FORCE_INLINE void _mm_clflush(void const*p)
                                          ^
In file included from src/support/viterbi_768/SSE2NEON.h:123:0,
                 from src/support/viterbi_768/spiral-neon.c:27:
src/support/viterbi_768/SSE2NEON.h: In function ‘_mm_setzero_si128’:
/usr/lib/gcc/arm-linux-gnueabihf/6/include/arm_neon.h:5792:1: error: inlining failed in call to always_inline ‘vdupq_n_s32’: target specific option mismatch
 vdupq_n_s32 (int32_t __a)
 ^~~~~~~~~~~
In file included from src/support/viterbi_768/spiral-neon.c:27:0:
src/support/viterbi_768/SSE2NEON.h:312:33: note: called from here
src/support/viterbi_768/SSE2NEON.h:230:2:
  (x)
  ~~~                             
src/support/viterbi_768/SSE2NEON.h:312:33:
  return vreinterpretq_m128i_s32(vdupq_n_s32(0));
                                  
Makefile:1946: recipe for target 'spiral-neon.o' failed
make: *** [spiral-neon.o] Error 1

JvanKatwijk, 2018-12-11:
Can you send me the 'qt-dab.pro" file you are using

kaits68, 2018-12-11:
I can  use only libre office at send this fail
[qt-dab.pro.docx](https://github.com/JvanKatwijk/qt-dab/files/2668746/qt-dab.pro.docx)

JvanKatwijk, 2018-12-11:
# you might select SSE if you are compiling on a x64 with SSE support

# and you might select NEON if you are compiling for an arm (however

# have a look at the config section for neon then)

#CONFIG += NEON_RPI2

CONFIG += NEON_RPI3

#CONFIG += SSE

#CONFIG += NO_SSE

You selected NEON_RPI3, in some (many) cases that gives an error. Try
NEON_RPI2 or NO_SSE

Furthermore, I see that you have selected more than one device

# comment or uncomment for the devices you want to have support for

# (you obviously have libraries installed for the selected ones)

CONFIG += dabstick

CONFIG += sdrplay

#CONFIG += rtl_tcp

CONFIG += airspy

CONFIG += hackrf

#CONFIG += elad_s1 # does not work yet

It might be best to comment out the lines for the devices you do not have.
Just to make things easier
(Commenting out is by preceding the line with a #)

kaits68, 2018-12-11:
work with Qt_DAB-ARM.Appimage if this doit evry start time execute file

closed 2020-09-27 by JvanKatwijk