JvanKatwijk/qt-dab #104: Quit button does not quit

andimik, 2018-05-27:
If you start qt-dab from the console and want to exit by pressing the QUIT button, you have to press ctrl+c otherwise it will show and the process hangs

> dabProcessor is stopping
> taak voor replay eindigt hier
> going to delete dabProcessor
> deleted dabProcessor
> .. end the radio silences

JvanKatwijk, 2018-05-27:
OK, I cannot reconstruct that, tonight I'll run it on an Ubuntubox

JvanKatwijk, 2018-08-29:
> Maybe this only occurs in Kubuntu? I still have the same problem.
>
> Will try it on my Rpi3.

Is this still happening?

andimik, 2018-08-29:
Under Kubuntu: Yes, still the same, after pressing quit the GUI disappears, but the console is blocked unless I press CTRL+C.

But under my RPI3 (Stretch, Linux raspberrypi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux) it works. The console is then ready to use.

```
dabProcessor is stopping
taak voor replay eindigt hier
going to delete dabProcessor
deleted dabProcessor
.. end the radio silences
back in main program
It is done

radioInterface is deleted
pi@raspberrypi:~/apps/qt-dab2/build $ 
```

andimik, 2018-09-01:
It's Kubuntu 18.04

andimik, 2018-12-03:
Dear Jan, I found what caused that: It was the hidden `.qt-dab.ini`.

I've remembered it only today, removed it (completely) and now it closes. 

Unfortunately I did not save it so I cannot tell you what (wrong entry) caused it.

closed 2018-12-03 by andimik