JvanKatwijk/qt-dab #105: Segfault in file open dialog - KDE issue?

andimik, 2018-05-27:
I get a segfault when I switch between filetypes:

Open a raw file. Play it.
Open an iq file, but don't select one and cancel
Open an sdr file -> segfault

```
added items to combobox
Suggested size for outputbuffer = 2560
stream opened
stream started
3 file input (.sdr)
kf5.kio.core: KSambaShare: Could not find smb.conf!
bitDepth set to 10
Starting 1

Period = 16000
dabProcessor is stopping
taak voor replay eindigt hier
kf5.kio.widgets: No node found for item that was just removed: QUrl("file:///home/andreas/test.eti")
file  cannot open
file ?: No such file or directory
Segmentation fault (core dumped)

```

This `test.eti` exists in the same directory, but why does qt-dab look for an eti-file?

I've also copied one `*.raw` and one `*.sdr` (but no `*.iq`) file to /tmp, same issue.

But I fear this is a bug in KDE only, see https://api.kde.org/frameworks/kio/html/kdirmodel_8cpp_source.html

I'm using Kubuntu 18.04.

JvanKatwijk, 2018-05-27:
> frameworks/kio/html/kdirmodel_8cpp_source.html

On fedora and Ubuntu this does not happen. Qt-DAB does not know about eti
files
Can you replay it while running gdb, i.e.

gdb qt-dab....
then when the prompt appears, just 'r' for run

The GUI will show, and if you repeat the sequence you described you should
see it crash
then on the prompt type 'bt' (which stands for backtrace)

It should give an indication which part of the software shows this
undesired behaviour

andimik, 2018-05-27:
Ok, here is the output.

```
$ gdb ./qt-dab-1.1-alpha 
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
...
(gdb) r
Starting program: /home/andreas/apps/qt-dab/qt-dab-1.1-alpha 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8fe2700 (LWP 6909)]
[New Thread 0x7fffdb171700 (LWP 6910)]
...
[New Thread 0x7fffcbf65700 (LWP 6911)]
[Thread 0x7fffcbf65700 (LWP 6911) exited]
[New Thread 0x7fffcbf65700 (LWP 6912)]
[Thread 0x7fffcbf65700 (LWP 6912) exited]
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
[New Thread 0x7fffcbf65700 (LWP 6913)]
[Thread 0x7fffcbf65700 (LWP 6913) exited]
[New Thread 0x7fffcbf65700 (LWP 6914)]
[Thread 0x7fffcbf65700 (LWP 6914) exited]
[New Thread 0x7fffcbf65700 (LWP 6915)]
Hostapis: 2

Api 0 is ALSA
Api 1 is OSS
[Thread 0x7fffcbf65700 (LWP 6915) exited]
Investigating Device 0
 (output):item 1 wordt stream 0 (HD-Audio Generic: CX20590 Analog (hw:1,0))
[New Thread 0x7fffcbf65700 (LWP 6916)]
Investigating Device 1

 (output):item 2 wordt stream 1 (pulse)

[Thread 0x7fffcbf65700 (LWP 6916) exited]
[New Thread 0x7fffcbf65700 (LWP 6917)]
[Thread 0x7fffcbf65700 (LWP 6917) exited]
Investigating Device 2

 (output):item 3 wordt stream 2 (default)

added items to combobox
Suggested size for outputbuffer = 2560
[New Thread 0x7fffcbf65700 (LWP 6918)]
stream opened
[New Thread 0x7fffcb764700 (LWP 6919)]
stream started
[New Thread 0x7fffaffff700 (LWP 6920)]
[New Thread 0x7fffaf7fe700 (LWP 6921)]
1 file input (.raw)
kf5.kio.core: KSambaShare: Could not find smb.conf!
[New Thread 0x7fffaeffd700 (LWP 6922)]
bitDepth set to 10
Starting 1

Period = 8000
[New Thread 0x7fffad85b700 (LWP 6923)]
dabProcessor is stopping
[Thread 0x7fffad85b700 (LWP 6923) exited]
taak voor replay eindigt hier
[Thread 0x7fffaeffd700 (LWP 6922) exited]
kf5.kio.widgets: No node found for item that was just removed: QUrl("file:///home/andreas/test.eti")
file  cannot open
file ?: No such file or directory

Thread 1 "qt-dab-1.1-alph" received signal SIGSEGV, Segmentation fault.
0x000055555557132e in RadioInterface::newDevice (this=0x5555561445c0, deviceName=...)
    at ../radio.cpp:1140
1140            inputDevice     -> stopReader ();
[... 31 lines cut ...]
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff5950523 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#20 0x00007ffff5951ff5 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#21 0x00007ffff59292eb in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#22 0x00007fffed0c21c0 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#23 0x00007ffff0fb9287 in g_main_context_dispatch ()
   from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007ffff0fb94c0 in ?? () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007ffff0fb954c in g_main_context_iteration ()
   from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#26 0x00007ffff4e1f8ef in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff4dc49ea in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
---Type <return> to continue, or q <return> to quit---
#28 0x00007ffff4dcda84 in QCoreApplication::exec() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x000055555556a9b3 in main (argc=<optimized out>, argv=<optimized out>)
    at ../main.cpp:117
(gdb) q
A debugging session is active.

        Inferior 1 [process 6904] will be killed.

Quit anyway? (y or n) y
```

JvanKatwijk, 2018-05-27:
Thanks

I think I can identify a sequence of events where a pointer - that should
not be NULL - is NULL actually
I;ll fix it tomorrow night, during daytime I look after a 2-year old (the
youngest of the family)

andimik, 2018-06-17:
Fixed in https://github.com/JvanKatwijk/qt-dab/commit/20bf3bc7f2d412a3491c36bf6aa975b67cdb1f07

closed 2018-06-17 by andimik