JvanKatwijk/qt-dab #106: Quits of own accord

NomisGninnac, 2018-05-31:
Qt-DAB is quitting at Random intervals

andimik, 2018-05-31:
Could you please give us more details? Which OS? What version? What hardware is used?  Is your reception good?

I have never faced randomly quits.

NomisGninnac, 2018-06-01:
Apologies, I should have stated its Raspberry Pi 3 on latest compile.

I couldn't clearly see a log file in Qt-DAB directory.  Might try and launch it from command line to see what that says on quit.  

Timing varies as to when it quits.

NomisGninnac, 2018-06-01:
Segmentation fault
Might be an SD card fault or software fault.  What do you think?

NomisGninnac, 2018-06-02:
I've decided this is a hardware fault on the RPi3.

It appears segmentation faults happen to all sorts of software when a faulty sector exists on the SD Card.

Will close issue now.

closed 2018-06-02 by NomisGninnac

JvanKatwijk, 2018-06-29:
Thanks for the messages, I could not react since I was on holidays and -
believe it or not - with a hardly functioning internet connection,
I really can recommend that