JvanKatwijk/qt-dab #107: Segmentation fault

NomisGninnac, 2018-07-06:
Using RPi3 and latest code.

Randomly quits at any time.

The console gives ::

alternative start carriers -714 -692
alternative start carriers -714 -692
Segmentation fault

JvanKatwijk, 2018-07-06:
please have it run under gdb

NomisGninnac, 2018-07-26:
unable to get it to run under gdb.

issue still relevant, have compiled as of 26/7/18  and still getting the problem after a while.  Longest runtime 20hrs or so.  But all quit with segmentation fault.

NomisGninnac, 2018-07-27:
pi@raspberrypi:~/qt-dab/linux-bin $ gdb ./qt-dab-2.0 
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./qt-dab-2.0...(no debugging symbols found)...done.
(gdb) run gdb ./qt-dab-2.0 
Starting program: /home/pi/qt-dab/linux-bin/qt-dab-2.0 gdb ./qt-dab-2.0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x72a53440 (LWP 17386)]
libEGL warning: DRI2: failed to authenticate
[New Thread 0x6ed7e440 (LWP 17387)]
[New Thread 0x6e57e440 (LWP 17388)]
[New Thread 0x6dd7e440 (LWP 17389)]
[New Thread 0x6d57e440 (LWP 17390)]
qt5ct: using qt5ct plugin
main: Detected system language "en_GB"
main: Set language "en_GB"
main: Error while loading language specifics "en_GB" use English "en_GB" instead
[New Thread 0x6ac4b440 (LWP 17391)]
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
[New Thread 0x6609c440 (LWP 17392)]
[Thread 0x6609c440 (LWP 17392) exited]
[New Thread 0x6609c440 (LWP 17393)]
[Thread 0x6609c440 (LWP 17393) exited]
[New Thread 0x6609c440 (LWP 17394)]
connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
Hostapis: 2

[Thread 0x6609c440 (LWP 17394) exited]
Api 0 is ALSA

Api 1 is OSS

Investigating Device 0

 (output):item 1 wordt stream 0 (bcm2835 ALSA: - (hw:0,0))

Investigating Device 1

 (output):item 2 wordt stream 1 (bcm2835 ALSA: IEC958/HDMI (hw:0,1))

Investigating Device 2

 (output):item 3 wordt stream 2 (sysdefault)

Investigating Device 3

 (output):item 4 wordt stream 3 (dmix)

[New Thread 0x6609c440 (LWP 17397)]
[Thread 0x6609c440 (LWP 17397) exited]
Investigating Device 4

 (output):item 5 wordt stream 4 (default)

added items to combobox
Suggested size for outputbuffer = 0
stream opened
[New Thread 0x6609c440 (LWP 17398)]
stream started
OK, functions seem to be loaded
[New Thread 0x69fc6440 (LWP 17399)]
[Thread 0x69fc6440 (LWP 17399) exited]
[New Thread 0x69fc6440 (LWP 17400)]
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
samplerate set to 2048000
Supported gain values (29): 49.6 48.0 44.5 43.9 43.4 42.1 40.2 38.6 37.2 36.4 33.8 32.8 29.7 28.0 25.4 22.9 20.7 19.7 16.6 15.7 14.4 12.5 8.7 7.7 3.7 2.7 1.4 0.9 0.0 
[New Thread 0x687c5440 (LWP 17401)]
Starting 1

[New Thread 0x67024440 (LWP 17402)]
[New Thread 0x656ff440 (LWP 17403)]
[New Thread 0x64eff440 (LWP 17404)]
[Thread 0x656ff440 (LWP 17403) exited]
protLevel 3, bitRate 64
we now have DAB
[New Thread 0x64eff440 (LWP 17405)]
[Thread 0x64eff440 (LWP 17404) exited]
protLevel 3, bitRate 64
we now have DAB
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId

Thread 19 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x64eff440 (LWP 17405)]
0x75cf56c8 in QByteArray::resize(int) ()
   from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5

JvanKatwijk, 2018-07-28:
> html

OK
that already helps. It shows that you have DAB, rather than DAB+ what is
more commonly used (and mor advanced).
We only have DAB+ here.

The fragments of DAB I have do not show the message "no transportId", the
message indicates an unexpected occurrence
of some data.

Pls run it again, and then, at the end when the segfault appears, give in
the gdb environment the command "bt", that will
give a stacktrace where we can deduce which function causes the trouble

andimik, 2018-07-29:
I will test it with my Rpi3 as well in the next days. Can receive old DAB via Satellite 4 East.

NomisGninnac, 2018-08-01:
I suspect I need to look for a specific DAB+ USB dongle.  I assume my current DABSTICK is not DAB+

I will re-run under GDB and report back with stacktrace

andimik, 2018-08-02:
You are wrong. This is done in software only.

andimik, 2018-08-02:
Could you please try 11A and for example Union Jack or Jazz FM Stereo? They are in DAB+ nationwide.

NomisGninnac, 2018-08-02:
On choosing Jazz FM Stereo I did indeed get DAB+
Console:
[New Thread 0x64eff440 (LWP 4336)]
[Thread 0x656ff440 (LWP 2056) exited]
we now have DAB+

------

Will continue to run in debug mode to get the stacktrace.
Currently in 11A listening mainly to TalkRadio which isn't DAB+

andimik, 2018-08-02:
Of course you will get the message with DAB+ on the console. 

Please test if it crashes also at DAB+.

NomisGninnac, 2018-08-02:
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./qt-dab-2.0...(no debugging symbols found)...done.
(gdb)  run gdb ./qt-dab-2.0
Starting program: /home/pi/qt-dab/linux-bin/qt-dab-2.0 gdb ./qt-dab-2.0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x72a53440 (LWP 2036)]
libEGL warning: DRI2: failed to authenticate
[New Thread 0x6ed7e440 (LWP 2037)]
[New Thread 0x6e57e440 (LWP 2038)]
[New Thread 0x6dd7e440 (LWP 2039)]
[New Thread 0x6d57e440 (LWP 2040)]
qt5ct: using qt5ct plugin
main: Detected system language "en_GB"
main: Set language "en_GB"
main: Error while loading language specifics "en_GB" use English "en_GB" instead
[New Thread 0x6ac4b440 (LWP 2042)]
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
[New Thread 0x6609c440 (LWP 2043)]
[Thread 0x6609c440 (LWP 2043) exited]
[New Thread 0x6609c440 (LWP 2044)]
[Thread 0x6609c440 (LWP 2044) exited]
[New Thread 0x6609c440 (LWP 2045)]
connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
Hostapis: 2

Api 0 is ALSA

Api 1 is OSS
[... 78 lines cut ...]
load glyph failed err=15 face=0x6907a0, glyph=3566
load glyph failed err=15 face=0x6907a0, glyph=3566
sorry no transportId
sorry no transportId
sorry no transportId
sorry no transportId

Thread 21 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x64eff440 (LWP 4339)]
0x75cf56d4 in QByteArray::resize(int) ()
   from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
(gdb) bt
#0  0x75cf56d4 in QByteArray::resize(int) ()
   from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#1  0x0003f300 in motObject::addBodySegment(unsigned char*, short, int, bool)
    ()
#2  0x00038bf4 in padHandler::build_MSC_segment(std::vector<unsigned char, std::allocator<unsigned char> >) ()
#3  0x000396a0 in padHandler::processPAD(unsigned char*, short, unsigned char, unsigned char) ()
#4  0x000369e0 in mp2Processor::addtoFrame(std::vector<unsigned char, std::allocator<unsigned char> >) ()
#5  0x00033054 in audioBackend::process(short*, short) ()
#6  0x00047b2c in mscHandler::run() ()
#7  0x75cf1b20 in ?? () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#8  0x75c0ffc4 in start_thread (arg=0x64eff440) at pthread_create.c:335
Backtrace stopped: Cannot access memory at address 0x38f0386
(gdb)

andimik, 2018-08-03:
I asked you to test the segfault issue with DAB+, not DAB

JvanKatwijk, 2018-08-03:
Thanks for the info. I now have a rough idea where the error occurred, not
yet what the error causes.
I made a source version with some print statements in it, that should help
verifying whether or not
the variables involved have their assumed values.

If you can compile and run it,the output might shed some light over what
variable (value) is the offender
The "tgz" file is in my dropbox, the link is

https://www.dropbox.com/s/6rx4ajrzv5szf87/qt-dab-test.tgz?dl=0

Thanks so far
best
jan

NomisGninnac, 2018-08-03:
Will report back over weekend.  Currently testing DAB+.

andimik, 2018-08-05:
I've tried with a short 3 min sample from BBC which I found on the Ukrainian's programmer page.

I had no problems in these 3 min. But I tried with a raw file.

> Period = 8000
> protLevel 3, bitRate 128
> we now have DAB
> End of file, restarting

Will test it with a real signal as well.

andimik, 2018-08-06:
What was the DAB programme you were listening to? I just saw it is in 64 kBit, but Jan removed the SID and label statement in the console some time ago.

NomisGninnac, 2018-08-06:
The DAB (not DAB+) channel was TalkRadio in block 11A.  I cannot remember the individual programme that it crashed in.  

I am currently running Jazz FM in Block 11.  It seems listening to DAB+ channels does not lead to errors of any kind.

I will restart debugging using QT-DAB-test with TalkRadio (DAB only) once the previous test gets to 70 hours.

I will post the console and stacktrace in a new post once a crash occurs.

NomisGninnac, 2018-08-06:
This is a dump from tonight (6/8/18 11pm)

Was tuned into standard DAB channel TalkRadio on block 11A

pi@raspberrypi:~/Desktop/qt-dab-test/linux-bin $ gdb ./qt-dab-2.0 
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./qt-dab-2.0...done.
(gdb) run gdb qt-dab-2.0 
Starting program: /home/pi/Desktop/qt-dab-test/linux-bin/qt-dab-2.0 gdb qt-dab-2.0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x728cc440 (LWP 23288)]
libEGL warning: DRI2: failed to authenticate
[New Thread 0x6ebfb440 (LWP 23289)]
[New Thread 0x6e3fb440 (LWP 23290)]
[New Thread 0x6dbfb440 (LWP 23291)]
[New Thread 0x6d3fb440 (LWP 23292)]
qt5ct: using qt5ct plugin
main: Detected system language "en_GB"
main: Set language "en_GB"
main: Error while loading language specifics "en_GB" use English "en_GB" instead
[New Thread 0x6aac4440 (LWP 23293)]
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
[New Thread 0x65f1f440 (LWP 23294)]
[Thread 0x65f1f440 (LWP 23294) exited]
[New Thread 0x65f1f440 (LWP 23295)]
[Thread 0x65f1f440 (LWP 23295) exited]
[New Thread 0x65f1f440 (LWP 23296)]
connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
[... 257 lines cut ...]
    b=0x64dfc248 "\377\364\204\300nĕSE6\227+-2\252\025U\027\365`\005TD\016Ih&\243i\256\243*\260\252\352.\313O4\373/@Ӱ4\373\315<\373o:\353\357A<\223\331ht\352\360\354\254\343J\327ƪ\330\372\272\330\371\255k\346\265\363\346\276kZ\326\322Z\211\271\233ιA\021P\300\215\237\221Ok\201\003\"vKr\310\b\326\346\324\061\367qiwE@ΆG7¸7\311l\322Vu\337\v\263\362Y\244h̍\242\316u\335-\343\350\242\345\372\347k<63<\255;7\230\361\356t\031HG5 p\212\304A\b\317\033\035\"B\375\021\252SE)\243\226\275\214!\223\\\255%\256", last=<optimized out>, CI_flag=<optimized out>)
    at src/backend/data/pad-handler.cpp:252
#5  0x0003f858 in padHandler::processPAD (this=this@entry=0x750fe4, 
---Type <return> to continue, or q <return> to quit---
    buffer=buffer@entry=0x64dfc248 "\377\364\204\300nĕSE6\227+-2\252\025U\027\365`\005TD\016Ih&\243i\256\243*\260\252\352.\313O4\373/@Ӱ4\373\315<\373o:\353\357A<\223\331ht\352\360\354\254\343J\327ƪ\330\372\272\330\371\255k\346\265\363\346\276kZ\326\322Z\211\271\233ιA\021P\300\215\237\221Ok\201\003\"vKr\310\b\326\346\324\061\367qiwE@ΆG7¸7\311l\322Vu\337\v\263\362Y\244h̍\242\316u\335-\343\350\242\345\372\347k<63<\255;7\230\361\356t\031HG5 p\212\304A\b\317\033\035\"B\375\021\252SE)\243\226\275\214!\223\\\255%\256", last=<optimized out>, 
    L1=<optimized out>, L0=174 '\256') at src/backend/data/pad-handler.cpp:86
[... cut ...]
    blkno=<optimized out>) at src/backend/msc-handler.cpp:237
#10 0x0003663c in mscHandler::run (this=0x477d28)
    at src/backend/msc-handler.cpp:136
#11 0x75b95b20 in ?? () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#12 0x75ab3fc4 in start_thread (arg=0x64dff440) at pthread_create.c:335
#13 0x7584bbc8 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:76
---Type <return> to continue, or q <return> to quit---
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

andimik, 2018-08-07:
I confirm the random seg faults with Talkradio, even under my PC running under Kubuntu 64bit

I'm using a Linux Sat receiver pointing to 4 East, which feeds eti-tools, then make a modulated DAB signal with odr-dabmod and then I load a raw file from '/dev/stdin' 

`wget -q -O - 'http://ip_of_my_box:8001/1:0:1:50:0:1:30300F:0:0:0:' | ts2na -p 1063 -s 12 | na2ni | odr-dabmod eti2raw.ini | qt-dab-2.0`
(BTW: gdb does not work with this long piped command ...)

I also got some `sorry no transportId` messages. Then it crashed after a while. I have no idea when.

JvanKatwijk, 2018-08-07:
Can you record that into a file? That would really help

andimik, 2018-08-07:
Sure Jan :relaxed:

I have recorded an eti file in the morning of some hundreds megabytes and will play it and check if this one crashes as well.

In the meanwhile please install ODR-DabMod :wink:

But I fear that also has to do with DLS as the static text from talkradio changes each second to nonsense special characters in qt-dab

JvanKatwijk, 2018-08-07:
It shows why the segment violation happens, the direct cause for that has
been eliminated. The repair is pretty crude though
and I have to dig deeper into the issue of trasportId's being 0 or not

andimik, 2018-08-08:
Good news so far. 

With the latest bf355f8 I had no seg fault for more than 19h 30 min and no "sorry no transportId" message.

I'll add a howto in the `README.md` as it is quite simple to load a raw from the stdin:

```
mkfifo stdin.raw
cat eit-file.eti | odr-dabmod ini-file.ini > stdin.raw
```

then you can observe qt-dab with `gdb` in a different console

[image]

But I discovered that the DLS in qt-dab prints characters which are NOT part of the DLS. In case of talkRADIO the DLS is static, see screenshot from XPADxpert and watch the (zipped) small video I took.

[qt-dab_video.zip](https://github.com/JvanKatwijk/qt-dab/files/2271188/qt-dab_video.zip)

[image]

JvanKatwijk, 2018-08-08:
I took a rather crude approach, eliminating some control paths that
would/could lead to strange phenomena such as transportId 0 and negative
sizes for
data segments
If/when you can send a link to a fragment, I can do some research in what I
was doing wrong

NomisGninnac, 2018-08-08:
I will continue testing alongside and posting.  I will try to keep it brief and accurate as I am able to avoid endless emails and notifications.

JvanKatwijk, 2018-08-11:
> [image: screenshot from 2018-08-11 19-19-55]
> <https://user-images.githubusercontent.com/26828910/43994852-a6c9afb2-9d9b-11e8-81dd-483d4cf16124.png>
>
> As you can see I've been running the same channel for 72 hours and 11
> minutes... No errors whatsoever, no stack to report. I think we can
> probably close the issue if we're all in agreement.

Thanks.