JvanKatwijk/qt-dab #108: Implementation Idea - Bluetooth Bridge

daniel-loader, 2018-07-09:
Just found this project and wondered if there's a clear direction, or even just a suggestion of a good direction to go with in respect to bridging this DAB reception to A2DP bluetooth.

Idea being that if you can expose this as a bluetooth device you could change tuning/stations with next and previous track on the client device as well as expose the name of the station as a Track Name or other metadata to the client device.

The use-case for this would be as a car computer to be able to add DAB to a car with A2DP bluetooth but no DAB natively.

I understand this is outside of the scope of the project as it is, but hopefully this may prompt some ideas in this direction as a Pi ZeroW + SDR would make an ideal retrofit.

Having seen DAB to FM bridges I'm not sold that's the right way to go and robs the driver of the ability to use native controls for tuning on the head unit or steering wheel as you'd be able to do with a bluetooth source.

Thanks anyway for a great project!

JvanKatwijk, 2018-07-12:
Interesting idea. Ther eis a version of the dab-cmdline software that is/
can be controlled through an IP port.
I do not have any experience with bluetooth, but - as a matter of fact - I
was (more or less) working on some
extensions to the dab-library sources, to make it easier to switch a
channel or a service.

I'll give adding a "bluetooth server" to the dab-library software some
attention,

best
jan

daniel-loader, 2018-07-12:
That's pretty useful, would be useful even for my bluetooth headphones to be able to scan to next station with the next/last track buttons too!

JvanKatwijk, 2018-07-14:
I have been working on the TCP/IP version, works OK.
Next step is to look at an interface using bluetooth. I presume that the
interface is merely giving commands like "next channel" or "next service"?

daniel-loader, 2018-07-14:
I imagine the AVRCP specification is key here as it handles the client being able to signal to the server.

https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#Audio/Video_Remote_Control_Profile_(AVRCP)

Given it handles metadata in addition to the media controls itself you could present the metadata you currently display when tuned in, Station etc 

AVRCP 1.5 spec includes playlists from what I can see from https://www.bluetooth.org/docman/handlers/DownloadDoc.ashx?doc_id=260861 (Section 5.10 p40) so in addition to being able to dumb skip stations in order you should be able to display the list of tuned stations and switch accordingly, if the client device supports the 1.5 spec.

1.6 includes cover art so that could include the station artwork.

Basically from what I've glanced over the AVRCP spec could expose all of the backend tuner metadata to a client to interface with directly.

In addition to bluetooth headsets it should be useful for TV soundbars that support A2DP or other home speakers. Rather than the speaker being a dumb audio output sink it could be the control interface for the tuner itself.

Thoughts?

JvanKatwijk, 2018-08-27:
Can you suggest a protocol. As such it is quit well possible to build a (simple) server. I do not know anything about android and phones (and would like to keep it that way), however I am absolutely willing to consider building a version with a simple bluetooth server as controller

JvanKatwijk, 2018-10-31:
In the mean time I have been working at/looking into bluetooth and example 7 from the dab-cmdline
repository has a (kind of) bluetooth server. The basics of communication using bluetooth In C and C++
are (more or less) clear. The next step is to get communication in java going, and then the issue is to get things done on android.
My experience with android is 0 (zero), so I could use some help here

JvanKatwijk, 2018-11-25:
Still working on the idea. It seems to me there are two things to consider, 
a. a bluetooth connected headset, 
b. bluetooth control

I have been working on the second issue for a while and right now I created a pretty simple android client - using an old tablet - with the DAB processing being done in a server (running on an RPI 2/3)
While the GUI on the tablet works, it certainly needs attention to make it more attractive
(see the dab-server repository for details)

One of the issues is which device is making the sound audible, the server or the client.
If you look at e.g a mobile environment, then it would make sense to have a phone or tablet as control, and have the server send the audio to some speaker sets. That is the current setup.

If, on the other hand, you consider the tablet/phone or whatever as just a handheld device
that you always have "on your body", then it would make sense to have the sound being delivered
to a headphone connected on the tablet/phone.

Anyway, these were the first steps working with bluetooth and with android,

Comments are (more than) welcome

danielloader, 2018-11-25:
Sorry about the delay, been very busy!
What you've suggested is way more than I expected so anything here was going to be good work - hopefully it's useful to not just you, but many others. 

In my usecase I outlined initially having the audio piped to the bluetooth control client is preferable, though I can see a usecase for using bluetooth to control the server with a locally connected speaker.

In my case I don't necessarily need a bluetooth client as the car system has one, though it's handy to have one as a testing platform when doing the backend. 

Can see it being useful with bluetooth headphones that have next track controls on them to skip through tuned stations.

Any way thanks for investigating this and hope it's proving to be interesting for you!

JvanKatwijk, 2018-11-26:
The next step is extending the server/client such that other devices are
supported. This was basically an exercise in applying bluetooth
and android. So it is most likely that a next extension is the output of
qt-dab over Bluetooth to a wireless headphone or a speakerset

Just nice to explore some areas that I did not work with before

JvanKatwijk, 2018-11-29:
the interesting thing is that sending the output over bluetooth is available.
Assuming you are running a Linux box, then install the bluetooth software (bluez)
The approach from https://markus.jarvisalo.dy.fi/2017/12/making-the-raspberry-pi-3-a-bluetooth-audio-receiver/ on an RPI 3 was implemented this afternoon, and I could listen - using a bluetooth connection - with a phone connected to the RPI while the dab software was running on my laptop.
So, it seems all ingredients are available

closed 2020-09-27 by JvanKatwijk