JvanKatwijk/qt-dab #111: 2.0 does not show signal in spectrum view ... receiving works

DG9BFC, 2018-07-15:
title says all ... i used 1.1 for setting up antenna (for best signal to noise ratio and fine error-less receive on all stations in my area)
2.0 shows the "cross" in right window (or 4 spots) but no signal in left window of spectrum view
receiving works fine in both versions
using the windoze version on a win 10 tablet
any more details needed for finding that bug?
dg9bfc sigi
ps ... superb software :-) thumbs up

andimik, 2018-07-15:
Here it works with 2.0 under kubuntu

DG9BFC, 2018-07-15:
hmmm ... is there a setting to be done in a config file(uncomment signal meter)??
greetz sigi

andimik, 2018-07-15:
I had once the situation that the spectrum window did not appear.

JvanKatwijk, 2018-07-15:
Apologies I just forgot to upload a fixed version - has detected the error
on Linux already.
Is fixed
best
jan

DG9BFC, 2018-07-16:
aehem ... is it fixed now in version 2.0 ???

JvanKatwijk, 2018-07-16:
It should be, on my W10 box it runs fine

andimik, 2018-07-16:
Me too, but following the instructions also worked for me.

JvanKatwijk, 2018-07-16:
The windows-bin.zip file was updated and contains a new version of
qt-dab-2.0

closed 2019-05-05 by JvanKatwijk