JvanKatwijk/qt-dab #112: SDRPlay not working with SDRPlay 2

hitechguy18, 2018-08-03:
seems the auto gain / manual gain just brakes everything
added items to combobox
Suggested size for outputbuffer = 0
stream opened
stream started
hwVer = 2
devicename = \\?\usb#vid_1df7&pid_3010#1713000020#{7d7faf9d-8656-464a-8ab5-7922beb22b88}
Starting 1

If i use ExtIO_SDRplay_RSP2.dll V1.2 Build 0710 everything works
added items to combobox
Suggested size for outputbuffer = 0
stream opened
stream started
Last error = 0
got a status cmd 100
inputRate = 2000000
hardware type = 3
Hw open successful
setting freq to 218640000
got a status cmd 102
got a status cmd 103
restart reader entered (218640000)
restart reader returned with 32768
Restart met block 32768
now we have restarted the reader
Starting 1

JvanKatwijk, 2018-08-03:
You are using an EXTIO lib. I did not look at that for years. I'll have a
look

JvanKatwijk, 2018-08-03:
I just ran a quick test with (a) an RSP II and an RSP 11A, selected the
appropriate extio dll's and did not encounter a problem

Can you more more specific on your set up?

2018-08-03 18:14 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> You are using an EXTIO lib. I did not look at that for years. I'll have a
> look
>
>
>
>> seems the auto gain / manual gain just brakes everything
>> added items to combobox
>> Suggested size for outputbuffer = 0
>> stream opened
>> stream started
>> hwVer = 2
>> devicename = \?\usb#vid_1df7&pid_3010#1713000020#{7d7faf9d-8656-464a-
>> 8ab5-7922beb22b88}
>> Starting 1
>>
>> If i use ExtIO_SDRplay_RSP2.dll V1.2 Build 0710 everything works
>> added items to combobox
>> Suggested size for outputbuffer = 0
>> stream opened
>> stream started
>> Last error = 0
>> got a status cmd 100
>> inputRate = 2000000
>> hardware type = 3
>> Hw open successful
>> setting freq to 218640000
>> got a status cmd 102
>> got a status cmd 103
>> restart reader entered (218640000)
>> restart reader returned with 32768
>> Restart met block 32768
>> now we have restarted the reader
>> Starting 1
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980
> +31 (0) 628260355

hitechguy18, 2018-08-03:
This is what installed Right Now
From
https://www.sdrplay.com/downloads/
SDRuno – v1.22 (13th Jan 2018)
(RSP1/RSP1A/RSP2) Includes hardware driver
I did have
SDRuno – v1.23 (19th June 2018) 
(RSPduo) Includes hardware drive
 [uninstalled now even if it did have the driver for SDRPlay RSP2 just not the new one ]
along with
SDR Console 3.0.1 Release
https://www.sdr-radio.com/Console

Working 
SDR Console 3.0.1 & SDRuno – v1.22
if there was some thing up with my SDRplay RSP2 driver none of these will be working ... right

Some how i must of gotten/ had an odd   ExtIO_SDRplay_RSP2.DLL ver 
Folder:C:\Program Files (x86)\SDRplay then copied to Both qt-dab-2.0 & qt-dab-1.1-alpha Folders

qt-dab-1.1-alpha > ExtIO_SDRplay_RSP2.dll - failed to map DLL functions
note: qt-dab-1.1-alpha >  device: SDRplay works fine under SDRuno – v1.22 (13th Jan 2018) install
note: qt-dab-1.1-alpha >  device: SDRplay work fine under SDRuno – v1.23 (19th June 2018) install
 I did do a few re Install an a fresh of qt-dab-2.0 +SDRuno – v1.22  so thi may be why ever thing is now braking in qt-dab-2.0

it must of been this one
ExtIO Plugins – v1.2 (13th July 2018)
(RSP1/RSP1A/RSP2) Includes hardware driver/API
from https://www.sdrplay.com/downloads/

the ExtIO_SDRplay_RSP2.DLL in this one seem to like working in Both qt-dab-2.0 & qt-dab-1.1-alpha well it was until i decided to reinstall **SDRuno – v1.22 (13th Jan 2018)** over **ExtIO Plugins – v1.2 (13th July 2018)** and re copy the C:\Program Files (x86)\SDRplay\ExtIO_SDRplay_RSP2.DLL to other folders [Doh]

ExtIO_SDRplay_RSP2.DLL the seems to be 2 of them for me  
SDRuno – v1.22 (13th Jan 2018) / ExtIO Plugins – v1.2 (13th July 2018) < however this one Dll seem a little more universal due SDRuno – v1.23 (19th June 2018)  was installed an evy thing else was still working 
ExtIO_SDRplay_RSP2.DLL was happy to run in qt-dab-2.0 untill i put 
SDRuno – v1.22 (13th Jan 2018) ExtIO_SDRplay_RSP2.DLL dill in... qt-dab-1.1-alpha& qt-dab-2.0 are both dont like this dll

hitechguy18, 2018-08-03:
If you have Discord: hi_tech_guy_18 #4775  + Teamviewer (get code form discord) I'm sure you can nit pick at this problem

hitechguy18, 2018-08-03:
ah yes  i do have a few versions  of
ExtIO_SDRplay_RSP2.dll one moment

hitechguy18, 2018-08-03:
Working [ExtIO_SDRplay_RSP2- Working.zip](https://github.com/JvanKatwijk/qt-dab/files/2258473/ExtIO_SDRplay_RSP2-.Working.zip)
Not Working[ExtIO_SDRplay_RSP2- Not Working.zip](https://github.com/JvanKatwijk/qt-dab/files/2258478/ExtIO_SDRplay_RSP2-.Not.Working.zip)

JvanKatwijk, 2018-08-03:
OK,

could you give the most recent qt-dab version a try (in the most recent
window-bin.zip in the releases section in the qt-dab repository),
there were quite some changes a.o with the API

hitechguy18, 2018-08-03:
this one 
qt-dab-2.0
https://github.com/JvanKatwijk/qt-dab/releases/tag/qt-dab-2.0
and this one
qt-dab-1.1-alpha
https://github.com/JvanKatwijk/qt-dab/releases/tag/qt-dab-1.1-alpha

hitechguy18, 2018-08-03:
Working ExtIO_SDRplay_RSP2- Working.zip qt-dab-2.0 & qt-dab-1.1-alpha liked this one
Not WorkingExtIO_SDRplay_RSP2- qt-dab-2.0 & qt-dab-1.1-alpha both hated this one Both= api errors

JvanKatwijk, 2018-08-03:
The confusing thing is that I run qt-dab-2.0 with the most recent extio-dll
(I do not have sdruno installed)
The older qt-dab-1.1-xxx will nof run with the 2.13 api (some
incompatibility between 2.13 and older versions)

I run some tests tomorrow

hitechguy18, 2018-08-03:
Trust me it confusing me too

JvanKatwijk, 2018-08-04:
Running with the configuration is specified here does not show problems,
in fact the extio-dll works well with other programs as well

You can see on the attached picture the version of the extio-dll.

do you also have problems with running with the included sdrplay handler?
As said, the 2.13 lib is incompatible with the older sdrplay libs (a
parameter is added to a callback function), so older precompiled
versions, like the 1.1 version of qt-dab will not work when addressing the
built-in sdrplay handler

closed 2020-09-27 by JvanKatwijk