JvanKatwijk/qt-dab #115: core getting dumped on execution

aswinp17, 2018-10-17:
*** Error in `./qt-dab-2.0': realloc(): invalid pointer: 0x00000000006bb1b0 ***
Aborted (core dumped)

aswinp17, 2018-10-17:
i am using a 64x ubuntu 14.04. compiled and build the code using cmake and make .Then on running the executable file the above error occurs.

aswinp17, 2018-10-17:
ran the executable in gdb and the log showed 

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
*** Error in `/home/inntot/test/qt-dab/build/qt-dab-2.0': realloc(): invalid old size: 0x0000000000681ce0 ***

Program received signal SIGABRT, Aborted.
0x00007ffff7a47c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff7a47c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff7a4b028 in __GI_abort () at abort.c:89
#2  0x00007ffff7a842a4 in __libc_message (do_abort=do_abort@entry=1, fmt=fmt@entry=0x7ffff7b96350 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff7a8f007 in malloc_printerr (action=<optimised out>, str=0x7ffff7b92586 "realloc(): invalid old size", ptr=<optimised out>)
    at malloc.c:4998
#4  0x00007ffff7a93467 in _int_realloc (av=<optimised out>, 
    oldp=0x681cd0 <vtable for std::basic_filebuf<char, std::char_traits<char> >@@GLIBCXX_3.4+112>, oldsize=<optimised out>, nb=<optimised out>)
    at malloc.c:4236
#5  0x00007ffff7a942f9 in __GI___libc_realloc (oldmem=0x681ce0 <QListData::shared_null>, bytes=32) at malloc.c:3031
#6  0x00007ffff528ee3e in QListData::realloc(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff528eefa in QListData::append(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff537bd8c in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff5377e29 in qRegisterResourceData(int, unsigned char const*, unsigned char const*, unsigned char const*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff3186f43 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#11 0x00007ffff7dea1da in call_init (l=<optimised out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffded8, env=env@entry=0x7fffffffdee8)
    at dl-init.c:78
#12 0x00007ffff7dea2c3 in call_init (env=<optimised out>, argv=<optimised out>, argc=<optimised out>, l=<optimised out>) at dl-init.c:36
#13 _dl_init (main_map=0x7ffff7ffe1c8, argc=1, argv=0x7fffffffded8, env=0x7fffffffdee8) at dl-init.c:126
#14 0x00007ffff7ddb29a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#15 0x0000000000000001 in ?? ()
#16 0x00007fffffffe27f in ?? ()
#17 0x0000000000000000 in ?? ()

JvanKatwijk, 2018-10-17:
which qwt are you using

aswinp17, 2018-10-17:
qwt6.1.3

JvanKatwijk, 2018-10-17:
I was not clear. On Ubuntu >= 14.04 one may choose between qt4 and qt5 and
for each of them there exists a qwt version
While it is possible to compile and link qt4 with a qwt version, linked
again qt5, it gives roughly the error you described

So, the real question is
qt4 or qt5
against which qt-x version is the qwt lib compiled

aswinp17, 2018-10-17:
its qt5

aswinp17, 2018-10-17:
yes

JvanKatwijk, 2018-10-17:
Then I do not know the answer, too little info to help you

aswinp17, 2018-10-17:
thanks , fixed the error when re compiled with linking qt4. :+1:

JvanKatwijk, 2018-10-17:
end well. all well

JvanKatwijk, 2018-10-17:
> hai got some problem again when i try to open a .iq file in qt-dab the
> program crashes on segmentation fault.
>
> this is the log :
>
> Starting program: /home/inntot/test/testq/qt-dab/build/qt-dab-2.0
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> [New Thread 0x7fffe8568700 (LWP 12868)]
> [New Thread 0x7fffe7d67700 (LWP 12869)]
> [New Thread 0x7fffe6447700 (LWP 12870)]
> [New Thread 0x7fffe5c46700 (LWP 12871)]
> [New Thread 0x7fffe5445700 (LWP 12872)]
> main: Detected system language "en_GB"
> main: Set language "en_GB"
> main: Error while loading language specifics "en_GB" use English "en_GB"
> instead
> ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
> ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM
> cards.pcm.center_lfe
> ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
> [New Thread 0x7fffc9f44700 (LWP 12874)]
> [Thread 0x7fffc9f44700 (LWP 12874) exited]
> [New Thread 0x7fffc9f44700 (LWP 12875)]
> [Thread 0x7fffc9f44700 (LWP 12875) exited]
> [... 226 lines cut ...]
> Program terminated with signal SIGSEGV, Segmentation fault.
> The program no longer exists.
> (gdb) bt
> No stack.
> (gdb)

The appimage - can be downloaded from the github site - is generated in an
ubuntu trusty environment.
Pls check whether or not the behaviour is there the same.

aswinp17, 2018-10-17:
hi i checked the appimage loaded the program. but on opening an .iq file the program crashed

aswinp17, 2018-10-17:
[image]
[image]
[image]

JvanKatwijk, 2018-10-17:
Ok I love see if I can let the same happen

aswinp17, 2018-10-17:
please let me know how can i fix this.

JvanKatwijk, 2018-10-17:
The error does not appear on my Ubuntu 18 and not on my Fedora system, so
the reason for the error is not clear.
Can you send me a link to the .iq file, then at least we know we are
talking about the same input
(btw I'll be absent this afternoon)

andimik, 2018-10-17:
Please note that the frequency for 11D is different 

It's 222.064 Mhz

aswinp17, 2018-10-17:
OK ,  can you please send link to the .iq file you are using ?

andimik, 2018-10-17:
I can confirm that qt-dab crashes when you want to load the iq file. 

But if you rename it to *.raw it works. It's u8, too.

Thanks for the file. This was from 2015 when the BR 11D had DAB, too.

andimik, 2018-10-17:
Strange, now it works.

JvanKatwijk, 2018-10-17:
The issue is recognized and an temporary solution is created.
The issue shows that sometimes a nice gadget - in this case showing the
progress of the consumption of an input file -
introduces problems.
Reading files will be redesigned such that that progress bar (switched off
now) is again introduced

The current git content - and the appimage - do show a progress bar when
reading a file,
however there will not be any progres reported/

JvanKatwijk, 2018-10-18:
It was an interesting error, did not appear on recent qt implementations.
Took some redesign of the file readers but - as far as I can see - the
progress-bars
are working again

aswinp17, 2018-10-18:
yeah it was :). glad it was fixed so quickly