JvanKatwijk/qt-dab #116: Build error using cmake on Arch

zygmund2000, 2018-10-26:
Hi,

I've built qt-dab many times using cmake on Arch system but today I can't 
I have qwt  6.1.3-3 but and error like this:

-- The C compiler identification is GNU 8.2.1
-- The CXX compiler identification is GNU 8.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.5.3") 
CMake Warning at CMakeLists.txt:107 (find_package):
  By not providing "FindQwt.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qwt", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qwt" with any of
  the following names:

    QwtConfig.cmake
    qwt-config.cmake

  Add the installation prefix of "Qwt" to CMAKE_PREFIX_PATH or set "Qwt_DIR"
  to a directory containing one of the above files.  If "Qwt" provides a
  separate development package or SDK, be sure it has been installed.

CMake Error at CMakeLists.txt:109 (message):
  please install qwt library

JvanKatwijk, 2018-10-26:
In the directory cmake/Modules a file with the name FindQwt.cmake is now
needed. The CMakeLists.txt file contained a typo as well that is corrected

zygmund2000, 2018-10-26:
Well, so where can I find this file? I don't have it in whole system.

zygmund2000, 2018-10-28:
Thanks, now it works, but I found another issue: pulseaudio output is broken or missing, plays only with eg. hw:1,0.

JvanKatwijk, 2018-10-28:
OK,  that is difficult to address for me, on my Fedora and Ubuntu boxes
(just briefly tested) it works fine. Do you use a portaudio from a
repository
or did you compile it yourself (and did you upgrade/update your system?)

zygmund2000, 2018-10-28:
At the moment I can only say, my every other app plays with pulseaudio and I see it in pavucontrol mixer but qt-dab no.
Some time ago pulse out was available in select audio device list, today is not.
https://i.imgur.com/ksMSNjj.png

zygmund2000, 2018-10-28:
I think it is portaudio bug, I'll try build it instead of Arch default package.

zygmund2000, 2018-10-29:
Yes, it is something wrong with portaudio because Audacity also can't play to pulseaudio, I asked some other people and they confirm same issue.

zygmund2000, 2018-11-06:
This pulseaudio issue was affected by new alsa-lib v1.1.7 small bug, so don't worry about it.

JvanKatwijk, 2018-11-06:
Thanks for the effort and the report

closed 2018-11-06 by zygmund2000