JvanKatwijk/qt-dab #120: QT stops playing, SNR decreases and Coarse offset running

olilesp, 2018-12-17:
Hi

since weeks, perhaps since using it with a new stick, QT stops playing any DAB station, the SNR decreases to 1-4 (habitually 19-26) and the coarse information will up or down randomly ans speedy.
I'm using RT2832 DAB Stick FC0013. Before, using a RTL SDR DAB Stick RTL2832 RT802.
Drivers are updated with ZADIG

This problem shows on QT 2.0 from July,3 or in the Continuous build.
Autogain on or not, this is the same.
This "stop" receiving is  while 5-10 minutes or 30 seconds randomly... 
If i select another station from the list, i can't get any sound
To resolve it, i must click on RESET and i get again all the stations available.

JvanKatwijk, 2018-12-17:
Thanks for the report

few questions
you can still use the old stick and see the same behaviour?
you ar erunning under Windows, is the behaviour using an extio plugin the
same

JvanKatwijk, 2018-12-18:
I ran some tests with an FC0013 (Noxon) stick this morning. It seems the device
is quite sensitive to changes in the temperature. In the first half hour the device
was not very stable in the selected frequency. The second test, a few hours later, the
frequency stability was reasonable

olilesp, 2018-12-28:
Yes. Tested too this night. Seems to be stable after hours. 
Solved.

closed 2019-02-27 by olilesp