JvanKatwijk/qt-dab #122: Curious CPU usage patten

troffasky, 2019-01-14:
CPU usage seems to follow a "sine-wave" pattern whilst listening with Qt-DAB. The cycle is about 30 seconds. Goes as low as 5% and as high as 30%.
Running the latest AppImage on Debian.

troffasky, 2019-01-14:
Screenshot:
[image]

andimik, 2019-01-15:
I tried both, the compiled version (see 1st screenshot) and the appimage (2nd).

[image]

[image]

Remark: The increase at the end is due to starting the screencapture programme.

It's about 1 min each. I'm using the feed on Satellite, but tuned to the same channel like you.

JvanKatwijk, 2019-01-15:
Looks interesting, but I cannot reproduce it. Can you describe the system
uou are using (number of cores),  You run on a pretty fast cpu I see
(mine is app 2.5G). The load I normally have is more or less equally
distributed over 4 cores (not surprisingly, I tried to configure the load
such
that 4 cores could do the job easily)

troffasky, 2019-01-15:
It's an Athlon X4 860k.  It occurs to me that Gkrellm is configured to aggregate all 4 cores together in that view, so I will re-test with them shown individually.
However, I don't think it's just an artefact of how Gkrellm shows the CPU utilisation, as Qt-DAB's assessment of its own CPU usage follows the same trend as Gkrellm.

troffasky, 2019-01-15:
Not really sure this screenshot throws any more light on the situation:
[image]
Oddly enough, 'top' shows Qt-DAB using 49-51% CPU consistently which contradicts what gkrellm says. dstat agrees that usr CPU time rises and falls:
[image]

JvanKatwijk, 2019-01-16:
Still strange. I installed gkrellm and on my - much slower - laptop, the
load is - more or less - continuous over
time. You have a pretty fast machine, and processing is essentially clock
based, so apparently the processors
have to wait from time to time on data to be available for processing.  The
"load" on my 2.4G laptop is shown
in the picture

andimik, 2019-01-16:
Jan, could you upload the picture in github.com? It has not been transferred with your reply.

JvanKatwijk, 2019-01-16:
[image]

JvanKatwijk, 2019-01-16:
Distribution of load plays a role when running on e.g. an RPI2, the original version overloaded one of the
cores, now it is more or less evenly distributed over the 4 cores
Anyway, it is an interesting issue. Thanks for sharing it

troffasky, 2019-02-10:
welle.io also exhibits the same behaviour.

closed 2019-02-10 by troffasky