JvanKatwijk/qt-dab #124: Dependency breaks wine

Twilight-Logic, 2019-02-20:
A dependency - portaudio19-dev - used by qt-dab broke wine on my system. Not sure why a dev package should uninstall binaries, but this uninstalled wine and left the system in a state where it was no possible to re-install it again. Removing portaudio19-dev allowed wine to be re-installed again and everything brought back to normality. I'm flagging this up in case it happens to anyone else. I am compiling on Linux Mint 19 with Cinamon.

> Start-Date: 2019-02-19  23:38:29
> Commandline: /usr/bin/apt install portaudio19-dev
> Requested-By: johnc (1000)
> Install: libportaudiocpp0:amd64 (19.6.0-1, automatic), portaudio19-dev:amd64 (19.6.0-1), uuid-dev:amd64 (2.31.1-0.4ubuntu3.3, automatic), libjack-dev:amd64 (1:0.125.0-3, automatic), libjack0:amd64 (1:0.125.0-3, automatic)
> Remove: simplescreenrecorder:amd64 (0.3.8-3), libfluidsynth1:i386 (1.1.9-1), libjack-jackd2-0:amd64 (1.9.12~dfsg-2), libjack-jackd2-0:i386 (1.9.12~dfsg-2), wine-stable:amd64 (3.0.0~xenial), gstreamer0.10-plugins-good:amd64 (0.10.31-3+nmu4ubuntu2.16.04.3), gstreamer0.10-plugins-good:i386 (0.10.31-3+nmu4ubuntu2.16.04.3), winehq-stable:amd64 (3.0.0~xenial), libasound2-plugins:i386 (1.1.1-1ubuntu1), wine-stable-i386:i386 (3.0.0~xenial), libsdl-mixer1.2:i386 (1.2.12-14)
> End-Date: 2019-02-19  23:38:33

JvanKatwijk, 2019-02-21:
The explanation could be that wine in the Ubuntu version uses the
portaudio-18 version.
(for whatever reason, portaudio-18 is as far as I know still the default
portaudio version on Ubuntu).
The precompiled wine then depends on the old portaudio version.

Twilight-Logic, 2019-02-21:
Once I removed portaudio19-dev and restored wine back on my system I checked to see if portaudio is installed. It is:

> $ dpkg -l |grep portaudio
> ii  libportaudio2:amd64                                         19.6.0-1                                    amd64        Portable audio I/O - shared library

However, there seems to be no libportaudio2-dev package, only libportaudio19-dev. There is a libportaudiocpp0 packahe though described as 'C++ bindings - shared'. Would this be the same as a -dev package?

JvanKatwijk, 2019-02-21:
libportaudio2-dev is the package. I'll have to look at my Ubuntu box to see
what packages there are. In your previous mail it showed
that portaudiocpp0 is installed together with portaudio19-dev.

JvanKatwijk, 2019-02-21:
Well,
the strange thing is that the Ubuntu 18 (and mint 19 is based on that) had
as default (and only) portaudio package the portaudio-2 package, which is
the -19 version. So it is not obvious that wine was compiled against
portaudio-v18. Strange

If you want borh qt-dab and wine, you can always use the appImage. The
drawback is that it is compiled in a Ubuntu 14 environment and
always requests a passwd since it installes the udev files for dabsticks

best
jan

closed 2020-09-27 by JvanKatwijk