JvanKatwijk/qt-dab #125: Linux version now working with RSP2

Twilight-Logic, 2019-02-20:
I managed to compile this yesterday but although it works with RTLSDR, it does not work properly with the SDRplay RSP2.

The RSP2 comes up as an device option and I can select it, but I get no stations. the behaviour is different to when it is used with RTLSDR. With RTLSDR you select a channel and then scan it and if you pick one where stations are detected they will be listed. With the RSP2 it repeatedly cycles quite quickly though all the channels, some quicker than others and detects nothing. Selecting auto gain seems to allow it to take more time, but again no channels detected.

I'm running on Linux 19 with Cinnamon. I have version 2.13 of the SDRplay API installed. Works fine with CubicSDR. With the exception of the device selection, setting lna state to 3 as per your example graphic, and experimenting with agc, I am using pretty much default settings. I compiled QtDab-2.1.

JvanKatwijk, 2019-02-21:
Does not look good. If you have the sdrplay attached and select a channel
where you are certain that there is dabdata, what does the spectrum show?

Twilight-Logic, 2019-02-21:
Overload! I don't think I can attach images here, but I compared the RTL with the SDRplay and the spectrum display showed overload on the SDRplay. I found that increasing the LNA state selector to 6 allowed stations to be found and the signal to be decoded. I still need to improve my antenna, but it does seem to work after all.

JvanKatwijk, 2019-02-21:
Yes that is certainly a possibility. The rsp2 is pretty sensitive and even
with a small indoor antenna I usually have the lna state > 4

closed 2019-05-05 by JvanKatwijk