JvanKatwijk/qt-dab #126: qt-dab project with Raspi 3 and Adafruit 3.5" TFT touchscreen

Spectrumanalyzer, 2019-03-17:
Dear Jan,
Beste Jan,
[image]
[image]

Many congratulations on your qt-dab project!
I integrated a Raspi3 running your Appimage +  Adafruit 3.5" TFT touchscreen in my existing self-made audio console.  It works like a charm :-)

Just wanted to let you know this.

Groetjes uit België
Joris

phil-audio, 2019-03-20:
Hello Joris

can you give me a little more precision on your realization.
I am really beginner in the middle RPI3 and I would also like to be able to make a DAB + reader with this RPI3.
I am from the region of Tournai in Belgium.

Did you compile an image from Rasbian?
or have you taken the DT-dab-ARM.appimage 2.0 version?

I have a screen KUMAN 3.5 HDMI
Thank you

Spectrumanalyzer, 2019-03-20:
Dear Phil,

I indeed used the QT-dab-ARM appimage by Jan Van Katwijk.  Don't forget to use 'sudo chmod +x filename' to make it executable and to make sure the drivers for your SDR Dabstick are installed.

hope this helps,

phil-audio, 2019-03-20:
Thank you Joris for the information, I will test this, I have never used the file APPimage,
to follow, I'll let you know.
your audio output comes from the raspberry or did you add a card in addition
I have in stock a X400 suptronics
Thank you Philippe

Spectrumanalyzer, 2019-03-20:
I use the RPI and the quality is very good

rgrds 
Joris

Spectrumanalyzer, 2019-03-20:
RPI- audio I mean..

JvanKatwijk, 2019-03-21:
Whow, that really looks nice
Thanks

ab1jx, 2019-04-27:
No signals here in the US.  That's politics I think.
[image]

Spectrumanalyzer, 2019-04-28:
From what I read on the internet, there are no DAB transmissions in the US. That's a pity because it works really well.  Maybe you can move to Europe :)

JvanKatwijk, 2019-04-28:
I am sorry to hear that, if you are interested to see what DAB is, I can
send you a link to a recording

ab1jx, 2019-04-28:
Yes, it's fairly well documented that it won't work, I just didn't check first.  https://en.wikipedia.org/wiki/Digital_radio  So we don't have DAB and we don't have DVB-T, guess that just leaves SDR for the dongles.

This sounds very similar to Radio Mondiale https://en.wikipedia.org/wiki/Digital_Radio_Mondiale but over VHF instead of HF.  I sort of had that working a few years ago, but if I remember right finding a receiver with enough bandwidth was an issue.

JvanKatwijk, 2019-04-28:
While both DAB and DRM use OFDM, there are differences. From an
implementers point of view, DRM is more difficult than DAB,
DAB frames are separated by a section without any transmitted data, in DRM
the frames follow each other directly
Furthermore, while DAB data is encoded in a differtial psk style, for DRM,
with QAM16 and 64 more work is to be done to
reconstruct the original data.

I do not know whether or not there is DRM is the US

best
jan

ab1jx, 2019-04-28:
As far as I can see we just have IBOC https://en.wikipedia.org/wiki/In-band_on-channel, which is proprietary.  The FCC isn't willing to give a slot in the 173 MHz range I think.  Frequencies are actually auctioned here, it's quite commercial.

andimik, 2019-04-29:
Have a look at https://github.com/argilo/gr-nrsc5

This is for HD Radio which is used in US.

Not tested from my side, as the only were tests in Romania and Germany.

ab1jx, 2019-05-01:
That's transmit (using Gnuradio), https://github.com/theori-io/nrsc5/ is for receive using librtlsdr.

closed 2020-09-27 by JvanKatwijk