JvanKatwijk/qt-dab #127: compilation fail on RPi3

NomisGninnac, 2019-04-07:
g++ -c -pipe -std=c++11 -O3 -ffast-math -isystem /usr/include/arm-linux-gnueabihf/qt5 -O2 -Wall -W -D_REENTRANT -fPIC -DGITHASH=\"b4c7bc6\" -DMSC_DATA__ -DPRESET_NAME -D__THREADED_BACKEND -D__COMPUTE_ECHO__ -DHAVE_RTLSDR -DHAVE_RTL_TCP -D_SEND_DATAGRAM_ -DTRY_EPG -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_CORE_LIB -I. -I. -I. -Isrc -Iincludes -Iservice-description -Iincludes/protection -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/mot -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/support -Iincludes/support/viterbi-jan -Idevices -Idevices/rawfiles -Idevices/wavfiles -Iincludes/scopes-qwt6 -Ispectrum-viewer -Iimpulse-viewer -Itii-viewer -isystem /usr/local/include -isystem /usr/local/include -isystem /usr/include/qt4/qwt -isystem /usr/include/qt5/qwt -isystem /usr/include/qt4/qwt -isystem /usr/include/qwt -I/usr/local/qwt-6.1.4-svn/ -Idevices/rtlsdr-handler -Idevices/rtl_tcp -Iincludes/backend/data/epg -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtXml -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o moc_rtl_tcp_client.o moc_rtl_tcp_client.cpp
g++ -Wl,-O1 -o linux-bin/qt-dab-2.5 main.o radio.o dab-processor.o audio-descriptor.o data-descriptor.o timesyncer.o sample-reader.o ofdm-decoder.o phasereference.o phasetable.o freq-interleaver.o tii_table.o tii_detector.o fic-handler.o fib-decoder.o protTables.o protection.o eep-protection.o uep-protection.o msc-handler.o galois.o reed-solomon.o rscodec.o charsets.o firecode-checker.o frame-processor.o backend.o backend-driver.o backend-deconvolver.o mp2processor.o mp4processor.o faad-decoder.o pad-handler.o data-processor.o virtual-datahandler.o tdc-datahandler.o ip-datahandler.o mot-handler.o mot-object.o mot-dir.o journaline-datahandler.o crc_8_16.o log.o newssvcdec_impl.o Splitter.o dabdgdec_impl.o newsobject.o NML.o audio-base.o newconverter.o audiosink.o viterbi-handler.o fft-handler.o dab-params.o band-handler.o text-mapper.o dab_tables.o ensemble-printer.o iqdisplay.o virtual-input.o rawfiles.o raw-reader.o wavfiles.o wav-reader.o spectrum-viewer.o impulse-viewer.o tii-viewer.o rtlsdr-handler.o rtl-dongleselect.o rtl_tcp_client.o epgdec.o qrc_resources.o moc_radio.o moc_dab-processor.o moc_sample-reader.o moc_ofdm-decoder.o moc_phasereference.o moc_tii_table.o moc_fic-handler.o moc_fib-decoder.o moc_mp2processor.o moc_mp4processor.o moc_faad-decoder.o moc_data-processor.o moc_pad-handler.o moc_virtual-datahandler.o moc_tdc-datahandler.o moc_ip-datahandler.o moc_mot-object.o moc_audio-base.o moc_audiosink.o moc_iqdisplay.o moc_rawfiles.o moc_raw-reader.o moc_wavfiles.o moc_wav-reader.o moc_spectrum-viewer.o moc_impulse-viewer.o moc_tii-viewer.o moc_rtlsdr-handler.o moc_rtl-dongleselect.o moc_rtl_tcp_client.o   -lfftw3f -lusb-1.0 -ldl -lportaudio -lz -lsndfile -lsamplerate -lfaad /usr/lib/libqwt-qt4.so.5 -lQt5Widgets -lQt5Gui -lQt5Network -lQt5Xml -lQt5Core -lGLESv2 -lpthread 
iqdisplay.o: In function `IQDisplay::~IQDisplay()':
iqdisplay.cpp:(.text+0x30): undefined reference to `QwtPlotItem::detach()'
iqdisplay.o: In function `IQDisplay::DisplayIQ(std::complex<float>*, float)':
iqdisplay.cpp:(.text+0x1f8): undefined reference to `QwtPlotItem::detach()'
iqdisplay.cpp:(.text+0x210): undefined reference to `QwtPlotSpectrogram::setData(QwtRasterData*)'
iqdisplay.o: In function `IQDisplay::IQDisplay(QwtPlot*, short)':
iqdisplay.cpp:(.text+0x370): undefined reference to `QwtPlotItem::setRenderThreadCount(unsigned int)'
iqdisplay.cpp:(.text+0x3c4): undefined reference to `QwtPlotSpectrogram::setColorMap(QwtColorMap*)'
iqdisplay.cpp:(.text+0x49c): undefined reference to `QwtRasterData::setInterval(Qt::Axis, QwtInterval const&)'
iqdisplay.cpp:(.text+0x4c4): undefined reference to `QwtRasterData::setInterval(Qt::Axis, QwtInterval const&)'
iqdisplay.cpp:(.text+0x4f4): undefined reference to `QwtRasterData::setInterval(Qt::Axis, QwtInterval const&)'
iqdisplay.cpp:(.text+0x50c): undefined reference to `QwtPlotSpectrogram::setData(QwtRasterData*)'
iqdisplay.o:(.data.rel.ro._ZTV15SpectrogramData[_ZTV15SpectrogramData]+0x10): undefined reference to `QwtRasterData::setInterval(Qt::Axis, QwtInterval const&)'
iqdisplay.o:(.data.rel.ro._ZTV15SpectrogramData[_ZTV15SpectrogramData]+0x14): undefined reference to `QwtRasterData::pixelHint(QRectF const&) const'
iqdisplay.o:(.data.rel.ro._ZTV15SpectrogramData[_ZTV15SpectrogramData]+0x24): undefined reference to `QwtRasterData::contourLines(QRectF const&, QSize const&, QList<double> const&, QFlags<QwtRasterData::ConrecFlag>) const'
spectrum-viewer.o: In function `spectrumViewer::ViewSpectrum(double*, double*, double, int)':
spectrum-viewer.cpp:(.text+0x284): undefined reference to `QwtPlotCurve::setSamples(double const*, double const*, int)'
spectrum-viewer.o: In function `spectrumViewer::spectrumViewer(RadioInterface*, QSettings*, RingBuffer<std::complex<float> >*, RingBuffer<std::complex<float> >*)':
spectrum-viewer.cpp:(.text+0x1548): undefined reference to `QwtPlot::setCanvasBackground(QBrush const&)'
spectrum-viewer.cpp:(.text+0x15a8): undefined reference to `QwtPlotGrid::setMajorPen(QPen const&)'
spectrum-viewer.cpp:(.text+0x1618): undefined reference to `QwtPlotGrid::setMinorPen(QPen const&)'
spectrum-viewer.cpp:(.text+0x16dc): undefined reference to `QwtPlotSeriesItem::setOrientation(Qt::Orientation)'
spectrum-viewer.cpp:(.text+0x1758): undefined reference to `QwtPlotMarker::QwtPlotMarker(QString const&)'
impulse-viewer.o: In function `impulseViewer::impulseViewer(RadioInterface*, RingBuffer<float>*)':
impulse-viewer.cpp:(.text+0x620): undefined reference to `QwtPlot::setCanvasBackground(QBrush const&)'
impulse-viewer.cpp:(.text+0x680): undefined reference to `QwtPlotGrid::setMajorPen(QPen const&)'
impulse-viewer.cpp:(.text+0x6f0): undefined reference to `QwtPlotGrid::setMinorPen(QPen const&)'
impulse-viewer.cpp:(.text+0x7b8): undefined reference to `QwtPlotSeriesItem::setOrientation(Qt::Orientation)'
impulse-viewer.o: In function `impulseViewer::showImpulse(int)':
impulse-viewer.cpp:(.text+0xe38): undefined reference to `QwtPlotCurve::setSamples(double const*, double const*, int)'
tii-viewer.o: In function `tiiViewer::ViewSpectrum(double*, double*, double, int)':
tii-viewer.cpp:(.text+0x560): undefined reference to `QwtPlotCurve::setSamples(double const*, double const*, int)'
tii-viewer.o: In function `tiiViewer::tiiViewer(RadioInterface*, RingBuffer<std::complex<float> >*)':
tii-viewer.cpp:(.text+0x10ac): undefined reference to `QwtPlot::setCanvasBackground(QBrush const&)'
tii-viewer.cpp:(.text+0x110c): undefined reference to `QwtPlotGrid::setMajorPen(QPen const&)'
tii-viewer.cpp:(.text+0x117c): undefined reference to `QwtPlotGrid::setMinorPen(QPen const&)'
tii-viewer.cpp:(.text+0x1240): undefined reference to `QwtPlotSeriesItem::setOrientation(Qt::Orientation)'
tii-viewer.cpp:(.text+0x12b4): undefined reference to `QwtPlotMarker::QwtPlotMarker(QString const&)'
moc_iqdisplay.o:(.data.rel.ro+0x74): undefined reference to `QwtPlotItem::legendChanged()'
moc_iqdisplay.o:(.data.rel.ro+0x78): undefined reference to `QwtPlotSpectrogram::draw(QPainter*, QwtScaleMap const&, QwtScaleMap const&, QRectF const&) const'
moc_iqdisplay.o:(.data.rel.ro+0x7c): undefined reference to `QwtPlotRasterItem::boundingRect() const'
moc_iqdisplay.o:(.data.rel.ro+0x80): undefined reference to `QwtPlotItem::getCanvasMarginHint(QwtScaleMap const&, QwtScaleMap const&, QRectF const&, double&, double&, double&, double&) const'
moc_iqdisplay.o:(.data.rel.ro+0x88): undefined reference to `QwtPlotItem::updateLegend(QwtPlotItem const*, QList<QwtLegendData> const&)'
moc_iqdisplay.o:(.data.rel.ro+0x8c): undefined reference to `QwtPlotItem::legendData() const'
moc_iqdisplay.o:(.data.rel.ro+0x90): undefined reference to `QwtPlotItem::legendIcon(int, QSizeF const&) const'
moc_iqdisplay.o:(.data.rel.ro+0x94): undefined reference to `QwtPlotSpectrogram::pixelHint(QRectF const&) const'
moc_iqdisplay.o:(.data.rel.ro+0x98): undefined reference to `QwtPlotSpectrogram::interval(Qt::Axis) const'
moc_iqdisplay.o:(.data.rel.ro+0x9c): undefined reference to `QwtPlotSpectrogram::renderImage(QwtScaleMap const&, QwtScaleMap const&, QRectF const&, QSize const&) const'
moc_iqdisplay.o:(.data.rel.ro+0xa0): undefined reference to `QwtPlotRasterItem::imageMap(Qt::Orientation, QwtScaleMap const&, QRectF const&, QSize const&, double) const'
collect2: error: ld returned 1 exit status
Makefile:468: recipe for target 'linux-bin/qt-dab-2.5' failed
make: *** [linux-bin/qt-dab-2.5] Error 1

JvanKatwijk, 2019-04-07:
> QSettings*, RingBuffer<std::complex >*, RingBuffer<std::complex >*)':

Hi Simon

To me it seems that you the qwt library cannot be found. On my rpi 3 -
running stretch - I do not have the same problem

NomisGninnac, 2019-04-07:
Quite right.  I had it pointed at a .so2 file that was incorrect.  I copied the wrong path.

Will close now.

closed 2019-04-07 by NomisGninnac