JvanKatwijk/qt-dab #129: FindQWT.cmake file collision

ra1nb0w, 2019-04-19:
on macOS

```
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'cmake/Modules/FindQWT.cmake'
  'cmake/Modules/FindQwt.cmake'
```

ra1nb0w, 2019-04-19:
And just a side note:

```make
/Users/user/tmp/qt-dab/./includes/support/ringbuffer.h:181:2: warning: 'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
        PaUtil_WriteMemoryBarrier();
        ^
/Users/user/tmp/qt-dab/./includes/support/ringbuffer.h:87:40: note: expanded from macro 'PaUtil_WriteMemoryBarrier'
#   define PaUtil_WriteMemoryBarrier() OSMemoryBarrier()
                                       ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has been explicitly marked
      deprecated here
void    OSMemoryBarrier( void );
        ^
```

JvanKatwijk, 2019-04-19:
Thanks for the report
What is the spelling for the correct qwt lib on mac (I am not a mac owner)

wrt the ringbuffer code the code is derived from the portaudio library, do
you have a suggestion
for the correct naming of the memory barrier

j

ra1nb0w, 2019-04-20:
> Thanks for the report What is the spelling for the correct qwt lib on mac

from the project site seems Qwt

> wrt the ringbuffer code the code is derived from the portaudio library

maybe this can help https://github.com/google/leveldb/commit/594cc987af2e0af6417c4ac2b947ee8cdad59e5e
or see
https://github.com/harfbuzz/harfbuzz/issues/345

JvanKatwijk, 2019-04-21:
> maybe this can help google/leveldb@594cc98
> harfbuzz/harfbuzz#345 <https://github.com/harfbuzz/harfbuzz/issues/345>

Thanks for the link.
The FindQWT does not seem to be needed, so it is renamed
I changed the barrier for apple according to the link you sent, hope it
works now

JvanKatwijk, 2019-04-21:
I was too early
The FindQwt module is used when compiling on my Fedora box
The FindQWT modules is used when compiling on an Ubuntu box, as is done by
travis to generate an appImage

ra1nb0w, 2019-04-21:
```patch
diff --git a/CMakeLists.txt-qt4 b/CMakeLists.txt-qt4
index c4f7bc6..c4f2aa8 100755
--- a/CMakeLists.txt-qt4
+++ b/CMakeLists.txt-qt4
@@ -92,7 +92,7 @@ add_definitions (-DPRESET_NAME)
        set(CMAKE_INCLUDE_CURRENT_DIR ON)
        find_package(Qt4 REQUIRED COMPONENTS QtCore QtNetwork)
        include(${QT_USE_FILE})
-       find_package(QWT REQUIRED)
+       find_package(Qwt REQUIRED)
        if (NOT QWT_FOUND)
           message(FATAL_ERROR "please install qwt library")
        endif ()
```
and
```bash
git rm cmake/Modules/FindQWT.cmake
```
and should work.

Also patch
```c++
diff --git a/includes/dab-constants.h b/includes/dab-constants.h
index e7b5225..0a66f40 100755
--- a/includes/dab-constants.h
+++ b/includes/dab-constants.h
@@ -35,7 +35,7 @@
 #include       <cstring>
 #include       <unistd.h>
 
-#ifndef        __FREEBSD__
+#if !(defined __FREEBSD__) && !(defined __APPLE__)
 #include       <malloc.h>
 #endif
 
@@ -43,7 +43,7 @@
 //#include     "iostream.h"
 #include       "windows.h"
 #else
-#ifndef        __FREEBSD__
+#if !(defined __FREEBSD__) && !(defined __APPLE__)
 #include       "alloca.h"
 #endif
 #include       "dlfcn.h"
```

thank you

JvanKatwijk, 2019-04-21:
Including alloca and malloc dated back from the stone ages, so they are now commented out and in a next editing will be removed

ra1nb0w, 2019-04-24:
ok. please fix
```patch
diff --git a/includes/support/ringbuffer.h b/includes/support/ringbuffer.h
index 1c84dec..691b106 100644
--- a/includes/support/ringbuffer.h
+++ b/includes/support/ringbuffer.h
@@ -79,7 +79,7 @@
  */
 #ifdef __APPLE__
 #include <libkern/OSAtomic.h>
-#ifdef LEVELDB_ATOMIC_PRESENT
+#ifdef OSATOMIC_DEPRECATED
 #include <atomic>
 #endif
 #
@@ -88,11 +88,12 @@
        however, these barriers are superior to compiler-based ones. */
 
 inline void Apple_MemoryBarrier() {
-#if defined(LEVELDB_ATOMIC_PRESENT)
+#ifdef OSATOMIC_DEPRECATED
   std::atomic_thread_fence(std::memory_order_seq_cst);
 #else
   OSMemoryBarrier();
-#endif  // defined(LEVELDB_ATOMIC_PRESENT)
+#endif
+}
 #   define PaUtil_FullMemoryBarrier()  Apple_MemoryBarrier()
 #   define PaUtil_ReadMemoryBarrier()  Apple_MemoryBarrier()
 #   define PaUtil_WriteMemoryBarrier() Apple_MemoryBarrier()
```
`LEVELDB_ATOMIC_PRESENT` is not defined in any .h of XCode.

not it builds fine on macOS. thank you

ra1nb0w, 2019-04-24:
Sorry, but I don't see the patch that I sent.

JvanKatwijk, 2019-04-24:
which one?
as said, the one with the include files is implemented differently since
the includes were superfluous and remainders
from early versions

ra1nb0w, 2019-04-24:
https://github.com/JvanKatwijk/qt-dab/issues/129#issuecomment-486263879
about `ringbuffer.h`

JvanKatwijk, 2019-04-24:
Oops, I misread your earlier mail today. It should be OK now
thanks

ra1nb0w, 2019-04-24:
no. you lost `+}` after `#endif`

JvanKatwijk, 2019-04-24:
right again

ra1nb0w, 2019-04-24:
pay attention that you have added \` on `#ifdef OSATOMIC_DEPRECATED` line 82 of `includes/support/ringbuffer.h`

ra1nb0w, 2019-04-24:
and add `build/` to `.gitignore`

closed 2019-05-01 by JvanKatwijk

JvanKatwijk, 2020-01-04:
I do not have your address, so I'll do it this way. From time to time I get a question from someone who wants to compile/install qt-dab on the mac. Do you have a (kind of) advice for that?

ra1nb0w, 2020-01-05:
for dependencies you can use [macports](https://www.macports.org/)
then adding something like
```
INCLUDEPATH += /opt/local/include /opt/local/libexec/qt5/lib/qwt.framework/Versions/6/Headers
LIBS                  += -L/opt/local/lib
```
to qt-dab.pro and enable/disable the features you need.

I also changed
```
diff --git a/includes/backend/mm_malloc.h b/includes/backend/mm_malloc.h
index 0790a7e..dfc6fb2 100755
--- a/includes/backend/mm_malloc.h
+++ b/includes/backend/mm_malloc.h
@@ -40,7 +40,7 @@ OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 #if defined(_WIN32) || defined(_WIN64)
 #define MALLOC(a) _mm_malloc(a, 16)
 #else
-#include <malloc.h>
+#include <malloc/malloc.h>
 #define MALLOC(a) memalign(16, a)
 #endif
```