JvanKatwijk/qt-dab #130: Compilation on RPi 3B+ successful, but no signal is received

apaj, 2019-05-02:
Hello Jan,

thanks so much for the wonderful software. Everything went quite simply on the new Raspberry Pi, under Ubuntu Mate, concerning the installation. I attache the `qt-dab.pro` (it's not really a zip file, i just renamed it so github allows it to attach) so that you are familiar with my configuration.
However, once I connect the dabstick (820T2) and run ./qt-dab-2.5 - nothing happens. 
I attach the log file (also not a zip, just a plain text file) of a run and a screenshot of the widgets for easier debugging. 
I tried with the ARM appimage, but to no avail, unfortunately.
Would you, please, help me out here?
I am in Vienna, Austria if that means anything in the context of available bands.
Regards,
Aleksandar
[image]
[log.zip](https://github.com/JvanKatwijk/qt-dab/files/3138538/log.zip)
[qt-dab.pro.zip](https://github.com/JvanKatwijk/qt-dab/files/3138543/qt-dab.pro.zip)

NomisGninnac, 2019-05-03:
Did you make the rules.d file? if not just install gqrx, that program does it for you.

andimik, 2019-05-03:
In Vienna the channels are 11C (Citymux), 12C from Bratislava and 5D (Bundesweiter Mux).

On channel 13 you won't receive anything in Vienna.

apaj, 2019-05-03:
Thank you NomisGninnac and andimik for your responses.

@NomisGninnac , could you please specify which exactly file do you mean? 
As you advised, I tried installing `gqrx`, but then `apt` redirected me to a different package, called `gqrx-sdr`, so I installed that:

```
 sudo apt-get install gqrx-sdr 
```
I wasn't sure which `rules.d` did you mean, so I did a search with the following results:

```
apaj@ubuntu-pib3p:~$ sudo find / -iname "rules.d"
find: ‘/run/user/1000/gvfs’: Permission denied
/usr/share/polkit-1/rules.d
/etc/udev/rules.d
/lib/udev/rules.d
/snap/core/6534/etc/udev/rules.d
/snap/core/6534/lib/udev/rules.d
/snap/core/6816/etc/udev/rules.d
/snap/core/6816/lib/udev/rules.d
```
however, all of these are directories... Can you please provide more details, what exactly did you mean?

@andimik , yesterday I did try some random channels, but I didn't catch anything - don't know which I randomly clicked... Now that you gave me a list, I tried 11C, and voila! I get signal there! I am not sure if this has to do with the installation of the package I mentioned above... Anyway, there is nothing in 12C and 5D, still.

Here is a screenshot of a working system.

Even though we have it all running now, I would wait a bit before closing this issue, as I am very eager to hear more about the `rules.d` file from @NomisGninnac - please, I would appreciate more information greatly. Thanks!

Regards,
Aleksandar

[image]

JvanKatwijk, 2019-05-03:
Hello
As Animik states, on channel 13 there is no DAB signal. Looking at the
image and the log it seems that the software is working
properly, however, there is no signal.

I assume you tried all the channels
If so, there is the issue of an antenna.

JvanKatwijk, 2019-05-03:
OK
our mails crossed, I see that you receive something now.
The rules.d file ensure that you have - as a non-root user- access through
USB. It seems to be working

apaj, 2019-05-03:
Hello Jan, thanks so much for joining in.

Right, but where is the file supposed to be?

JvanKatwijk, 2019-05-03:
/etc/udev/rules/.d is the location for the rules files, the file is
probably called rtl-sdr.rules
When your system starts the ".rules" files in this directory are read and
the system is set accordingly

For most kind of devices such rules files can be found on the net.

apaj, 2019-05-03:
Thank you very much, @JvanKatwijk , @andimik and @NomisGninnac !

closed 2019-05-03 by apaj

andimik, 2019-05-03:
There are currently tests on 5D, official start is 28 May.

12C from Slovakia is quite strong in parts of Vienna, seems your place (or your equipment) is not the best.

If you want, there is a German speaking forum at https://www.rundfunkforum.de/index.php where several guys from Vienna are actively posting there.