JvanKatwijk/qt-dab #132: Dependency missing in Raspbian Buster

happysat, 2019-07-27:
Hello Jan,

Executing QT-Dab Appimage on Raspbian Buster gives the following missing Dep.:
./Qt_DAB-ARM.AppImage
./Qt_DAB-ARM.AppImage: error while loading shared libraries: libicui18n.so.57: cannot open shared object file: No such file or directory

I did install sudo apt-get install libicu-dev but same error, i think those are at v63.
Which Dependency do i need.?

Regards Alex

JvanKatwijk, 2019-07-27:
Thanks for the report. Right now I am not at home.
I am running (and building) the appImage on stretch, without problems.
\When I am back I'll download a Buster version

It is not completely clear to me what the library in question is used for,
but I'll come back to you on this issue tuesday or later
best
jan

JvanKatwijk, 2019-07-30:
Well I know what the problem is, but it'll take some time before being able
to solve it.
The problem is that - since the tool to restructure dependencies in a set
of dependent libraries
is not working correct on the RPI - I have to make some manual changes.
On Stretch this works since my changes cause the dependencies to be external
to the set libraries forming the appImage and the affected libraries are
just available on
a Stretch system.
They are - however - not available on the Buster version, so the "solution"
is that I have to
build a version on and for Buster

Sorry,
best
jan

happysat, 2019-08-01:
Thanks Jan for explanation, we wait :)

Kind regards Alex

JvanKatwijk, 2019-08-02:
Creating an AppImage that will work both on Stretch and Buster seems a long way off. What I have done now is replace the appimage for the rpi in the releases section with a script that will just
download everything needed and create a working qt-dab-2.7.1 version.  See the releases section

closed 2020-09-27 by JvanKatwijk