JvanKatwijk/qt-dab #133: compilation issue

H2G2LB, 2019-09-06:
First many thanks for this nice program.
Using a brand new installation of Ubuntu 19.04 on a Beelink BT3pro 
I compiled your program.
To my surprise it worked.
I also tried the appimage version.
The appimage version is OK but doesn't display *.jpg data it is stored OK in the /temp directory
but the display windows is very small and empty.
Now for my compiled version:
Every thing is OK except for every windows being twice too big
I'm running on a 1920x1080 display and with all useful windows open from your program it is not enough.
Is there any simple solution to this problem ? 
I'm using an RSP1 device and various RTL2832 ones .
The main unique advantage for me of your program is the constellation display.
I'm 70Km from Paris and a recent SFN transmitter on Channel 6A started deteriorating a lot 
my reception of this channel.
Your software is the only one that spotted it with the constellation looking as an X.
Thanks again for a very nice software.

JvanKatwijk, 2019-09-07:
Right now I am not at home for just one more week. I'll have a look when I
am back
If I understand you correctly, the scaling of the widgets is incorrect,
i.e. twice as large as it should be. One thing you could try is to comment
out line 110 in main.cpp, that relates to a setting to handle widgets
calling by qt and see what happens.
The line is within an ifdef checking the version number of qt.
Best
Jan

H2G2LB, 2019-09-07:
Thanks for the quick answer 
I'll try this ASAP and keep you informed.

H2G2LB, 2019-09-07:
It worked perfectly many thanks.
In fact your script refused to work on my frequently updated Ubuntu 19.04 because of a missing 
portaudio.o file.
So I ran it on another machine with a LTS 18.04 version and imported the qt_dab_2.7.1 file. 
Thanks again.

JvanKatwijk, 2019-09-07:
Good to hear.
I plan to buy another laptop when I am back, one with a higher screen
resolution than my old one (13.. to ???). There are more reported issues
with widgetsizes on high(er) resolution screens

Best
Jan

andimik, 2020-09-27:
As this is solved you can close this issue

closed 2020-09-27 by JvanKatwijk