JvanKatwijk/qt-dab #138: Failed to map API.DLL functi!!!

bg1pc, 2019-09-08:
Failed to map API.DLL functi!!!

JvanKatwijk, 2019-09-08:
Can you be slightly more specific? I understand that you are running a
windows version, but can you specify what you have done

bg1pc, 2019-09-08:
win7

JvanKatwijk, 2019-09-08:
Thanks, but that is not enough I am afraid. If you want me to be able to
address an issue then you should tell me the steps you have taken such that
I can try to reconstruct what happens and address the issue

bg1pc, 2019-09-08:
w7下装好后，不能起动，显示无法映射API.DLL函数！
After W7 is installed, it can't start, display... API. DLL function cannot be mapped!

bg1pc, 2019-09-08:
After downloading 4 files, install them in turn

JvanKatwijk, 2019-09-08:
How does this help me understanding your problem?

bg1pc, 2019-09-08:
[image]

bg1pc, 2019-09-08:
[image]
[image]

bg1pc, 2019-09-08:
ok???

JvanKatwijk, 2019-09-08:
Please send the picture to my mailaddress.uploading pictures does not work

JvanKatwijk, 2019-09-16:
I cannot read the pictures that you sent. Please try to send a picture using my regular email address.

closed 2020-01-01 by JvanKatwijk