JvanKatwijk/qt-dab #139: Not working on Ubuntu 18.04

frspin, 2019-10-23:
Latest version is not working on my Ubuntu 18.04/64 bit

Compile is without errors but at startup I get:

spin@franco:/usr/local/src/Ham/sdr-j/qt-dab$ linux-bin/qt-dab-2.7.3
realloc(): invalid pointer
Annullato (core dump creato)

Same for previous version on my PC, updated wit "git pull", and with a new "git clone" command.

Regards

Franco Spinelli

JvanKatwijk, 2019-10-23:
OK,
it works fine on my fedora box. Tomorrow I'll test with Ubuntu
Did you compile using qmake or cmake
best
jan

JvanKatwijk, 2019-10-23:
ps
does the appImage run?
j

andimik, 2019-10-23:
I have Ubuntu 18.04 bionic 64bit as well and have no problems with latest version.

Have you tried to delete `~/.qt-dab.ini`?

frspin, 2019-10-24:
Il 23/10/19 21:28, JvanKatwijk ha scritto:
> OK,
> it works fine on my fedora box. Tomorrow I'll test with Ubuntu
> Did you compile using qmake or cmake

qmake

Regards

Franco Spinelli

frspin, 2019-10-24:
Il 23/10/19 22:07, JvanKatwijk ha scritto:
> ps
> does the appImage run?

I will try this this evening

Regards

Franco Spinelli

frspin, 2019-10-24:
Il 23/10/19 22:22, andimik ha scritto:
> I have Ubuntu 18.04 bionic 64bit as well and have no problems with 
> latest version.
> 
> Have you tried to delete |~/.qt-dab.ini|?

No. I will try this evening

Regards

Franco Spinelli

JvanKatwijk, 2019-10-24:
I installed a virtual machine with Ubuntu 18.04 and no issue arises.
If you still have a crash please do the following
a. make clean
b. edit in qt-dab.pro the lines
QMAKE_CFLAGS    +=  -O3 -ffast-math
QMAKE_CXXFLAGS  +=  -O3 -ffast-math
#QMAKE_CFLAGS   +=  -g
#QMAKE_CXXFLAGS +=  -g
#QMAKE_LFLAGS   +=  -g

into
#QMAKE_CFLAGS   +=  -O3 -ffast-math-bin/
#QMAKE_CXXFLAGS +=  -O3 -ffast-math
QMAKE_CFLAGS    +=  -g
QMAKE_CXXFLAGS  +=  -g
QMAKE_LFLAGS    +=  -g

3. qmake/make an executable
4. run it with gdb
   gdb ./linux-bin/qt-dab-2.7.3

and run

if a crash occurs, give the command bt

and send me the output, then we might see where the error manifested itself

best
jan

frspin, 2019-10-24:
Il 24/10/19 15:21, JvanKatwijk ha scritto:
> I installed a virtual machine with Ubuntu 18.04 and no issue arises.

Appimage is correctly working
Removing ~/.qt-dab.ini don't change anything

> If you still have a crash please do the following
> a. make clean
> b. edit in qt-dab.pro the lines
> QMAKE_CFLAGS += -O3 -ffast-math
> QMAKE_CXXFLAGS += -O3 -ffast-math
> #QMAKE_CFLAGS += -g
> #QMAKE_CXXFLAGS += -g
> #QMAKE_LFLAGS += -g

I gon't have
QMAKE_CFLAGS += -O3 -ffast-math
QMAKE_CXXFLAGS += -O3 -ffast-math
but
QMAKE_CXXFLAGS	+= -std=c++11
QMAKE_CFLAGS	+=  -flto -ffast-math
MAKE_CXXFLAGS	+=  -flto -ffast-math

> 
> into
> #QMAKE_CFLAGS += -O3 -ffast-math-bin/
> #QMAKE_CXXFLAGS += -O3 -ffast-math
> QMAKE_CFLAGS += -g
> QMAKE_CXXFLAGS += -g
> QMAKE_LFLAGS += -g

Commented/uncommented as suggested

> 
> 3. qmake/make an executable
> 4. run it with gdb
> gdb ./linux-bin/qt-dab-2.7.3
> 
> and run
> 
> if a crash occurs, give the command bt
> 
> and send me the output, then we might see where the error manifested itself
> 

This is the output:

Starting program: /usr/local/src/Ham/sdr-j/qt-dab/linux-bin/qt-dab-2.7.3
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
realloc(): invalid pointer

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: File o directory non esistente.
(gdb) bt
#0  0x00007ffff4396e97 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff4398801 in __GI_abort () at abort.c:79
#2  0x00007ffff43e1897 in __libc_message (action=action@entry=do_abort,
fmt=fmt@entry=0x7ffff450eb9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff43e890a in malloc_printerr (str=str@entry=0x7ffff450cefd
"realloc(): invalid pointer") at malloc.c:5350
#4  0x00007ffff43f0eba in __GI___libc_realloc (oldmem=0x5555558433a0
<QListData::shared_null>, bytes=32) at malloc.c:3174
#5  0x00007ffff139d2d1 in QListData::realloc_grow(int) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff139d36f in QListData::append(int) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff146b358 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff1466b0d in qRegisterResourceData(int, unsigned char
const*, unsigned char const*, unsigned char const*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff131c693 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff7de5733 in call_init (env=0x7fffffffdf38,
argv=0x7fffffffdf28, argc=1, l=<optimized out>) at dl-init.c:72
#11 0x00007ffff7de5733 in _dl_init (main_map=0x7ffff7ffe170, argc=1,
argv=0x7fffffffdf28, env=0x7fffffffdf38) at dl-init.c:119
#12 0x00007ffff7dd60ca in _dl_start_user () at /lib64/ld-linux-x86-64.so.2
---Type <return> to continue, or q <return> to quit---
#13 0x0000000000000001 in  ()
#14 0x00007fffffffe27b in  ()
#15 0x0000000000000000 in  ()

Regards

Franco Spinelli

JvanKatwijk, 2019-10-24:
I seems to be a library issue. As far as I can see, the error occurs in the
initialisation phase.
Can you do a make clean followed by a make and send me the full output of
that? I can compare it to
the output of the make as done on my VM.
To make that a good comparison, I assume you updated the 18.04 version.
Did you use the "script" to load all required libraries?

Btw, the appImage was made in Ubuntu 16
best
jan

frspin, 2019-10-24:
Il 24/10/19 19:10, JvanKatwijk ha scritto:
> I seems to be a library issue. As far as I can see, the error occurs in the
> initialisation phase.
> Can you do a make clean followed by a make and send me the full output of
> that? I can compare it to
> the output of the make as done on my VM.

Attached the file obtained with
make > output.txt 2>&1

> To make that a good comparison, I assume you updated the 18.04 version.

Updated to latest update

> Did you use the "script" to load all required libraries?

No
Previous version of qt-dab was working without problems. Last working
version was the first with 2.7.3 number from Github.

Regards

Franco Spinelli

/usr/lib/x86_64-linux-gnu/qt4/bin/uic forms/technical_data.ui -o ui_technical_data.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic forms/dabradio.ui -o ui_dabradio.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic forms/audio-description.ui -o ui_audio-description.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic forms/data-description.ui -o ui_data-description.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic spectrum-viewer/scopewidget.ui -o ui_scopewidget.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic impulse-viewer/impulse-widget.ui -o ui_impulse-widget.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic tii-viewer/tii-widget.ui -o ui_tii-widget.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic devices/rtlsdr-handler/rtlsdr-widget.ui -o ui_rtlsdr-widget.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic devices/sdrplay-handler/sdrplay-widget.ui -o ui_sdrplay-widget.h
/usr/lib/x86_64-linux-gnu/qt4/bin/uic devices/rtl_tcp/rtl_tcp-widget.ui -o ui_rtl_tcp-widget.h
g++ -c -m64 -pipe -std=c++11 -g -isystem /usr/include/qt4 -O2 -Wall -W -D_REENTRANT -DGITHASH=\"1ce8b6c\" -DPRESET_NAME -D__THREADED_BACKEND -D__HIGH_PRECISION__ -DHAVE_RTLSDR -DHAVE_SDRPLAY -DHAVE_RTL_TCP -D_SEND_DATAGRAM_ -DTRY_EPG -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -I. -I. -Isrc -Iincludes -Iservice-description -Iincludes/protection -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/mot -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/support -Iincludes/support/viterbi-jan -Idevices -Idevices/rawfiles-new -Idevices/wavfiles-new -Iincludes/scopes-qwt6 -Ispectrum-viewer -Iimpulse-viewer -Itii-viewer -I../../../../include -I../../../../include -I/usr/include/qwt -Idevices/rtlsdr-handler -Idevices/sdrplay-handler -Idevices/rtl_tcp -Iincludes/backend/data/epg -I. -I. -o main.o main.cpp
g++ -c -m64 -pipe -std=c++11 -g -isystem /usr/include/qt4 -O2 -Wall -W -D_REENTRANT -DGITHASH=\"1ce8b6c\" -DPRESET_NAME -D__THREADED_BACKEND -D__HIGH_PRECISION__ -DHAVE_RTLSDR -DHAVE_SDRPLAY -DHAVE_RTL_TCP -D_SEND_DATAGRAM_ -DTRY_EPG -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -I. -I. -Isrc -Iincludes -Iservice-description -Iincludes/protection -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/mot -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/support -Iincludes/support/viterbi-jan -Idevices -Idevices/rawfiles-new -Idevices/wavfiles-new -Iincludes/scopes-qwt6 -Ispectrum-viewer -Iimpulse-viewer -Itii-viewer -I../../../../include -I../../../../include -I/usr/include/qwt -Idevices/rtlsdr-handler -Idevices/sdrplay-handler -Idevices/rtl_tcp -Iincludes/backend/data/epg -I. -I. -o radio.o radio.cpp
radio.cpp: In member function ‘void RadioInterface::select_presetService(QString, QString)’:
radio.cpp:1892:6: warning: unused variable ‘res’ [-Wunused-variable]
 bool res;
      ^~~
g++ -c -m64 -pipe -std=c++11 -g -isystem /usr/include/qt4 -O2 -Wall -W -D_REENTRANT -DGITHASH=\"1ce8b6c\" -DPRESET_NAME -D__THREADED_BACKEND -D__HIGH_PRECISION__ -DHAVE_RTLSDR -DHAVE_SDRPLAY -DHAVE_RTL_TCP -D_SEND_DATAGRAM_ -DTRY_EPG -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtXml -I/usr/include/qt4 -I. -I. -Isrc -Iincludes -Iservice-description -Iincludes/protection -Iincludes/ofdm -Iincludes/backend -Iincludes/backend/audio -Iincludes/backend/data -Iincludes/backend/data/mot -Iincludes/backend/data/journaline -Iincludes/output -Iincludes/support -Iincludes/support/viterbi-jan -Idevices -Idevices/rawfiles-new -Idevices/wavfiles-new -Iincludes/scopes-qwt6 -Ispectrum-viewer -Iimpulse-viewer -Itii-viewer -I../../../../include -I../../../../include -I/usr/include/qwt -Idevices/rtlsdr-handler -Idevices/sdrplay-handler -Idevices/rtl_tcp -Iincludes/backend/data/epg -I. -I. -o dab-processor.o dab-processor.cpp
dab-processor.cpp: In member function ‘virtual void dabProcessor::run()’:
dab-processor.cpp:146:8: warning: variable ‘avgValue_nullPeriod’ set but not used [-Wunused-but-set-variable]
 float  avgValue_nullPeriod = 0;
        ^~~~~~~~~~~~~~~~~~~
dab-processor.cpp:147:8: warning: variable ‘avgValue_testPeriod’ set but not used [-Wunused-but-set-variable]
 float  avgValue_testPeriod = 0;
        ^~~~~~~~~~~~~~~~~~~
[... cut ...] r -Idevices/rtl_tcp -Iincludes/backend/data/epg -I. -I. -o newssvcdec_impl.o src/backend/data/journaline/newssvcdec_impl.cpp
g++ -c -m64 -pipe -std=c++11 -g -isystem /usr/include/qt4 -O2 -Wall -W -D_REENTRANT -DGITHASH=\"1ce8b6c\" -DPRESET_NAME -D__THREADED_BACKEND -D__HIGH_PRECISION__ -DHAVE_RTLSDR -DHAVE_SDRPLAY -DHAVE_RTL_TCP -D_SEND_DATAGRAM_ -DTRY_EPG -DQT_NO_DEBUG -DQT_XML_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/q

JvanKatwijk, 2019-10-24:
are you compiling against Qt4 or Qt5?
It seems you are compiling with Qt4, at least the processing of thewidgets
is done by a Qt4 tool

The software is - since a few years - assuming Qt5, that might be an issue
best
jan

Op do 24 okt. 2019 om 20:14 schreef frspin <notifications@github.com>:

> Il 24/10/19 19:10, JvanKatwijk ha scritto:
> the
>
> Attached the file obtained with
> make > output.txt 2>&1
>
>
> Updated to latest update
>
>
> No
> Previous version of qt-dab was working without problems. Last [...]

frspin, 2019-10-24:
Il 24/10/19 20:31, JvanKatwijk ha scritto:
> are you compiling against Qt4 or Qt5?
> It seems you are compiling with Qt4, at least the processing of thewidgets
> is done by a Qt4 tool
> 
> The software is - since a few years - assuming Qt5, that might be an issue
> best

I was using qmake command, normally compiling for Qt4
Previous version was working. I have a 2.7.1 version on my disk, working
without problems.

Now I have used explicity qmake for Qt5 and program is working

So problem solved

Regards and thanks

Franco Spinelli

JvanKatwijk, 2019-10-24:
Ok good to hear
best
jan

closed 2020-01-01 by JvanKatwijk