JvanKatwijk/qt-dab #141: EPG: EIB files are not been saved

andimik, 2019-11-02:
I just found out that when observing the content of an EPG data stream the console output message says (in my case)

> going to write file /tmp/qt-pictures/eE3C201.EIB

but in the **tmp**-folder there are just the small icons. **Not the mentioned EIB file**, which I even cannot find somewhere below the root `/`

[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Remark: **EHB files** from the EPG are saved, indeed.

JvanKatwijk, 2019-11-02:
Tonight I'll have a look at it

JvanKatwijk, 2019-11-02:
Interesting. The code flow is pretty simple and the epg handler writes the
file after converting the input to xml format

can you sen me the input file?

best
jan

JvanKatwijk, 2019-11-06:
Andreas

Ik found an old file with DAB data, containing an EPG service. The EPG data
contains both EHB and EIB files. The
decoding software I am using skips the EIB files as being "not epg". I
cannot find any data on what the file format
in EIB files is. The EHB data is stored as xml data in files, and the name
of the file contains the serviceId of
the service it is describing.

As short term solution I changes the code such that EIB files are skipped
(and not mentioned anymore)

best
jan

closed 2020-04-18 by JvanKatwijk