JvanKatwijk/qt-dab #142: No syncing from version 3 - RPi3

NomisGninnac, 2019-11-11:
Previous 2.7.1 version keeps sync

latest editions don't sync, sync light flashes green then red then green repeatedly

Sounds terrible

JvanKatwijk, 2019-11-11:
what is your input device

NomisGninnac, 2019-11-11:
Input is Dabstick

JvanKatwijk, 2019-11-11:
The sync handling - in the continuous build - is now precisely the one in
2.7.1
Interesting is that performance for 12 bit data (RSP and airspy) is worse -
more false positives -
so I made the parameter "threshold" official. I do not have a working
dabstick so I cannot compare the behaviour of
2.7.1 and 3.0. The default settingfor "threshold" (to be set in the ini
file) is 3, which makes the
parameters used equal to those in 2.7.1., I use different values

NomisGninnac, 2019-11-11:
Have compiled from latest commit,  the problem remains the same.

Am struggling to find the 2.7.1 commit on github as I suspect it will be some time before the problem is healed.

andimik, 2019-11-11:
You could use the 2.7 release. Or you could test which commit was working by using `git reset` command.

But let me test this with rawfiles, I fear I had the same experience.

JvanKatwijk, 2019-11-14:
The latest version was tested by me with a - borrowed - dabstick as well as more than a dozen files containing "raw" images and I do not see a problem. Two parameters are important
threshold_1 sets the treshold for a first test, if passed the data will be interpreted for FIC data
threshold_2 sets the threshold for continuous use its value is higher, since in general the correlation value when the first datablock is "in place" is much higher than when the offset of the data to be computed by the correlation is 40 to 50 samples off.
Default values now are 2 and 5, for me that works fine

NomisGninnac, 2019-11-21:
Still having chronic sync problems as of latest.  Audio is terrible, squeeling at times.   2.7.1 works fine.

JvanKatwijk, 2019-11-22:
I cannot create a situation where the phenomenon as described happens with as configration an RPI 3 and a noxon stick. The interesting thing is that the difference in synchronization code between 2.7 and current is that 2 constants that appeared in 2.7 are now replaced by parameters.
Experimenting with these parameters did not give a clue. The best thing is to dump a segment of the input and send it to me, then I can have a look

NomisGninnac, 2019-11-23:
How do I dump a segment of the input?  What is your email so I can send it?   Many thanks.

andimik, 2019-11-23:
Use a file hoster and insert the link here.

NomisGninnac, 2019-11-23:
https://gofile.io/?c=29DlIg

JvanKatwijk, 2019-11-23:
Interesting phenomenon.
It turns out that the synchronization as such was fine, in both 2.7 and the
current version the synchronisation
was lost on the same spots in the data. However, I adapted the code for
handling a loss in synchronization during processing
for some experiments (counting etc). For the time being I changed that back
to the way it was handled in 2.7.
Your file now shows - apart from some sissing in the beginning - one
"sissing" moment.

Your signal is extremely weak,  the SNR indication related two very small
numbers. I adapted the computation and the
SNR measured from the file data is now 1dB,
Signal quality seems pretty well though as shown in the constellation
diagram

NomisGninnac, 2019-11-26:
Latest compile is even worse, cannot keep sync at all , again i'm falling back to 2.7.1

Dump:
https://gofile.io/?c=qL6HvG

JvanKatwijk, 2019-11-26:
Interesting
With 2.7 and 2.7.1 the behaviour of the most recent dump is the same  as
with 3.1: no synchronization whatsoever, while
with 2.7 and 2.7.1 and 3.1 there is a reasonable amount of synchronization
with the previous dump.
Looking at the spectrum the signal is just too weak

NomisGninnac, 2019-11-26:
How does 2.7.1 deal with it all so well?   I am unable to get a stronger signal.  

Will stick with 2.7.1.

JvanKatwijk, 2019-11-27:
The sychronization code for 3.x and 2.7.x is the same, the code is the
same, and the two samples you sent show exactly
the same behaviour on both 2.7.1 and 3.1 (i.e. same moments in gaining and
loosing control)

NomisGninnac, 2019-11-27:
yet there is no audio disturbance in 2.7.1  whereas version 3.x can't handle weak signals at all.

very strange.   I think it needs study......

NomisGninnac, 2019-12-14:
Apart from a few issues with BBC channels, this issue has been resolved as of current commit (14 December 2019)

closed 2019-12-14 by NomisGninnac