JvanKatwijk/qt-dab #146: Stereo stations reported as mono on 3.x releases

fustinoni-net, 2019-12-07:
Hi,
first of all thank you for sharing this great software that I use appreciate.

I just want to report that some Italian station, reported as stereo in the 2.7.1 release and in other receiver are reported to be mono ( no green indicator) in versions 3.0 and 3.11 running on a raspberry pi 4. Let me know if you need a raw dump.

Again, thank you for your great work.
Have a nice day
Enrico

andimik, 2019-12-07:
Yes, confirmed.

fustinoni-net, 2019-12-07:
You can download the file at:

[http://fustinoni.net/12c.sdr](http://fustinoni.net/12c.sdr)

it's almost 342MB

JvanKatwijk, 2019-12-07:
Thanks
Just looking at the issue, thanks for the file
You are absolutely right that there is a difference. However, the 2.X
series were wrong and added the stereo sign even when it should not be
Actually for DAB+ streams it always added the stereo sign as long as no
error occurred

best
jan

fustinoni-net, 2019-12-07:
On a Sony portable radio and in a car receiver the station are reported to be stereo too.

I'm referring to RDS RELAX station. Forgot to mention it before

JvanKatwijk, 2019-12-07:
Portable radio and car receiver are DAB I assume. Tomorrow I'll have a
closer look

fustinoni-net, 2019-12-07:
Yes, DAB :-)

JvanKatwijk, 2019-12-08:
Interesting/
The configuration parameters in the stream miss any information (all bits
ero), but the audio output has two channels with different
content. So I will change the "stereo" setting in the next build.

best jan

fustinoni-net, 2019-12-08:
Thank you!

Have a nice day
Enrico

andimik, 2019-12-08:
Jan, do we really need that stereo icon?

JvanKatwijk, 2019-12-08:
No of course not, it is one of those things that survived all changes since
long time.
Triggered by the question yesterday I looked into the code and found out
that for DAB+ basically every service will
have the "stereo" on, so the added value is 0.

mxlbzn, 2019-12-16:
Even 3.11-Beta is still incorrectly reporting mono for some stereo streams. It seems to be the case for HE-AAC v2 with parametric stereo.

JvanKatwijk, 2019-12-16:
In the continuous build the stereo indicator only depends on whether or not the output was in 2 channels

mxlbzn, 2019-12-17:
OK, it seems to be fixed in continuous build (bug was in 3.11-beta), as far as I can verify with one accessible multiplex, with HE-AAC v1 and v2 streams, though (all stereo).

JvanKatwijk, 2019-12-17:
The indicator is basically a left over, I am considering removing it in the
next release and using that space for the
FIC indicator, since that indicator is "ensemble/channel" related and not
just "service related" as the rest of the
technical data widget

closed 2020-01-01 by JvanKatwijk