JvanKatwijk/qt-dab #147: Local time confusing format

mxlbzn, 2019-12-17:
Local time is displayed with confusing format, e.g. **2019-dec-17:20:45**, which corresponds to ISO8601-like **2019-12-17 20:45**, which would be much clearer.

mxlbzn, 2019-12-17:
(someone including me would prefer displaying original UTC time)

JvanKatwijk, 2019-12-17:
why

mxlbzn, 2019-12-17:
Never mind, the issue is primarily the confusing format looking like frozen ...HH:MM:SS datetime.

JvanKatwijk, 2019-12-18:
The most recent continuous build has an adapted time format. Note that it
is the time as recorded in the transmission,
so if you play a file, you get the time of the recording rather than
current time.

andimik, 2019-12-18:
I love this feature, as you know when a file was recorded

mxlbzn, 2019-12-18:
OK, just removing the first colon is enough. I understand that replacing also the month name to number would be confusing for those living in places where date with confusing order is used.

And I thought that the local time is calculated using computer locale time zone, didn't know about LTO (local time offset) info present in the stream. That's good.

JvanKatwijk, 2019-12-19:
Well I just had a quick look at the ISO standard and that states that
between date and time a letter T is set. I do not think that
improves readability, so I'll leave it as it is now

sjlongland, 2019-12-25:
Actually… related to this, it appears the "date" part sometimes lags behind.  Noticed this yesterday, and I see today it's doing it again.

[image]

Note that up the top-right corner, it shows the *correct* date, while `qt-dab` is showing yesterday's date.

sjlongland, 2019-12-26:
Yep, without touching the application… after 10AM it ticked over to the 26th… I think somewhere it is using UTC for the date, but local time for the time.

When your local time-zone is UTC+10, it displays the wrong date between midnight and 10AM.

andimik, 2019-12-26:
Thank you for the investigation!

JvanKatwijk, 2019-12-26:
Interesting

It seems that the LTO is added by the hours field in the time (just modulo
24), but if the raw "hours" is not within the
range 0 - 23 no adjustment of the day takes place.
(Awful since changing the day may lead to overflow/underflow such that the
month field has to be adapted, and of course
when that may lead to changing the year)

I' see to it that it will be corrected

sjlongland, 2019-12-26:
So is this local time actually showing the local system time, or a timestamp broadcast in the radio data stream?

JvanKatwijk, 2019-12-27:
It is the timestamp in the data stream. So if you play from a recording you
see the time of the recording

mxlbzn, 2019-12-28:
The newer build has changed to show UTC time instead of local time, without changing the label/tooltip.. ?

closed 2020-01-01 by JvanKatwijk