JvanKatwijk/qt-dab #149: Spectrum in SDR files broken

andimik, 2019-12-31:
I don't know since when this happens and if Airspy or HackRF or RTLSDR behavior different, but saving into an *.sdr file means, that in the played sdr file the spectrum on the left is nearly empty.

Using latest qt-dab built from source under Ubuntu.

Live reception:
[image]

SDR-file from the same live source (taken just some seconds before)
[image]

The funny thing is, when I convert this recorded sdr into a raw file by

```
sox -e signed-integer -r 2048e3 -B -t raw -c 1 -b 16 foo.sdr -e unsigned-integer -r 2048e3 -t raw -b 8 foo.raw
```

then it's ok.

[image]

But I found some sdr files on my laptop where the spectrum is ok, indeed. I assume they were taken under Windows with an older version.

JvanKatwijk, 2019-12-31:
Hi Andreas

AIRSpy and SDRplay donot show that behaviour. Looks strange. Are you saving
the file on an external usb device or onthe internal drive?

andimik, 2020-01-02:
It's the local drive (~ or /tmp).

An Airspy also shows different windows, please have a look at the y-axis ...

Live reception:

[image]

SDR-file taken some seconds before:
[image]

Can you check if the saved (and reloaded) samples have the correct format? Because when converting it I get this error message:

> sox -e signed-integer -r 2048e3 ***-B*** -t raw -c 1 -b 16 foo.sdr -e unsigned-integer -r 2048e3 -t raw -b 8 foo.raw

> sox WARN dither: dither clipped 240045 samples; decrease volume?

Afterwards I cannot play the raw-file:
[image]

It only works with

> sox -e signed-integer -r 2048e3 ***-L*** -t raw -c 1 -b 16 foo.sdr -e unsigned-integer -r 2048e3 -t raw -b 8 foo.raw

So I found out that the saved sdr-files have two different formats: ***Airspy (and maybe HackRF) uses s16le, whereas an rtlsdr-stick saves it in s16be format.***

What format is used for loading the pre-recorded *.sdr files?

That would explain the issue. Could you check?

JvanKatwijk, 2020-01-02:
Hi Andreas

If/when you use the general "dump" facility, dumping the output is
somewhere centrally - long after the samples are read-in.
The reader, esponsible for dumping, does not know anything about the device
where the samples are coming from, apart from the nrbits per channel
The only thing is gets when dumping is the nr of bits per sample, such that
the samples - the device handler delivers them as
(float, float), can be converted to int16 values.Then the samples are
recorded in a wav file.
An ".sdr" file is nothing more (or less) than a ".wav" file, two channels,
16 bit int values and a samplerate of 2048.

If/when you use the experimental "raw" facility, the files are written as
raw files, however, the dabstick files are 8 bits
and written as 8 bit files. The hackrf samples are also written as 8 bit
files, the Airspy and SDRplay samples as 12/14 bit int files

I assume you were using the "general" dump facility
helpful?

andimik, 2020-01-02:
Yes, dump as that produces sdr files.

JvanKatwijk, 2020-01-02:
But if the processing of samples of different sources is the same - by the
same program elements - then
it is strange if the structure of their contents is different. But they are
not raw files, they are valid RIFF files
so I do not understand the sox commands you are giving

JvanKatwijk, 2020-01-02:
Btw the scaling as shown in your earlies mail seems to be different from
the original input and the
recorded file. However, as you can see, the signal/noise ratio is in both
cases roughly the same,
The scale (left side of the spectrum) is determined by the number of bits
passed from the device. From
a recorded file I cannot decipher what the input had as bitrange. It seems
to be 10 bits here, that
should give -60 dB as bottom. The filereader program reads in floats, so
the conversion is done
by the libsndfile library

closed 2022-08-28 by JvanKatwijk