JvanKatwijk/qt-dab #152: rtl_tcp input broken

mxlbzn, 2020-01-04:
Somewhere 3.11-beta and 3.2-beta, rtl_tcp input support broke.

After connection and selecting proven channel, FIC drops to 0% (is set initially to 24% for some reason) and then nothing happens. (I don't see any difference in console output between working and non-working versions.)

mxlbzn, 2020-01-04:
Update: I can see a difference on the server. In both cases, freq is 219488000 set regardless of initial default channel setting. After selecting the channel, it's not seen on server in the broken version.

JvanKatwijk, 2020-01-04:
rtl_tcp does not have much priority, however, the changes are simple
The "restartReader" function in most other device handlers was changed,
apparently rtl_tcp was just forgotten (I never use it)
In rtl_tcp_client.cpp the function restartReader should look like

bool    rtl_tcp_client::restartReader   (int32_t freq) {
        if (!connected)
           return true;
        vfoFrequency    = freq;
//      here the command to set the frequency
        sendVFO (freq);
        connect (&toServer, SIGNAL (readyRead (void)),
                 this, SLOT (readData (void)));
        return true;
}

and of course the rtl_tcp_client.h function should be modified accordingly

Changes are applied in the latest build, not tested, where I am right now I
do not have
dongles or other devices

mxlbzn, 2020-01-04:
I don't use it as well, started to play with it just today.

The fix works, great service! :)

closed 2020-01-04 by mxlbzn