JvanKatwijk/qt-dab #153: Raspberry Pi CMake build broken: emmintrin.h: No such file or directory

sjlongland, 2020-01-19:
```
[ 69%] Building CXX object CMakeFiles/qt-dab-3.21-Beta.dir/src/support/viterbi-jan/viterbi-handler.cpp.o
[ 69%] Building CXX object CMakeFiles/qt-dab-3.21-Beta.dir/src/support/viterbi-spiral/viterbi-spiral.cpp.o
[ 70%] Building C object CMakeFiles/qt-dab-3.21-Beta.dir/src/support/viterbi-spiral/spiral-sse.c.o
/tmp/qt-dab/src/support/viterbi-spiral/spiral-sse.c:23:10: fatal error: emmintrin.h: No such file or directory
 #include <emmintrin.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/qt-dab-3.21-Beta.dir/build.make:1275: CMakeFiles/qt-dab-3.21-Beta.dir/src/support/viterbi-spiral/spiral-sse.c.o] Error 1
```

It would appear it tries to drag in the x86 SSE implementation of Viterbi, when it should be pulling in a ARM NEON version instead… digging around, I found the following `diff` gets it compiling on the Pi (but will break x86).

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 460e805..e0fca6e 100755
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -267,7 +267,7 @@ add_definitions (-DPRESET_NAME)
             ./includes/support/ensemble-printer.h
             ./includes/support/viterbi-jan/viterbi-handler.h
             ./includes/support/viterbi-spiral/viterbi-spiral.h
-              ./src/support/viterbi-spiral/spiral-sse.h
+            ./src/support/viterbi-spiral/spiral-neon.h
             ./includes/support/preset-handler.h
             ./includes/support/presetcombobox.h
             ./includes/support/smallqlistview.h
@@ -358,7 +358,7 @@ add_definitions (-DPRESET_NAME)
             ./src/support/history-handler.cpp
             ./src/support/viterbi-jan/viterbi-handler.cpp
             ./src/support/viterbi-spiral/viterbi-spiral.cpp
-            ./src/support/viterbi-spiral/spiral-sse.c
+            ./src/support/viterbi-spiral/spiral-neon.c
             ./src/scopes-qwt6/iqdisplay.cpp 
             ./spectrum-viewer/spectrum-viewer.cpp 
             ./correlation-viewer/correlation-viewer.cpp
```

I think the real fix is to define the implementation of the Viterbi algorithm being used someplace else based on processor architecture, and feed that in here.  I'll slot it into the back-log somewhere, but just documenting what I've seen in case others hit the same issue. :-)

JvanKatwijk, 2020-01-19:
#For x64 linux system uncomment SSE
#For any other system comment SSE out and uncomment NO_SSE
#CONFIG += SSE
CONFIG  += NO_SSE

Use the NO_SSE, SSE is just for x64

sjlongland, 2020-01-19:
Where's the `CONFIG` line in `CMakeLists.txt`?

JvanKatwijk, 2020-01-19:
good question, I am talking about the ".pro" file.

sjlongland, 2020-01-19:
For what it's worth `cmake` upstream haven't got a "standard" way to do this yet: https://gitlab.kitware.com/cmake/cmake/issues/18445

A "good-enough" work-around on Unix-like platforms might be to look at `uname -m` … if it's `x86_64`, there's a fair bet it'll support SSE (which was introduced with the Pentium III and the AMD Athlon XP/Duron)… if it's `armhf`, then probably try building with NEON.

After making those earlier changes to `CMakeLists.txt` I got some compiler errors regarding NEON instructions…
```
/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h:6733:1: error: inlining failed in call to always_inline ‘vdupq_n_s32’: target specific option mismatch
 vdupq_n_s32 (int32_t __a)
 ^~~~~~~~~~~
In file included from /tmp/qt-dab/src/support/viterbi-spiral/spiral-neon.c:27:
/tmp/qt-dab/src/support/viterbi-spiral/SSE2NEON.h:312:33: note: called from here
```

But that's progress… looks very similar to https://github.com/kcat/openal-soft/issues/54 so I'll try seeing if a few C compiler flags will crack this nut and get the CMake build system working again.

sjlongland, 2020-01-19:
Above patch + build with options:

`cmake -DRTLSDR=ON -DRTLTCP=ON -DHACKRF=ON -DCMAKE_C_FLAGS="-mfpu=neon" .`

got things to compile.  I have Qt-DAB 3.21 beta working on the Pi3 now.  I think CMake should have enough intelligence to at least tell us what OS and CPU architecture we're compiling for, and thus we can make some sane assumptions.  (i.e. pretty sure every x86-based Apple machine ever made supports SSE, most Windows machines running current versions would too, as would anything AMD64.)

I'll do some reading up on the CMake docs and see what can be figured out.

JvanKatwijk, 2020-01-20:
I added as options to the CMakeLists.txt file
VITERBI_SSE
VITERBI_NEON
When e.g. adding -DVITERBI_NEON=ON to the command line, the right files for inclusion are selected and the proper define is set

sjlongland, 2020-01-20:
On 21/1/20 6:36 am, JvanKatwijk wrote:
> I added as options to the CMakeLists.txt file
> VITERBI_SSE
> VITERBI_NEON
> When e.g. adding -DVITERBI_NEON=ON to the command line, the right files for inclusion are selected and the proper define is set

Fantastic.  I guess that reduces the urgency on the autodetection
aspect: we can just document "on ARM compile with -DVITERBI_NEON" in the
meantime.

Later on, we should be able to figure out which is needed and define the
appropriate parameter automagically.

Is there a non-optimised (i.e. non-SIMD) Viterbi implementation that is
selected if neither parameter is specified?

JvanKatwijk, 2020-01-20:
If nothing is specified a non-sse implementation is chosen. Note that these
implementations
are for the FIC blocks, not for the audio decoding. The "optimized"
versions are generated for a limited
size of the input, the audio segments often extend this length

closed 2020-09-27 by JvanKatwijk