JvanKatwijk/qt-dab #154: Misplaced slideshows: appearing under ~/NexGen_XML_export (notably 4KQ here in Brisbane)

sjlongland, 2020-01-19:
Just noticed this the other day… basically at work I noticed a mystery `NexGen_XML_export` sub-directory appear in my home directory.  This is with an older release of Qt-DAB, but I've reproduced the problem with current `master`:

```
RC=0 stuartl@rikishi ~ $ find Nexgen_XML_export/
Nexgen_XML_export/
Nexgen_XML_export/#Digital Output
Nexgen_XML_export/#Digital Output/4KQ
Nexgen_XML_export/#Digital Output/4KQ/slot_1.jpg
Nexgen_XML_export/#Digital Output/4KQ/slot_5.jpg
Nexgen_XML_export/#Digital Output/4KQ/slot_2.jpg
Nexgen_XML_export/#Digital Output/4KQ/slot_3.jpg
Nexgen_XML_export/#Digital Output/4KQ/slot_4.jpg
```

[image]

I don't get the problem with other DAB+ stations, e.g. all the ABC ones are fine.  It is something peculiar to [4KQ](http://www.4kq.com.au), and only something they've switched on since the new year.  (Previously they had MOT slideshows disabled.)

This directory is created regardless of whether `saveSlides` is set to `1` or `0`.  (On my RPi3 I have `showSlides=1` and `saveSlides=0`… still get the files created.  On my workstations, I have both set to `0` as I find the focus-stealing behaviour of the slide-shows annoying when I'm trying to get work done.)

I'll do some digging and see what can be uncovered.  Don't think it'll be anything big, it's probably a pattern-match issue.

sjlongland, 2020-01-19:
Just tried running from the `/tmp` directory, it appeared there instead.

Thinking about this, this is a bit of a security vulnerability… not sure if some smart-alec can "broadcast" a `.ssh/authorized_keys` file to replace such files on a target user's computer.  Thankfully the work-around is easy enough (unless the paths can be absolute).

sjlongland, 2020-01-19:
Right, the culprit might be here:

JvanKatwijk, 2020-01-19:
What probably happens is that the data is not transmitted as slides. If the
data is MOT, not slideshow, e.g. a directory with a number
of html type files - these are now stored regardless the setting of
"saveSlides".

Some german stations transmit a weather forecast with lists of city names
and forecasts for these cities, some do rolling news and traffic
information as html files.
I'll see to it that it will be subject to the setting of saveSlides (it is
pretty old code)

sjlongland, 2020-01-19:
Yeah, it looks like there's a lot of curious features in DAB/DAB+.  (Some ways I wonder why the ACMA didn't embrace DRM instead… but I digress.)  Lots of features means lots of edge cases, but it seems KISS (the principle, not the band) isn't in fashion any more.

I'm glad we're doing this on a comparatively "smart" computer and not a little embedded microcontroller. ;-)  Debugging this on a bare-metal firmware image would be hell.

sjlongland, 2020-04-12:
Okay, did some digging around… firstly, I noticed there was a lot of magic numbers surrounding `contentType` and `contentsubType`… so I defined an `enum` with all of those, packing them both into a `uint16_t`.

I also added some debug statements… if I tune to the offending station I actually see this:
```
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_2.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_3.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_4.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_5.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_1.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_2.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_3.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_4.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_5.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_1.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_2.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_3.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_4.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_5.jpg
Creating MOT object type 0x0000
going to write file Nexgen_XML_export/#Digital Output/4KQ/slot_1.jpg
Creating MOT object type 0x0000
…
```

So it's actually sending it as "general data", not telling QT-DAB that there's a slide-show image there.  I see the accompanying `MIME/HTTP` content type is 0x0001 (that is, "type" 0, "sub-type" 1), and that's probably the HTML web pages you were mentioning earlier.

If I tune to a different station, they do the slide-shows properly:
```
Creating MOT object type 0x0201
Creating MOT object type 0x0201
```

So that's part of the mystery of what's going on.

sjlongland, 2020-04-12:
Right, so not sure if 4KQ are doing the right thing, I shot them a note in case one of their technical people was trying to troubleshoot the problem with a slideshow-capable radio and wondering why nothing was showing up.  (Maybe some DAB+ radios just ignore the content type in the DAB+ stream and just display images if their size/format "looks right"?)

In the mean time, my nit of it just writing files willy-nilly, looks to be solved.  I traced what happened with `picturePath` and `showSlides`.  In `RadioInterface`, the constructor there checks if `showSlides` is 1, and if so, sets `picturePath`.

If it's 0, then `picturePath`'s value is decided by the `QString` constructor… which sets it to the empty string.  (That is, `picturePath.isEmpty()` is `true`.)  That string is then passed, by value, down through several layers of objects until it eventually hits `motObject`.

`motObject` doesn't check for an empty string, and will thus blindly append the name of the file to this empty string, producing a valid path relative to the current working directory.

So this is fixed by this commit: https://github.com/sjlongland/qt-dab/commit/ee48866f0e798a8b6116de4836e0cf5b22061df8

Whilst I was there, I changed those pass-by-value to pass-by-reference, to improve the efficiency just a little bit.

sjlongland, 2020-04-12:
```
going to write file /tmp/qt-pictures/Nexgen_XML_export/#Digital Output/4KQ/slot_1.jpg
going to write file /tmp/qt-pictures/Nexgen_XML_export/#Digital Output/4KQ/slot_2.jpg
going to write file /tmp/qt-pictures/Nexgen_XML_export/#Digital Output/4KQ/slot_3.jpg
going to write file /tmp/qt-pictures/Nexgen_XML_export/#Digital Output/4KQ/slot_4.jpg
going to write file /tmp/qt-pictures/Nexgen_XML_export/#Digital Output/4KQ/slot_5.jpg
```

Now they're going to a more sensible place.  I did notice my change broke MOT slideshows altogether because I was a bit overzealous checking for empty `picturePath` where I was… so that's re-located to the part that just concerns these "data" files.

Hopefully that is fixed.  I note the logic that actually displayed the slideshow is gone, seems f5f2843966f61a9fe6d1a4261641903705eecfcc ripped a lot of that out.  I'm not sure what the plans are for this feature, but the back-end logic should still be preserved.

JvanKatwijk, 2020-04-12:
I rearranged the code a little. Writing files from deep within the code is
a left over from long ago.
The approach now is that while collecting data for mot objects is done deep
inside the backend code,
as soon as an object is complete, it is transferred to the "radio.cpp"
file, i.e. the one handling the interface.

I added a variable
epgPath
that - when not set - will prohibit creating a file with epg encoded as xml.
if set, the file will be placed in the path given.

I added a variable
filePath
that - when not set - will prohibit creating a file for the object.
If set, the file will be placed in the path given.

The setting for pictures remains, it is now only used for saving slides

JvanKatwijk, 2020-04-12:
OK,
I Included your changes as well,
again
1. pictures will be saves in picturesPath
2. epg files will be saved in epgPath if that is not empty string
3. other files will be saves in filePath if that is not empty

Mot handling is now done in radio.cpp, after completing a frame in mot-object that frame
will be transferred to radio.cpp. Sorry for all your work in changing the mode of the parameters,
the "picturePath" parameter is now gone

Thanks for all the work!!!!

sjlongland, 2020-04-12:
No problems at all.  I nearly did create a `objectPath` variable and pass that down along-side `picturePath`, but then I saw `picturePath` was being used for EPG as well.

Having `epgPath`, `picturePath` and `filePath` definitely makes sense.  If I come up with a work-around for stations that aren't correctly tagging slideshows in the stream properly (so it doesn't break stations that use it for web pages) I'll make another pull request and re-open this issue.

My prime gripe though: files getting "randomly" created in `${HOME}`, is fixed though.  Many thanks. :-)

sjlongland, 2020-04-12:
Just thinking about the slideshows themselves, I wonder if rather than popping up a new window, it might be better to have a tab that displays the MOT Slideshow?

I'm thinking making the left-hand side a tab, one tab is the presets, the other is the MOT slideshow.  Maybe some of the debugging-related buttons can move to a third tab, not sure.

andimik, 2020-04-12:
I am also not satisfied that the slides are shown in the Detail window instead of an own pop up like before.

The problem is that they are resized there. Of course you can load them from the tmp folder.

andimik, 2020-04-12:
BTW, could you share us either a rawfile or even an eti file from this ominous mux containing 4KQ?

sjlongland, 2020-04-12:
@andimik Fair enough…

I found when it was a pop-up window, every time it changed it stole focus from whatever application I was doing.  I could tolerate that on the Raspberry Pi since it was basically _only_ doing DAB+ duties, but at work, it was a nuisance.

Currently, on my Pi3 it looks like this:
[image]

Now, if we put the left-hand side in a tab… it might look something like this:
[image]

We could have a second tab which shows the MOT slideshow (disabled for stations that don't provide it).
[image]

sjlongland, 2020-04-12:
I'll see if I can do a grab… might wait until they're doing the news or something so I'm not infringing some artist's copyright.

sjlongland, 2020-04-13:
Okay, I have a raw dump (`.sdr` file)… but it's kinda big, about 1.5GB, and I note writing this *and* decoding the stream caused the Raspberry Pi 3 some consternation regarding synchronisation so there may be bits missing.

My reception isn't the best either.  I'm just using a ¼-wave whip tuned for 145.7MHz… a little longer than needed for this frequency.

I'll see if I can grab a subset that demonstrates the problem (maybe the transmission of a single slide).

sjlongland, 2020-04-13:
Okay, success this time, did it on my AMD64 laptop.  I grabbed some of the news.  This is done using current `master` branch:

http://static.vk4msl.id.au/qtdab/2020/04/13-multiplex9a/multiplex9a.sdr

I'll delete the file after you've received it.

andimik, 2020-04-13:
Thanks for the file, I've extracted an eti with `$ eti-cmdline-wavfiles -F ./multiplex9a.sdr -O foo.eti`

and analysed with XPADxpert (see https://basicmaster.de/ )

[image]

This is similar to RAI from Italy, see screenshot, also here all slides were saved in the subdirectory `data/crit/[service]/`

[image]

@basicmaster Seems to be the same encoder issue, right?

andimik, 2020-04-13:
@JvanKatwijk 

[image]

His *.sdr looks the same as I have reported in January, see https://github.com/JvanKatwijk/qt-dab/issues/149

After converting it into a raw file 

`$ sox -e signed-integer -r 2048e3 -B -t raw -c 1 -b 16 ./multiplex9a.sdr -e unsigned-integer -r 2048e3 -t raw -b 8 foo.raw`

it looks like @sjlongland  Stuart will get in live reception.

[image]

Stuart: The file size is now 236.347.392 Bytes (= 8bit u8) compared to 472.694.784 Bytes in the *.sdr (=16 bit wav) ;-)

sjlongland, 2020-04-13:
Ahh right, wasn't quite sure how the `.sdr` format worked… I tried grabbing an excerpt using `dd` (with 1MB block size), but that produced an unreadable file so figured there was some sort of header involved.

I did wonder why the capture looked so low, indeed endianness would explain it.  This is how it looks live.

[image]

sjlongland, 2020-04-13:
Question in my mind, does RAI encode the slides as `ContentType=0`/`ContentSubType=0` as well?

That seems to be the issue here.  They're not sending the files out to accompany a HTML blob for display in a web browser, I think the intent is for these to be displayed as slides.  *That* would imply these should be `ContentType=2`/`ContentSubType=1` (as they're JPEG images).

JvanKatwijk, 2020-04-13:
The interpretation of MOT data is now in the main file (radio.cpp). Here in
the netherlands the services are accompanied by a single slide,
that is why in the current setting these labels are included in the
"technical data".
In some recordings these labels will change once or twice.

I am open to extend the processing of MOT, but as said, here in the
netherlands the use is limited.

basicmaster, 2020-04-13:
> @basicmaster Seems to be the same encoder issue, right?

The related `ContentName` parameter is e.g. used to distinguish different slides; it is not required to be a valid filename. However, decoders usually use much shorter values - instead of wasting precious bytes, like in the present case.

> Question in my mind, does RAI encode the slides as `ContentType=0`/`ContentSubType=0` as well?
>
> That seems to be the issue here. They're not sending the files out to accompany a HTML blob for [...]

Exactly - the MOT Slideshow requires specific content (sub) types, for JPEG and for PNG images. It is not allowed to use the general types here.

JvanKatwijk, 2020-04-13:
can you send me  copy of that file, then I can have a look

It seems we have to distinguish between MOT service labels (as done for
e.g. the dutch services) and real slideshows
Well, I'll have to look into that

JvanKatwijk, 2020-04-15:
Is it a reasonable assumption that whenever an audio service presets a
slide show, the files belonging to that
slide show are supposed to be stored in some directory, the path being part
of the name?

With some old (australian) recordings, this seems to be the case, the audio
services for the
dutch transmissions carry a MOT slide with a single filename (the same for
some other european transmissions).

If this assumption seems to be valid, then an option would be to present a
slide show separate from the "technical details",
while the single labels remain part of the technical details widget

JvanKatwijk, 2020-04-15:
I took a pragmatic approach: I added an option to be set in the ini file
motSlides=0 (the default), will place the mot objects in the technical data widget,
motSlides=1 will place mot objects in a separate widget

closed 2020-04-15 by JvanKatwijk