JvanKatwijk/qt-dab #155: Station sort order by ID

mxlbzn, 2020-02-16:
Currently the stations are sorted alphabetically by the station label, sort order cannot be changed.

Please allow also original sorting (by SubChId?). (Often the stations in the multiplex are ordered by the "importance". Also new stations in the multiplex would appear at the bottom, so could be easily spotted.)

I realize this might hot be desired, as the stations can be renamed by the user. So, having this id (together with 0xXXXX stationId) at least in the scan output would be nice.

andimik, 2020-02-16:
Have you seen the favorite feature?

JvanKatwijk, 2020-02-16:
The original sorting is an "undefined" sorting, depending on the quality of
the signal.
The software just collects data from the FIC, as soon as sufficient data is
collected for a service (data composed
from more than one element in the FIC), such that the service can be
selected, the servicename is
passed on to the GUI handler.
Sorting was added as a suggestion - and implementation - from stuart
longland (australie),
Of course there are different criteria that could be used for sorting:
there are two obvious criteria,
alphabetically or on the serviceId, the latter probably leading to
something like the "orginal order"

JvanKatwijk, 2020-02-16:
Adding the serviceId's to the scan is not hard, you can see the serviceIds
and notice that they do not necessarily form a sequence

JvanKatwijk, 2020-02-16:
well, if you were happier with the  servicenames being unsorted, that is
easy. Sorting is done on line 558 in the file radio.cpp,
commenting that line out causes the names to be visible in the order that
they are discovered

JvanKatwijk, 2020-02-16:
OK

I added two things.
a. the overview of the scan now shows for each service the serviceId
b. an option is to add "noSort=1" to the ini file. If set, the incoming
service names are NOT sorted

mxlbzn, 2020-02-18:
Configurability in the ini file is nice.

But I don't think it's very useful to have station list unsorted at all, by the (almost random?) order as it finds frames in the stream. As the sorting means alphabetical sorting, could be the behavior change to unsorted (noSort=1) to mean not alphabetically sorted, but sorted by subchannel ID ?

JvanKatwijk, 2020-02-18:
well, it could be ordered by serviceId. Ordering by subChannelId is less
attractive from an implementation point of view

andimik, 2020-02-18:
Well, please note, that sorting by SubChannel Id or SId is not sufficient for some countries. 

For example, in Croatia, public broadcaster HRT has Subchannels 11, 12 and 13 (lowest is 4). They even don't have the smallest SId ...

Therefore, I suggest to use the favourite list which is useful when you can receive more than one DAB ensemble.

JvanKatwijk, 2020-02-18:
It all depends what the purpose of the sorting is. The original idea was to
ease finding by ordering the service name, which is
of course a very legitimate and even helpful. The benefits of orderings
based on other criteria are not clear to me yet

JvanKatwijk, 2020-02-19:
This is as example the output of one of the ensembles I receive.
Servicenames not ordered,
What would be the benefit of having the serviceIds or the subchIds sorted?

devicename = SDRplay_RSPII_VID_1DF7&PID_3000_BUS_001_PORT_001
adding GrootNieuwsRadio serviceId 8961 subchId 15
adding 538              serviceId 83c7 subchId 1
adding 538TOP50         serviceId 8802 subchId 2
adding 100% NL          serviceId 83d3 subchId 6
adding Qmusic           serviceId 83c8 subchId 10
adding BNR Nieuwsradio  serviceId 83d1 subchId 9
adding Qmusic non-stop  serviceId 8804 subchId 4
adding Radio 10 60s&70s serviceId 8818 subchId 12
adding Sublime          serviceId 82a8 subchId 13
adding Veronica         serviceId 83e1 subchId 7
adding Radio Maria      serviceId 8816 subchId 3
adding Sky Radio Hits   serviceId 8808 subchId 16
adding KINK             serviceId 8358 subchId 14
adding SLAM!            serviceId 82a5 subchId 11
adding Sky Radio        serviceId 83c6 subchId 5
adding Radio 10         serviceId 83d2 subchId 17

mxlbzn, 2020-03-21:
I think that the serviceId could roughly corellate with radio station age. Might be useful for some. Ability to specify sort order parameter in (station name, serviceId, subchId) in config file would be great.

Current behavior, not sorting the stations by default, sucks, because the order is unstable.

JvanKatwijk, 2020-03-22:
What would be the advantage of ordering service names according to the
ServiceId compared to alphabetical ordering,
that is still not clear to me

andimik, 2020-03-22:
In lots of countries public radio stations have the same naming structure, like `Radio 1`, `Radio 2` and so on. This is easy when you want to zap.

But there are also countries where you **cannot** learn it from the name. The only way is the sorting by SId, as these IDs are similar. Of course only happens where DAB ensembles have public **and** private services on board.

Furthermore, the PI-Code in RDS and the SId in DAB have the task to distinguish between countries (1st digit) and national/regional/local (2nd digit).

But, I recommend to leave it as it is at the moment. The `preset` function is good enough in my opinion.

JvanKatwijk, 2020-03-24:
If you add the line
serviceOrder=1
to the ini file the sorting will be on SIds.
I still do not understand what the advantage is though

closed 2020-03-24 by JvanKatwijk

mxlbzn, 2020-09-09:
Humbly reopening the issue. It was "solved" by enabling sort order by a different param. Possibility of ordering by subchannel ID is desired, as well.

JvanKatwijk, 2020-09-10:
Such a friendly request cannot be ignored, user "serviceOrder=2" for
ordering on subChId's

mxlbzn, 2020-09-10:
It's a handy solution. Tested. Thanks.