JvanKatwijk/qt-dab #156: Please add compatibility with Navit, a open source navigation system.

Supermagnum, 2020-03-04:
I believe that Navit uses the TRAFF format for traffic information. TraFF is an XML-based format.
Links:
https://traffxml.gitlab.io/
https://www.navit-project.org/
https://wiki.navit-project.org/index.php/Traffic_in_Navit

See: https://github.com/navit-gps/navit/pull/766

TPEG:
https://www.worlddab.org/dab/technical-specifications

JvanKatwijk, 2020-03-05:
> See: navit-gps/navit#766 <https://github.com/navit-gps/navit/pull/766>

The qt-dab software sends - if configured - the TPEG frames to an ip-port.
I know how to
extract - and check validity of - TPEG frames, I do not have information on
interpreting the
contents.

any help with that would be appreciated

JvanKatwijk, 2020-03-05:
you can contribute by providing me some dab dumpfiles with content so that
I can have a look at what it is

Supermagnum, 2020-03-05:
Sorry I cant do that.. But there is DAB+ in Norway and two traffic message data channels.
One unencrypted ( NRK ) and one encrypted ( V-traffic ).

andimik, 2020-03-05:
Jan, I will share a 5 min raw file with SWR from Germany containing TPEG.

JvanKatwijk, 2020-03-06:
> See https://transferxl.com/08YNwsGdhJsSs which is zip compressed (800 MB).
>
> But I only get special characters as output in localhost:8888
>
> ÿ�ÿ�������bÓj�ÿ�ÿ�������bÓj�ÿ�ÿ����ÿ�bÓ���],Å��� ��������������0�ARD
> TPEG&Aktuelle Verkehrsinformationen der
> ARD!������������V����������ïŒ��²V†���� ���
> �›R�^S§ì������ ^ûü0����������� ��z����˜í
> �^�à’�������;���7$�?Ã#OÉ� ����”�
> ���†R�ýÏý„� ����$�0�� ����?Ã#OÉ��>½#N �þ)���%$�Fîu�=»
> ƒ�®rÛ�®rÚ1Ò‡»1Ò‡¸1Ò‡¶=»\ù��������
> ��L�^S§ì������ ^Y›4��������f����°�
> ���à’�����J�;���7$�hè#®Î� ����ô�
> ���º0�û��ã� ����l�0�� ����hè#®Î��f—#ºg�þ������� É¾�I�¼ä*hYäº$�������
> �ÐG�^S§ì����
> �^Dƒp^bÕ´�������������^�����ì ���à’�������<���8%�¿“#©“� ����ä�
> ���†R�ýõ��� ����l�0b„e ����¿“#©“��¾Ÿ#ªÅ�þ �����8CËä��_��<��� �Û
> �^S§ì��������P����������� /ò �^�à’���¨�p���lK�B©"é*� ����Œ� ���ë.��ÛýÁ�
> ����ô�p�ùPÓn� ����Œ� ���“��üû÷î� ����|� ���…"���
> ����B©"é*��?‹"Ôd��>""Ð¡��>ˆ"Ï”�þ��������”þ: <³h!œkr�'Ë�þü$��t���
> �§��^S§ì������ ^Ò×�[���
> -O
> �Nà’�������;���7$���#��� ����|�
> ��������ÿl� ����ü�0�� ������#������#�Î�þ�������(4’þZ��>���
> �Û'�^S§ì�������Ÿ ��&��
> .l

Well, yes. The TEPG output is definitely not source encoded xml.
The TPEG data consists of blocks, each block is either a type 0 (kind of
directory) or a type 1 (data) block
The Qt-DAB software checks the validity of the blocks by checking the crc
and passes the data - uninterpreted to the IP port.

JvanKatwijk, 2020-03-06:
Looking at some older recordings (and the internet) I found that many TPEG
messages are encrypted,
I did not find - so far - TPEG data that is xml, as soon as someone is
providing me with a raw image with
traffic info in xml encoded as TPEG I'll continue

andimik, 2020-03-06:
Just one question to understand, in case this is in the clear but in binary format, could we see it in the output?

I found a list of all TPEG services worldwide thanks to Günter Lorenz

http://www.fmlist.org/tpeg.php

I have lots of samples at home (or can receive them via satellite feed) and will tell you if I found xml.

JvanKatwijk, 2020-03-06:
Looking at one ensemble here and some (german) recordings, it seems that
most data in TPEG is encrypted
No idea how to decrypt that without a key. The dutch encrypt with key 0201,
the germans with key 0207, but then ???

The file you sent me was not encrypted, but contained, as you already
noticed, mostly binary data.
The main point is - I think - that while I am able to extract TPEG data, I
do not know how to interpret the bits and bytes.
The document I have differentiates between type 0 and type 1 frames (type 0
more or less description of type 1 data).
and I can extract the packets from the type 1 data, but for me there it ends

Supermagnum, 2020-03-06:
> Looking at one ensemble here and some (german) recordings, it seems that most data in TPEG is encrypted No idea how to decrypt that without a key. The dutch encrypt with key 0201, the germans with key 0207, but then ??? The file you sent me was not encrypted, but contained, as you already noticed, mostly binary data. The main point is - I think - that while I am able to extract TPEG data, I do not know how to interpret the bits and bytes. The document I have differentiates between type 0 and type 1 frames (type 0 more or less description of type 1 data). and I can extract the packets from the type 1 data, but for me there it ends Op vr 6 mrt. 2020 om 17:33 schreef andimik <notifications@github.com>:
> […](#)
> -- Jan van Katwijk

In Norway there are one unencrypted ( by NRK ) and one encrypted ( v-traffic ) if that is any helpful.

JvanKatwijk, 2020-03-06:
It would be helpful if you can provide a recording

Supermagnum, 2020-03-06:
> […](#)
> -- Jan van Katwijk

I can't, but I have asked for the recording on the NRRL ( Norwegian Hamradio association ) Facebook page. I hope that someone there can help.

andimik, 2020-03-07:
Please note that even professional software analysis gets the same output like Qt-DAB 😉

[001_stream12_16kbps_48bytes_per_frame_2306frames_55.344seconds.dab_address1.pkt.txt](https://github.com/JvanKatwijk/qt-dab/files/4301787/001_stream12_16kbps_48bytes_per_frame_2306frames_55.344seconds.dab_address1.pkt.txt)
[001_stream12_16kbps_48bytes_per_frame_2306frames_55.344seconds.dab_address1.dg.txt](https://github.com/JvanKatwijk/qt-dab/files/4301788/001_stream12_16kbps_48bytes_per_frame_2306frames_55.344seconds.dab_address1.dg.txt)

Source: rfmondial online (demo for NDR and German Bundesmux)

closed 2020-04-18 by JvanKatwijk