JvanKatwijk/qt-dab #157: Occaisional temporary loss of sync

NomisGninnac, 2020-03-15:
On RPi3 running latest compile.

Getting audio glitching every so often,perhaps loss of sync or some sort of latency issue perhaps, I don't know.  version 3.2 beta works well.

JvanKatwijk, 2020-03-15:
is there anything different in the load on the RPI?

NomisGninnac, 2020-03-15:
no.  I didn't notice any spikes.  I run the pi dedicated to qt-dab, nothing else runs on it.

JvanKatwijk, 2020-03-16:
Between 3.2 and 3.3 nothing was changed in the actual processing of the signal (from incoming samples to audio out), only in the way control data is registered and retrieved.
It would be interesting to see whether the problem appears in the front end part, i.e the actual
synchronization or in the decoding of the AAC data. 

I'll have to find out a way to add some instrumentation code

JvanKatwijk, 2020-03-16:
do you have DAB or DAB+ (i.e. mp2 or AAC audio frames)

JvanKatwijk, 2020-03-16:
I added a "-T" flag.
If you pass that plag as command line parameter, there will be some output
every 10 seconds
the output gives 2 lines
the first line 4 numbers:
the total amount of frames started to be recognized (around 104)
the number of good frames, i.e. that could be recognized and the contents of which is passed
on for further processing
the number of bad frames.
Usually on the start of listening to a channel, the number of bad frames will be higg since
no synchronization took place yet. Once synchronization is established, the number of
bad frames should be 0, each time it is not zero, something went wrong in syncing and the
content of the frame could not be used
The fourth number is the accumulated error ratio of the "FIC" handling. Errors in FIC usually
mean errors in the audio content as well.

The second line gives the number of samples missed on the 48000 samples/second rate output.
Due to timing and clock differences it might happen that that number is not always 0, but ideally it is
I just had my RPI3 run for 5 hours with only some glitches in the output when the "bad" frames count is not zero (which makes sense, since contents gets lost)

Hope this gives some insight in what causes the glitches

NomisGninnac, 2020-03-16:
it should do both according to the label but i cant see anything in the interface to tell me.

JvanKatwijk, 2020-03-16:
It is a commandline option
i.e ./linux/qt-dab...  -T

closed 2020-09-27 by JvanKatwijk