JvanKatwijk/qt-dab #164: Dabstick selection causes crash

MMSSolutions, 2020-04-17:
Hi, when I select DABstick I am able to see the software detects the RTL-SDR V3 dongle but as soon as I press scan or change one of the bands the application hangs and dissapears from the screen. I have tested Airspy, HackRF and SDRPlay which all work okay. I have tested a NooElec Smartee and RTLSDR V3 using Dabstick selection and they both make the app crash.

I am using Windows 10 and latest 3.4 build, same happened with previous version.

NooElec and RTL dongle both work fine in SDR# so drivers seem to be working okay.

Any thoughts? Is there a way to see the crash dump?

JvanKatwijk, 2020-04-17:
I'll look into it, I must admit that I am seldom using a stick

andimik, 2020-04-17:
Well, I fear some of the latest commits lets it crash. Same here with rawfiles and live reception.

JvanKatwijk, 2020-04-17:
My old dabsticks give normal results for what I can see

MMSSolutions, 2020-04-17:
Very strange, which Dabsticks are you testing with? Is there a way to see a crash log?

MMSSolutions, 2020-04-17:
Also, which driver do you have installed for your Dabstick?

JvanKatwijk, 2020-04-17:
I have two old sticks, a Noxon and a generic 820T based one.
I use the Osmocom driver which I compile myself

I run some more tests with the sticks tomorrow (the sticks need too much of
antenna to run from my chair)
There is most likely something wrong, it would be helpful if I can generate
the crash.

JvanKatwijk, 2020-04-17:
Can you send me a rawfile so that I can see what is happening

JvanKatwijk, 2020-04-18:
Well,
There is one condition where I can get Qt-DAB to crash under Windows using
a DABstick (but maybe with
other devices as well): If I start scanning when a service list is being
created.
That is reproducible so I will fix it this weekend

Further no encountered problems with my 2 dabsticks

andimik, 2020-04-18:
But I don't understand why the paths are different when you make it. 

All *.cpp.o files from the dir dab-cmdline have the expected path
all other *.cpp.o files (from the library) have this useless pwd path included

$ make
Scanning dependencies of target dab-scanner-airspy
[  2%] Building CXX object CMakeFiles/dab-scanner-airspy.dir/home/andreas/apps/dab-cmdline/devices/airspy-handler/airspy-handler.cpp.o
[  4%] Building CXX object CMakeFiles/dab-scanner-airspy.dir/main.cpp.o

Andreas

m1ctk, 2020-04-19:
Yes with me 'dabstick' will crash as soon as I press scan. my SDR play rsp1a will scan ok but stations are not saved or show up anyway even though they do for a second or so during the scan. bizarre ! version 2 works btw

JvanKatwijk, 2020-04-19:
Windows or Linux

m1ctk, 2020-04-19:
Windows 10 64bit

JvanKatwijk, 2020-04-19:
Ok, I have homework!

m1ctk, 2020-04-19:
Thanks for all your excellent work

JvanKatwijk, 2020-04-21:
I rewrote (part of) the code used in switching channels, there might have
been a timing issue there.
As said earlier, I do not have the problem so I cannot guarantee that it is
solved.
I uploaded a new windows installer.
Let me know the result

m1ctk, 2020-04-21:
Great but where do I downlead it or did you just replace the installer here ? https://github.com/JvanKatwijk/qt-dab/releases/tag/3.4

JvanKatwijk, 2020-04-21:
I just replaced the installer

m1ctk, 2020-04-21:
OK it won't crash now with my rsp1a whixch is fantastic but after a scan there is no station list showing. I can only get 1 mux worth showing, I can receive 6 here so if I want another mux to show I have to again scan for that mux...no way to show the full list ?

andimik, 2020-04-21:
If you use the presets for your favourite stations (which might be in different muxes) there is no need to scan again. It will tune to the right channel and play the right service.

m1ctk, 2020-04-22:
ok but I thought the scan result of all muxes would display in the box and you just highlight / click on the one you want to play, that would be for more useful for me anyway with over 100 stations I would like to run through

m1ctk, 2020-04-22:
Ah the XX button brings up a clickable full list ok thanks, great

JvanKatwijk, 2020-04-24:
The xx list is the history, I just saves all services ever selected (uintil
the list is cleared of course).
The presetlist is under your control, select a service, touch the name of
the selected service with the right mouse button
and it will appear in the presetlist.

The scan result will list the received ensembles and their components,
however, it will not save them.

m1ctk, 2020-04-24:
OK can you make it so they can be saved though ? plus one day ad support for airspy HF+ ?

andimik, 2020-04-24:
This is technically not possible. The bandwidth is limited. Please use another hardware.

https://airspy.com/airspy-hf-plus

m1ctk, 2020-04-24:
Yes ok it's working very well, my rsp1a seems to be very sensitive on DAB, Rouen received this morning here in Brighton UK  https://www.youtube.com/watch?v=BtL9BtdpdWA&t=6s

JvanKatwijk, 2020-04-25:
The latest version now asks whether or not you want to save the result of a
scan in a file (textfile, readable by e.g. LibreOfficeCalc)

m1ctk, 2020-04-26:
I just tried my RTL V3 and after a few secs after I hit scan your prog exists. All really good with my rsp1a though. it is excellent an sniffing out any hint of dx I find with 10A from Rouen. DAB player and Welle.io find nothing however

JvanKatwijk, 2020-04-26:
Apparently rtl V3 sticks behave differently from the "old" ones, I do not
have a decent spec, so for noe I am helpless here

m1ctk, 2020-04-26:
Trying another R820T and got a full scan THEN it exited ! Oh well

JvanKatwijk, 2020-04-26:
Do I understand it correctly that the software crashes with rtl V3 in the
scanner function?
How about other use of the rtl V3 (i.e. is it specific to the scanner)

m1ctk, 2020-04-26:
Coreect yes crashes after or at the start of a scan

JvanKatwijk, 2020-04-27:
I have a version of qt-dab now that generates some output on the console.
Using an RT820 based stick I get an occasional crash (app one out of 10 to
15 scans),
they seem to appear on the same place in the code.
I also noticed that with other devices (my old Noxon stick, the SDRplay RSP
1a and Airspy) I do not get the same phenomenon,
I did not manage to get them crashed, not normally and not while scanning.

My question is if you want to test with the attached version - preferably
running from a command window such that the
output remains visible - and send me the generated output in case of a
crash with a stick

The attachment is a zipped folder, unzip the folder, everything you need to
run is in that folder, and run the qt-dab-3.4.1 executable

If the output you get is - more or less - consistent with the output I get,
then I dive into windows, compile
and run from within windows with a debugger (I only cross compile from
Linux, no cross debugging tools)

Anyway, that will be a great help in solving this issue

https://www.dropbox.com/s/pe1ndi0kfvsa9eq/windows-qt-dab.zip?dl=0
Thanks in advance
best
jan

andimik, 2020-09-27:
As there is no progress, I suggest to close this issue.

closed 2020-09-27 by JvanKatwijk