JvanKatwijk/qt-dab #166: fdk-aac

DavideSpiniella, 2020-05-06:
hello everyone someone can help me, I installed the fdk-aac from source and when I try to do a make it ends with an error these are the most relevant rows
/usr/bin/ld: impossibile trovare -lfdk-aac
collect2: error: ld returned 1 exit status

JvanKatwijk, 2020-05-06:
what library is installed from the installation process? is there a
libfdk-aac installed in /usr/local/lib?
what system you are running?

DavideSpiniella, 2020-05-11:
I try with libfdk-aac as option  the library is installed in /usr/local/lib but is called libfdk-aac-dab.so I don't know if it is correct. I'm using a Kubuntu 19.10

JvanKatwijk, 2020-05-12:
That is the problem with different Linux distributions, libraries get
different names in different distribution
fdk-aac {
        DEFINES         += __WITH_FDK_AAC__
        INCLUDEPATH     += /usr/local/include/fdk-aac
        HEADERS         += ../includes/backend/audio/fdk-aac.h
        SOURCES         += ../src/backend/audio/fdk-aac.cpp
        LIBS            += -lfdk-aac
}

In this section in qt-dab.pro the configuration specification for including
fdk-aac is given,
you have to change the name of the library in LIBS from lfdk-aac ->
lfdk-aac-dab

DavideSpiniella, 2020-05-12:
I solved thanks you

closed 2020-05-15 by JvanKatwijk