JvanKatwijk/qt-dab #167: Don't recognize my dabstick

JackvanZ, 2020-05-10:
Hello,
Im using a rtl-sdr 2832u usb device.
It is working flawless with GQRX.
The dabstick is not being recognized under windows as Ubuntu with the appimager.
Can you help me?

JvanKatwijk, 2020-05-10:
What driver did you install in Ubuntu and what version ubuntu you are
running
(try to first address the issue for Linux)

andimik, 2020-05-10:
I had the same problem, but I could solve it. Will tell you later what I did.

JackvanZ, 2020-05-10:
Is this what you mean
Bus 001 Device 007: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Ubuntu 20.04 LTS
I use the appimager

Thanks! Jack

JackvanZ, 2020-05-10:
> I had the same problem, but I could solve it. Will tell you later what I did.

Thanks, let me know :-)

JackvanZ, 2020-05-10:
> What driver did you install in Ubuntu and what version ubuntu you are running (try to first address the issue for Linux) Op zo 10 mei 2020 om 16:51 schreef JackvanZ <notifications@github.com>:
> […](#)
> -- Jan van Katwijk

Is this what you mean
Bus 001 Device 007: ID 0bda:2838 Realtek Semiconductor Corp.### RTL2838 DVB-T
I have a RTL2838, i don't know of your software support it?
Ubuntu 20.04 LTS
I use the appimager

Thanks! Jack

JvanKatwijk, 2020-05-10:
Would be great to see it solved andreas!
Is it a typical Ubuntu 20.04 issue?

andimik, 2020-05-10:
Yes, Ubuntu (+ Ubuntu Mint) 20.04 was my first distribution where the DABSTICK first did not work.

I have checked my history and found these lines:

```
[...]
   16  git clone --depth=10 https://github.com/JvanKatwijk/qt-dab.git
   17  sudo apt-get install hackrf airspy
   18  sudo apt-get install rtl-sdr
   19  cd qt-dab/
   20  mkdir build
   21  cd build/
   22  qmake ..
   23  make
   24  ./linux-bin/qt-dab-3.4 
   25  sudo apt-get install librtlsdr0
   26  sudo apt-get install librtlsdr-dev 
   27  ./linux-bin/qt-dab-3.4 
[...]
```
In 24 I seem to have started Qt-DAB for the first time but the stick did not work.

Maybe you try to install the two packages mentioned in line 25 and 26.

JvanKatwijk, 2020-05-10:
It might be an issue with setting the udev rules

JvanKatwijk, 2020-05-10:
check the udev rules
(/etc/udev/rules.d/ ) there should be file for allowing "regular" user
access to the device

JackvanZ, 2020-05-10:
i'l get a error message.
In file included from ../dab-maxi/radio.cpp:83:
../devices/soapy/soapy-handler.h:34:10: fatal error: SoapySDR/Device.hpp: Bestand of map bestaat niet
   34 | #include <SoapySDR/Device.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Does this mean i'l have to install soapySDR?

andimik, 2020-05-10:
You should comment out all devices you do not have in `qt-dab.pro` line 296ff.

JackvanZ, 2020-05-10:
> You should comment out all devices you do not have in `qt-dab.pro` line 296ff or 232ff in `dab-mini.pro`.
i just did but i get the same error message

JvanKatwijk, 2020-05-11:
Yes, you have to install soapy yourself
Btw, if you are compiling on/for Ubuntu 20, you have to install the "old"
faad-2.8 library. It seems that the default
installed faad library is somehow incompatible with the DAB AAC frames

JackvanZ, 2020-05-11:
The appimager is working.
I'l get no error message but he find no stations.
Maybe the stick doesn't support DAB+
I recompile RTL-SDR and put this blacklist dvb_usb_rtl28xxu in the blacklist,conf
Thanks for your help, during the process i learned a lot.

andimik, 2020-05-13:
Seems soapy is now mandatory. I cannot compile without:

```
cmake .. -DRTLSDR=ON -DRTLTCP=ON  -DAIRSPY=ON -DHACKRF=ON
```
leads to

```
CMake Error at CMakeLists.txt:430 (message):
  please install libmirsdrapi-rsp.so

```

JvanKatwijk, 2020-05-13:
Can you be slightly more specific:
What are you compiling, libmirsdrapi-rsp is - if needed - loaded dynamically

andimik, 2020-05-13:
Sorry, forgot to mention, this error occurred when building dab-2 with cmake.

Using `cmake .. -DRTLSDR=ON` dab-maxi and dab-mini can be built, but dab-2 not. As you see, I did not specify any other device that a dabstick (rtlsdr).

The error comes from `set(SDRPLAY_V2 true)` in line 53 of `CMakeLists.txt`, I just had to comment it out.

But a rtlsdr is not (yet) supported by cmake. You can only use it with qmake.

So no need to change it in my opinion.

JvanKatwijk, 2020-05-14:
true, but dab-2 is (very) experimental. I put in in the same sourcetree
since it shares over 90 % of the sources and maintaining
consistency between different versions, each in their ow repository does
not work.
As far as I can see, the performance of dab-2 is slightly better than that
of qt-dab and its architecture offers more possibilities
for finetuning the settings of the device

andimik, 2020-05-14:
@JackvanZ

Does it work now?

JackvanZ, 2020-05-14:
Appimager works.. but i didn't receive dab+ signals. I have another dab+
receiver and that's working fine. Maybe my cheap dongle don't receive in
that range. However gqrx is working flawless. I downgraded my Ubuntu tot
V18.

andimik, 2020-05-14:
Can you see the signal in the spectrum in gqrx?

JvanKatwijk, 2020-05-14:
I assume you have a decent antenna,
what do you see on the spectrum display at channels where you expect to
find data

JackvanZ, 2020-05-14:
Could you tell me what that spectrum is? Inam realy a newbie :-)

JvanKatwijk, 2020-05-14:
touch the spectrum button on the main GUI, then you get a widget with a
spectrum of the incoming signal,
and - if something can be detected - a constellation diagram

andimik, 2020-05-14:
DAB is between 177 and 230 MHz. On https://www.fmlist.org/sendertabelle/dab-ww.php you can find your country and region.

JvanKatwijk, 2020-05-14:
The appImage will complain if the frequency of the selected channel in Band
III can not be set.
Just start the appImage, select as device the rtlsdr stick, touch the
spectrumbutton and select some channels, preferably ones you
know they carry DAB contents, and show us the results

JackvanZ, 2020-05-14:
Saw a signal on gqrx. Not to strong. I used another antenna and it works.
Thanks 4 the info and patients

Jack

closed 2020-05-15 by JvanKatwijk