JvanKatwijk/qt-dab #170: dab-maxi and dab-mini don't play subch 0 any more

andimik, 2020-05-13:
Very strange, there is no audio if this is subch0. Previous versions always worked (because I regularly record it with Qt-DAB), so the error is quite new.

dab-2 is not affected. Here you can listen to it.

In Slovenia it's Slo 1 Prvi, in Austria it's Arabella Relax. 

[Qt-DAB-10D-2019-jun-10-12-44.txt](https://github.com/JvanKatwijk/qt-dab/files/4620914/Qt-DAB-10D-2019-jun-10-12-44.txt)
[Qt-DAB-10D-2020-may-13-11-00.txt](https://github.com/JvanKatwijk/qt-dab/files/4620915/Qt-DAB-10D-2020-may-13-11-00.txt)

JvanKatwijk, 2020-05-13:
It has/had to do with startaddress 0 rather than subchId 0. I made a clear
separation between the options
to have backend code run in the main thread or in a separate thread:
running in the main thread had a studpid error

andimik, 2020-05-13:
Confirmed, now it works after https://github.com/JvanKatwijk/qt-dab/commit/ace2a3e08e225dd2b193afce84f778540091dd31

closed 2020-05-13 by andimik

JvanKatwijk, 2020-05-13:
> Confirmed, now it works after ace2a3e

The reason is was OK with dab-2 is that dab-2 did the backend in a separate
thread, that was the "old" code