JvanKatwijk/qt-dab #172: Sdrplay 3.07 api support

alphafox02, 2020-05-24:
It looks like the latest api you support is 3.06. Could support for sdrplay api 3.07 be added?

JvanKatwijk, 2020-05-24:
Is there a 3.07 version? I cannot find it

alphafox02, 2020-05-24:
There’s a bit of a typo I pointed out to them. If you go to the sdrplay website, download section and look at the Linux downloads it’ll say api 3.06 but click download and you’ll find that it downloads the 3.07 run files. I only happen to realize this after I look at their stream of press releases and noticed they specifically mentioned a new 3.07 api.

alphafox02, 2020-05-24:
You can also find it specifically referenced here,

https://www.sdrplay.com/rspdx-software/

JvanKatwijk, 2020-05-24:
I see
Not clear what - if any - the API differences are,
Using a precompiled version gives problems since the precompiled version(s)
are compiled with 3.06
If there are no - relevant API - differences then it can be solved by
changing the api match from equality to "at least"
What is your use case?

alphafox02, 2020-05-24:
I am honestly just getting familiar with sdrplay and have the rsp1a and I think duo on the way here - so nothing in hand yet. I was looking to include sdrplay and as many programs that work with it in a Linux distro I’ve created. I actually cannot even test at-dab here as far as receiving anything since I’m not in Europe and I don’t think there’s anything here where I’m at that would be broadcasting. I just figured I’d help others.

JvanKatwijk, 2020-05-24:
OK

I asked the sdrplay people what the differences are and to what extent they
relate to the - limited use - I make of
the API. In the 2.XX series there was - apart from 2.12 -> 2.13 - upwards
compatibility, and I presume that that will
be the case between 3,06 and 3.07.

Btw the RSP1a - my absolute favorite - works fine with 2.13 and 3.06, as
fas as I can see extensions in 3.07 relate to
the dx version (successor of the old RSP-II)

Anyway, as soon as I hear from the sdrplay people I'll take some action

best
jan

alphafox02, 2020-05-24:
Thank you and I am very excited to receive the rsp1a and the duo!

JvanKatwijk, 2020-05-24:
It is fixed already, 3.06 and 3.07 are - apart from one parameter that I a
do not use - compatibel
good luck

alphafox02, 2020-05-24:
Excellent! For AppImage as well? I’ll go look.

JvanKatwijk, 2020-05-24:
probably in the appImage tomorrow, when I upload the modified sources

alphafox02, 2020-05-27:
Some feedback, and it's most likely something I'm missing on my end, but when I start QT-Dab with the appimage sometimes it'll show mirics-sdrplay in the window and sometimes it'll say rsp1a. When it says mirics I can see in the terminal that the api has failed. 

api opened
api version 3.070000 detected

(qt-dab:4419): GLib-GIO-CRITICAL **: 21:53:28.345: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

(qt-dab:4419): GLib-GIO-CRITICAL **: 21:53:28.345: g_dbus_proxy_new: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
1 devices detected
sdrplay_api_SelectDevice failed sdrplay_api_Fail
De taak is gestopt

When it happens to say rsp1a in the gui then the api does not fail. I was thinking maybe it was something with blacklisting. Not sure.

JvanKatwijk, 2020-05-27:
It seems you do not have a device connected.

alphafox02, 2020-05-27:
Hi. I have both a rsp1a and a duo. Right prior to it saying api failure which happens more than it not failing, it’ll say 1 device detected. I’m not very familiar with qt-Dab but when I plug one or the other device in, then start qt-dab I can see that the terminal shows the “api failed” and when that happens the GUI shows mirics-SDR. If I then go to the drop down box in the GUI and flip flop devices around landing back on sdrplay3 the terminal will sometimes show no api failure and it’ll say rsp1a on the GUI. I can then go hit spectrum or whatever the option is to pop up another GUI showing the spectrum and it’s being populated with data - so I figure it must be working. Just not sure why I get the api failure sometimes. I’ll keep testing.

JvanKatwijk, 2020-05-27:
Did you also install the 2.13 library?
I have sometimes problems when starting qt-dab with the 3.06 library,
especially when sdrplay-v3 is in the ini file recorded as starting device

alphafox02, 2020-05-27:
Only api 3.07 and the associated soapysdrplay module built from source. Nothing else. In Cubicsdr 2.6a it works fine. I’ll look over everything again today and see if anything else stands out that may be causing an issue.

JvanKatwijk, 2020-05-27:
The 2.13 library is easier (well, at least for me), the 3.06 version is -
as you know - built as a service.
They can live together

JvanKatwijk, 2020-05-27:
Running the appImage it sometimes (actaully quite often) happens that when
the sdrplay-v3 is selected as "startup" device
(i.e. it was the device on closing the previous program invocation), the
init fails.
Deselecting that entry, by just clicking on the top of the list in the
combobox followed by selecting sdrplay-v3 helps (here)

alphafox02, 2020-05-27:
I think that’s what’s happening. As to the older api - I got confused on how they can exist together as the soapysdrplay module on pothosware vs the other soapysdrplay seems to be specific to the api. Hope that make sense. Like for the 3.07 api I couldn’t use what was on the pothosware page. It wouldn’t compile.

closed 2020-09-27 by JvanKatwijk