JvanKatwijk/qt-dab #173: Raspberry pi 4

mxdlinux, 2020-06-04:
hello 
I have a rpi4 4gb, I tried different ways to install qt-dab without success under raspbian and ubuntu 20.04 and 18.10, no AppImage launches and for installation via terminal, many libraries are obsolete.

andimik, 2020-06-04:
I have a Rpi4 under Raspbian.

Could successfully build latest dab-maxi with qmake. Works as expected with my stick. 

dab-mini fails with warnings regarding neon.

Could you please attach your `qt-dab.pro` file? Have you tried only to activate those input devices (hackrf, Airspy, etc.) you are really owning?

JvanKatwijk, 2020-06-04:
In the documentation (the "user's guide") there is a script for loading
libraries and compiling the stuff.
Assuming you are installing under raspbian buster, can you give some
details of what you did to install.
appImages run on x64 PC's, are you saying that the appImage does not run on
an x64 PC under Ubuntu 18.10 and 20.04?
So, it would be helpful if you specify
a. the raspbian version you are using for the installation
b. the result of executing the lines in the aforementioned script to load
the required libraries

Given that output, we can try to guide you further in the installation
process

best
jan

mxdlinux, 2020-06-05:
Hello
I tested the ARM version without succès .
My version of raspbian is the 10 buster but 32bit
the problem would come from there?

andimik, 2020-06-05:
https://www.raspberrypi.org/downloads/raspberry-pi-os/

This is not the reason.

Please post the error messages (paste it into a txt file and attach it here to avoid too long postings)

Have you installed all the necessary components?

JvanKatwijk, 2020-06-05:
There is no arm version

JvanKatwijk, 2020-06-05:
Oh I see,
Long time ago there was a kind of appImage for RPI, but that was strictly
bound to wheezy,  it was abandoned long time
ago.
The documentation - in the docs section of the sourcetree for qt-dab -
contains an extensive description
a.o. how to configure and set up qt-dab (or dab-mini)

JvanKatwijk, 2020-06-07:
If you run the scripts as given below - assuming a recent raspbian version on an RPI
the required libraries will be installed, the sourcetree for qt-dab is loaded and
the qt-dab executable is installed

   sudo apt-get update
   sudo apt-get install git cmake
   sudo apt-get install qt5-qmake build-essential g++
   sudo apt-get install pkg-config
   sudo apt-get install libsndfile1-dev qt5-default
   sudo apt-get install libfftw3-dev portaudio19-dev
   sudo apt-get install libfaad-dev zlib1g-dev 
   sudo apt-get install libusb-1.0-0-dev mesa-common-dev
   sudo apt-get install libgl1-mesa-dev libqt5opengl5-dev
   sudo apt-get install libsamplerate0-dev libqwt-qt5-dev
   sudo apt-get install qtbase5-dev

Assuming you want to run with a dabstick (RT2832 based stick), you can install support for 
these devices by executing

   git clone git://git.osmocom.org/rtl-sdr.git
   cd rtl-sdr/
   mkdir build
   cd build
   cmake ../ -DINSTALL_UDEV_RULES=ON -DDETACH_KERNEL_DRIVER=ON
   make
   sudo make install
   sudo ldconfig
   cd ..
   rm -rf build
   cd ..

Finally, you load the sourcetree by executing

  git clone https://github.com/JvanKatwijk/qt-dab.git

Once loaded, the following script will do all the work needed

   cd qt-dab/dab-maxi
   mkdir build
   cd build
   cmake ..  -DRTLSDR=ON 
   make

The executable is put in the build directory, installing is by

   sudo make install

Note that you best reboot before trying the executable since installing the software for the device changes the udev files, and the udev subsystem need to be reinitialized.

And, once the sourcetree is loaded it is best to look at the documentation, which is stored as a pdf file in the subdirectory "docs"

mxdlinux, 2020-06-07:
hello
I followed the same instructions and now it works.
The appimage still does not start and I have to start again some commands so that it works, I don't know why but in the end qt-dab is installed and works. 
many thanks for your help.

JvanKatwijk, 2020-06-07:
I think there is a misunderstanding of terms.
The appImage is a self contained executable, built for X64 architectures,
and available in the releases section of the repository
There is no appImage for arm based systems.

If you build an executable it is just that, an executable.

Anyway, glad it works

mxdlinux, 2020-06-07:
I just tested the application and my stick is well recognized but despite several scans, it finds nothing while with another application I have 44 radio available

JvanKatwijk, 2020-06-07:
Send an image of your screen, showing the main GUI, the controller widget
and the spectrum widget

mxdlinux, 2020-06-07:
[image]

JvanKatwijk, 2020-06-07:
Your selected channel 13F, and the spectrum does not look like a spectrum
for a DAB signal.
On what channel did/do you expect a DAB signal, and please show me the
picture after selection of that channel

mxdlinux, 2020-06-07:
[image]
[image]

JvanKatwijk, 2020-06-07:
There is something wrong with your signal, maybe the antenna?
It is strange that the spectrum as shown for channel 11B is a look-alike of
the spectrum as shown for channel 13F,

I assume you tried different gain settings for the device

To assess whether or not the software is working, you could try the file
the link points to
https://www.dropbox.com/s/1hkwqulaeavwpch/5C_DR%20Deutschland_2018-01-06_112205.raw?dl=0

If it plates correctly, it is your device that somehow gives strange results

andimik, 2020-09-27:
As there is no progress I suggest to close the issue.

closed 2020-09-27 by JvanKatwijk