JvanKatwijk/qt-dab #174: add ADALM PLUTO in the list of supported devices

DG9BFC, 2020-06-15:
i am sure a clever programmer can add the adalm pluto to the list of supported devices ... but sadly i am not a clever programmer :-(
the needed driver files are open so it should not be that difficult to add this sdr in the list
so ... please ... can you add the pluto in the list and add the needed drivers to the installer?!?
thank you sooooo much
dg9bfc sigi

JvanKatwijk, 2020-06-15:
well, if you send me (a link to) the API of the device I'll give it a try.
But I cannot test the result

andimik, 2020-06-15:
You could try to use a fifo.

`mkfifo stdin.raw`

Then stream a 8 bit unsigned signal from your device to this fifo. In Qt-DAB just open this fifo.

JvanKatwijk, 2020-06-15:
... and furthermore, there is an soapy interface driver for the pluto
Nevertheless, if the API is more or less similar to one of the devices
already supported, it would be an interesting experience to create one

DG9BFC, 2020-06-15:
https://wiki.analog.com/university/tools/pluto/users
here you should find further info and some links
should be easy for a programmer to find the needed packages
greetz sigi dg9bfc

DG9BFC, 2020-06-15:
and from here:
https://wiki.analog.com/university/tools/pluto/controlling_the_transceiver_and_transferring_data
.....

 PlutoSDR runs Linux. The device drivers allowing you to control the transceiver and capture samples are therefore part of to the Linux Industrial IO (IIO) framework. IIO is a kernel subsystem for analog to digital or digital to analog converters and related hardware. IIO communicates with user space via sysfs and a character devices. From a plain user interaction point of view this is quite intuitive, since everything is just a file. However when controlling the device from software, this can be quite painful, since you simple want to call a function or method, instead of doing string manipulation and file IO. LibIIO fills this gap, provides all sorts of device abstraction and handles all kind of IIO internals. LibIIO is cross platform and also provides different language bindings, so that you can control IIO devices from C, C++, C# or Python.

.......
hope that helps a bit
and ... if you not already own a pluto ... GO AND GET ONE :-)

andimik, 2020-06-15:
According to this link a PlutoSDR cannot receive DAB+ out of the box.

https://wiki.analog.com/university/tools/pluto/users/intro 

>  Receive (labeled 'Rx')
> 300 MHz - 3.8 GHz

Seems a hack is necessary: 

https://www.rtl-sdr.com/adalm-pluto-sdr-hack-tune-70-mhz-to-6-ghz-and-gqrx-install/
https://wiki.analog.com/university/tools/pluto/users/customizing

@DG9BFC 
Can you confirm you can receive VHF III Band (177 MHz - 230 MHz)?

DG9BFC, 2020-06-15:
yes ... definitely ...
also on fm band ...
with the hack it works 70 to 6000
even lower ... if you set local to 70 and use max sampling speed you can go down to 40 mhz ...
so dab on vhf should be easy

DG9BFC, 2020-06-15:
# fw_setenv attr_name compatible
# fw_setenv attr_val ad9364
# fw_setenv maxcpus
# pluto_reboot reset
every pluto owner knows (or should know) these 4 lines ...

andimik, 2020-06-16:
Can you set the sampling rate to 2,048 MS or 4,096 Ms?

JvanKatwijk, 2020-06-16:
For creating interface code to qt-dab one needs
a. open the device
b. set the samplerate to 2048000 and a bandwidth of app 1.5 MHz
c. ability to set and change frequency and gain settings
d. read data

I saw example code to open the device, set the samplerate, bandwidth to
large values
No clue how to change a frequency setting or a gain setting though

Wrt buying one: what would be the advantage over the hackrf, limeSDR for
higher frequencies and over airspy and sdrplay RSP
for "just" the frequencies to 2 GHz? I did not see any

andimik, 2020-06-16:
Me too.

A receiver which only can receive FM and DAB after a hack is really questionable.

I don't see any advantage to my Airspy and HackRF.

DG9BFC, 2020-06-16:
a man can not have enough sdr to try :-)
the bandwidth hack is well known and all pluto owners do that ... its not questionable ... it is stable and reliable ... and you only have to do that once ... the setting is permanent (no need to do the hack on every start)
advantage against hack rf?? hack rf is only 8 bits simplex while pluto is 12 bit fullduplex
the pluto is a good sdr for the higher bands ....goes up to 6 gig!
with the spectrum analyzer and tracking generator software you can tune antennas or filters (say you wanna build a bandpass filter for the dab band) ... only possible with a fullduplex sdr ...
ok ... for only receive dab ... a 10 bucks rtl dongle will do the job fine ... thats not the point here ... any sdr can be used for that ... the point is the pluto has lots of pros (also for other things and NOT only dab receive) ...

andimik, 2020-06-16:
Sorry, but I still can't see any advantage.

Could you please answer Jan's and my question? Can you set it to the needed sampling rate and bandwidth?

DG9BFC, 2020-06-16:
yes you can ... 500kc ... 1 meg 2 meg ...whatever you need (as far as i know up to 6 mhz bw but i never used that bw cause my pc is tooo slow for that)
max sampling is 60 megs (after the hack was made) but you can not squeeze  that much data over usb2 (so you then need to do decimation and use only say 1.5 or 2 megs bw)
gain setting can be set to manual (-10 to +73 db gain) or to auto (slow attack, hybrid, fast attack)

JvanKatwijk, 2020-06-16:
I believe you can, but so far I did not find a description of the "how".
As said, there is a fragment of code - dated 2017 - that - by means of
example - tells how to set up
the device, set an initial freq, bandwidth and samplerate, and shows how to
extract the data.
But I need some info on what to do when changing a frequency - e.g. do I
stop processing completely and restart
with a new frequency - and gain.

For the other devices, I have always found a - more or less complete -
functional description of an API, with functions
specified and some examples of how to use these.

wrt previous remark
I do not mean to disqualify the Pluto, no way, but for shortwave reception
it cannot be used (as one of my projects
I am rewriting my DRM decoder, which happens to be on short wave), for the
region to 2 GHz I am well equipped with
especially the SDRplay RSP.

Maybe if/when I am fed up with short wave and DAB I will look at things
like ATV, but that is 23cm,
and for experiments in the 2.4 GHz the Lime and Hackrf are  - at least for
the time being - more than sufficient

So, while Pluto might be a wonderful device, for the time being I do not
see any advantage of buying one

DG9BFC, 2020-06-16:
hack rf is only 8 bits
lime has much lower signal quality (spurious signal only 20 to 28 db below main signal)
and ... pluto can send datv signal ON ITS OWN (no pc needed!!) with a special firmware ...
ok no way for using the pluto on shortwave (besides adding mixers on both ports) ... but for all vhf uhf shf bands the pluto beats all other sdr with no problems ... 
as already said ... i am not a programmer so i can not help you with that ... but i could maybe ask another clever guy (evariste who made the datv firmware) to give us a few hints .... wait a minute i drop him a mail

DG9BFC, 2020-06-16:
as far as i know no need to completely stop the device ... you can change frequency and gain on the fly

DG9BFC, 2020-06-16:
maybe it would be best to have a settings box like in the rtl dongle (where you set gain, ppm correction etc) but for a first try we sure can just use auto gain ... and cause i have moddeed my pluto with a tuneable tcxo i can also live without any ppm correction (setting is done with a tiny screwdriver) ... and it is VERY VERY stable now ... 
i dropped a mail to evariste ... hopefully he will step in here and give us some hints

F5OEO, 2020-06-16:
Just a quick link to start with pluto...it uses libiio and is quiete easy to work. First example could be https://github.com/analogdevicesinc/libiio/blob/master/examples/ad9361-iiostream.c . Out of this scope with dab for now, but Pluto is running an ARM (part of the Zinq) which have sometimes enough processing to handle processes like dab (around same power as Raspebrry Pi2).

JvanKatwijk, 2020-06-20:
I added a preliminary - very experimental - version of a pluto interface.
It is highly unlikely that it works, but on start up it should print a lot,
telling to where in the code everything is OK

So, give it a try, and report back, then we can see whether in the end it
will be successful or not

DG9BFC, 2020-06-20:
hmm ... how??
can you send me a precompiled windows install to try??
if it helps ... i can offer a teamviewer meeting (with connected pluto and dab antenna)

JvanKatwijk, 2020-06-20:
I am sorry, but there is no support for libio in my cross compilation
environment, so no I cannot
create an windows executable with support for libiio

JvanKatwijk, 2020-06-20:
Well, it took most of my day, but the link
https://www.dropbox.com/s/yv9guuj2wou1wzq/windows-qt-dab.zip?dl=0
points to a zipped folder with a Qt-DAB for windows, with a first attempt
to support pluto
Now, what has to be done is
a. unzip the folder
b. start the program qt-dab-3.4.1.exe from a command window,
c. save everything that is printed
and send me the output
Then I can see whether or not it is worthwhile to continue with a remote
development

I do not know anything about windows and I want to keep it that way, so
debugging in a W environment is pretty tough
It would really be much easier  if you would have a Linuxbox for testing

anyway, give it a try

DG9BFC, 2020-06-20:
when starting shows an error libserialport-0.dll is missing please install new
hmmm
then i copied libserialport-0.dll from satsagen software in the qt fab folder and could get the soft starting
now it shows no pluto device found :-(

DG9BFC, 2020-06-20:
i found another libserialport-0.dll in sdr radio console v3 folder but with that copied in qt dab folder it does NOTstart at  all so i was guessing the one from the satsagen softare woud be ok ... (but maybe it is not that easy with just drag and drop the file in there?!?)

JvanKatwijk, 2020-06-20:
That is why windows is a disaster for developers, too many different dll's
https://www.dropbox.com/s/l5mpogfdvzlpol0/windows-qt-dab.zip?dl=0
In the zipped folder here, there are the official dll's

DG9BFC, 2020-06-20:
shows an error
[image]

JvanKatwijk, 2020-06-20:
That is bad, it means that the libstdc++ with which the Qt-DAB is compiled
is different from the one which is used for
the compilation of libiio.
Basically, this is end of story

JvanKatwijk, 2020-06-20:
https://www.dropbox.com/s/t3y4b84anfxc4nj/libstdc%2B%2B-6.dll?dl=0
try this one as replacement for the libc

DG9BFC, 2020-06-20:
ok at least it starts now ... but shows no pluto found :-(

JvanKatwijk, 2020-06-22:
well, if that is all the info it gives you cannot help me further. I think
we have to wait
for someone who is able to run the tests under Linux

DG9BFC, 2020-06-22:
maybe evariste can step again inside and give us a hint
i can additionally asc alberto who made the spectrum analyzer soft for the pluto
he sure also knows a bit about how to communicate with the device
at least you tried it :-)

DG9BFC, 2020-06-23:
i got an answer from alberto (copy from mail will follow) : 
...................................................................................(begin copy)....................................... 
hello Sigi
you must add an enviroment variablle to your windows:

Variable                 Value
IIOD_REMOTE      ip:192.168.2.1

where the value is the uri of your Pluto

or

JvanKatwijk must use iio_create_context_from_uri instead of iio_create_default_context function when creating context

ciao, Alberto
....................................................(end of copy)............................................

DG9BFC, 2020-06-23:
i hope that helps you so we finally can add the pluto :-)
i did not yet add that variable (and in fact i have no idea what alberto is talking about) ... but YOU sure know what he means ... come one we finally can solve that with the hint from alberto (i hope) ... greetz sigi

DG9BFC, 2020-06-23:
i am not a programmer but ...
in plutohandker cpp
change line 163
it is now
ctx = iio_create_default_context ();
and if i understood alberto correct it should be
ctx = iio_create_context_from_uri
.... no idea if that is correct ... just guessing
dg9bfc sigi

JvanKatwijk, 2020-06-25:
You should add what was said to your environment variable.
Changing create_default_context to ...from_url is only meaningful if I know
the url
(I'll be out rest of the day and most of tomorrow)

DG9BFC, 2020-06-25:
url of pluto by default is 192.168.2.1 ... 
how do i add the environment variable?!?
hmm ill ask alberto he sure may answer that

JvanKatwijk, 2020-06-27:
what is the network address of he adalm pluto?

JvanKatwijk, 2020-06-27:
Looking at the documentation and consulting a friend I found that the adalm
pluto itself is a device that is (should be)
connected to the net.
My suggestion is to adapt the driver in a way that you can specify the
network address, but then
we have to be certain that you know that address

DG9BFC, 2020-06-27:
you CAN connect to ethernet with an ethernet dongle ... then the adress is what the router gives the dongle as ip number
the default is using it via usb and then the default adress is 192.168.2.1
first get deqult adress working ... adding a user gui to change that to another adress is secod thing to do i guess :-)
greetz sigi dg9bfc

JvanKatwijk, 2020-06-28:
https://www.dropbox.com/s/ddequ0meduhxmhp/windows-qt-dab.zip?dl=0

As far as I understand, the pluto is a network device, and you configure it
with a network address
I made a version now where you can (have to) enter the network address in
the widget for the pluto control
Hope this helps

DG9BFC, 2020-06-28:
nothing in spectrum ... says no content found (at least it does not say no pluto found) ... hmmm does not work ... but we are getting closer i hope

JvanKatwijk, 2020-06-29:
I am not surprised, but please do as asked in the beginning: run the
software from a command window and report what is printed,
I added a lot of print statements to see what goed well and what does not.
Furthermore, the control widget shows a "state", so report that also back

DG9BFC, 2020-06-29:
[image]

accidently typed in the url in command line ... but also tried in your box
nothing to read in command line box ... and in pluto box only shown no content found try again ...spectrum empty

JvanKatwijk, 2020-06-29:
the url is not correct,  the software - with this url - cannot connect to
the device.
On the internet, there are several commands to investigate the settings in
the Pluto device

F5OEO, 2020-06-29:
@JvanKatwijk Can you commit yuour latest changes in order to inspect PLuto issue. If you can also make an AppImage to test under Linux, it could also be helpfull. On Windows, same here : not context found.

F5OEO, 2020-06-29:
Here is my initialisation (because iio_create_default_context() often return NULL)

`bool fmc_init()
{ 

    m_ctx = iio_create_default_context();
	if (m_ctx == NULL)
		m_ctx = iio_create_local_context();
	if (m_ctx == NULL)
		m_ctx = iio_create_network_context("pluto.local");
    if (m_ctx == NULL)
		m_ctx = iio_create_network_context("192.168.2.1");
	if (m_ctx == NULL)
	{
		fprintf(stderr, "No Pluto Found. Exiting\n");
		exit(1);
	}

	if (iio_context_get_devices_count(m_ctx) <= 0)
	{
		FMC_ERROR("No Pluto Found. Exiting\n")
		exit(1);
	}

    m_dev= get_ad9361_phy(m_ctx);

}`

JvanKatwijk, 2020-06-29:
The first few lines were what I had, then the comment was that an
environment variable had to be set to
the network address

Is it true that - whatever is happening - the network address is
192.168.2.1?
Anyway,
I'll adapt my code to match your code (Thanks!) and upload the stuff

DG9BFC, 2020-06-29:
no idea if that helps ... alberto (satsagen soft) ... found that there is a "white space" in the url

JvanKatwijk, 2020-06-29:
I changed the context finding code according to the example
I uploaded the committed sources (it might take some time before an
appImage is generated)
https://www.dropbox.com/s/ddequ0meduhxmhp/windows-qt-dab.zip?dl=0
this is a link to the windows folder with all the stuff, containing the
latest compiled version
Give it a try

DG9BFC, 2020-06-29:
getting closer ... now it says rx0 port not found

JvanKatwijk, 2020-06-29:
but what is the text printed on the command shell

DG9BFC, 2020-06-29:
[image]

nothing in command shell ...

andimik, 2020-06-29:
It looks like you started it first on the console, closed it and then started it in Windows.

I wonder because there is an empty command line after you have started it.

JvanKatwijk, 2020-06-29:
OK, I see (yes, it is windows of course),
here is a link to a version that prints a lot of data on the command shell
window
https://www.dropbox.com/s/n19jzhgmohqcxm9/windows-qt-dab.zip?dl=0

DG9BFC, 2020-06-29:
[image]

i left away the upper part (soundcard settings etc) ... cause i thought the lower end is where we have the problem

JvanKatwijk, 2020-06-29:
Well it indicates that it cannot set the sampling frequency

For the rest of the day I have to do other things, tomorrow I'll have a
look at the sampling frequency

F5OEO, 2020-06-29:
Jan,

SR without decimation is minimum 2.083KS :
See the documentation
https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361
If you need exactly this SR, you could use the decimation by 2 or 4. I
could help you on that if needed (you need to set a lowpass FIR)

NOTE:
In the minimum ADC rate is 25MSPS. Baseband rates below 2.083 MSPS (25MSPS/12)
require FIR decimation/interpolation to be set. In other words the FIR
filter needs to be configured and enabled. The minimum baseband rate with
the FIR filter (decimate by 4) enabled is: 25MSPS /(4*12) = 520.83 kSPS.

andimik, 2020-06-29:
See my posting dated 16th, asking for the sampling rate...

JvanKatwijk, 2020-06-29:
For DAB we need exactly 2048000 IQ samples/second.
If 2500000 is a decent option for the hardware, it is not big deal to do a
software conversion,
that is what is happening with the airspy as well, but it would be helpful
to just know what decent values are.
I must say that the documentation for the pluto lacks is below average
(Most others just give an APi with a concise description of the functions
and allowed and expected ranges for the value)

[30 more replies]

closed 2020-09-27 by JvanKatwijk