JvanKatwijk/qt-dab #175: Scan results wrong text output

andimik, 2020-06-29:
When an ensemble contains a data service, the output of the text files is different:

Qt-DAB's button "content" lists it correctly ("Spored" is an SPI/EPG)

```
Data Services
program name;;serviceId;subchannelId;start address;length (CU); bit rate; FEC; prot level; appType ; subService ; 

Spored          ;Slovenia;E4910002;22;852;12;16;0;EEP 3-A;7;no;;
```

but the text file generated by the scan is wrong (data is listed as audio service using the SID and bitrate of one of the audio services):

```
Spored          ;9203;DAB+ (plus);128;EEP 3-A;1/2;
```
[Qt-DAB-scan-2020-jun-29-21-29.txt](https://github.com/JvanKatwijk/qt-dab/files/4847895/Qt-DAB-scan-2020-jun-29-21-29.txt)
[Qt-DAB-10D-2020-jun-29-22-31.txt](https://github.com/JvanKatwijk/qt-dab/files/4847896/Qt-DAB-10D-2020-jun-29-22-31.txt)

Remark: The "content" button is much more detailed, so why not use the same data (columns) for the scan output?

JvanKatwijk, 2020-06-30:
Hi Andreas
The list of services was added to make the scanning more attractive: my
original idea for the
scanning was (and I use scanning still that way) to skip to the next
channel with data (a setting
in the ".ini" file lets you choose between the two different scan modes).
Anyway, some people asked for a scanning mode, a scan over all channels,
and then showing the
names of the services, something I did.

Since I am essentially a lazy person I only implemented code to show the
audio services, but only
showing the name of the services did not feel good, so I added a few key
characteristics.

The updated version now shows the names of the audio services.

Adding further to the description of the service is of course possible, but
getting a complete
scan over all channels with an exhaustive description of each service was
delegated to the dab-scanner

JvanKatwijk, 2020-06-30:
I added printing the name - and kind - of data services to the scan
function.
First audio services, then - if any - data services

andimik, 2021-07-28:
Sorry, did not close this issue, which has been fixed long ago (BTW: at that time I only could receive one mux from abroad as the Austrian mux even started later in August 2020 here) ...

In the scan-text-file the Bitrate, SID and other data are correct.

```
Data Services
program name;;serviceId;subchannelId;start address;length (CU); bit rate; FEC; prot level; appType ; subService ; 

Spored          ;Slovenia;E4910002;22;852;12;16;0;EEP 3-A;7;no;;
```

closed 2021-07-28 by andimik