JvanKatwijk/qt-dab #178: Build issue

hsantens, 2020-08-31:
perhaps I'm doing something wrong, but dab-mini does not build on my RPI4
- first some warning occur on various file
- the there are errors on radio.cpp

See output below. Do you have any idea? Cause I have no clue where to look. Thanks

> /home/pi/qt-dab/src/ofdm/fib-decoder.cpp: In member function ‘std::vector<serviceId> fibDecoder::insert(std::vector<serviceId>, serviceId, int)’:
> /home/pi/qt-dab/src/ofdm/fib-decoder.cpp:1517:23: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
>                ((baseN < n. SId) && (n. SId <= serv. SId)):
>                  ~~~~~~^~~~~~~~
> 
> /home/pi/qt-dab/src/backend/msc-handler.cpp: In member function ‘void mscHandler::stopService(descriptorType*)’:
> /home/pi/qt-dab/src/backend/msc-handler.cpp:225:20: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<Backend*>::size_type’ {aka ‘unsigned int’} [-Wsign-compare]
>   for (int i = 0; i < theBackends. size (); i ++) {
>                   ~~^~~~~~~~~~~~~~~~~~~~~~
> 
> /home/pi/qt-dab/src/backend/backend-deconvolver.cpp: In constructor ‘backendDeconvolver::backendDeconvolver(descriptorType*)’:
> /home/pi/qt-dab/src/backend/backend-deconvolver.cpp:31:69: warning: ‘new’ of type ‘uep_protection’ with extended alignment 16 [-Waligned-new=]
>                                                        d -> protLevel);
>                                                                      ^
> /home/pi/qt-dab/src/backend/backend-deconvolver.cpp:31:69: note: uses ‘void* operator new(std::size_t)’, which does not have an alignment parameter
> /home/pi/qt-dab/src/backend/backend-deconvolver.cpp:31:69: note: use ‘-faligned-new’ to enable C++17 over-aligned new support
> /home/pi/qt-dab/src/backend/backend-deconvolver.cpp:34:69: warning: ‘new’ of type ‘eep_protection’ with extended alignment 16 [-Waligned-new=]
>                                                        d -> protLevel);
>                                                                      ^
> /home/pi/qt-dab/src/backend/backend-deconvolver.cpp:34:69: note: uses ‘void* operator new(std::size_t)’, which does not have an alignment parameter
> /home/pi/qt-dab/src/backend/backend-deconvolver.cpp:34:69: note: use ‘-faligned-new’ to enable C++17 over-aligned new support
> 
> /home/pi/qt-dab/src/backend/data/pad-handler.cpp: In member function ‘void padHandler::dynamicLabel(uint8_t*, int16_t, uint8_t)’:
> /home/pi/qt-dab/src/backend/data/pad-handler.cpp:271:16: warning: variable ‘segmentno’ set but not used [-Wunused-but-set-variable]
>  static int16_t segmentno    = 0;
>                 ^~~~~~~~~
> /home/pi/qt-dab/dab-mini/radio.cpp: In constructor ‘RadioInterface::RadioInterface(QSettings*, const QString&, QWidget*)’:
> /home/pi/qt-dab/dab-mini/radio.cpp:220:65: warning: ‘new’ of type ‘dabProcessor’ with extended alignment 16 [-Waligned-new=]
>   my_dabProcessor = new dabProcessor (this, inputDevice, &globals);
>                                                                  ^
> /home/pi/qt-dab/dab-mini/radio.cpp:220:65: note: uses ‘void* operator new(std::size_t)’, which does not have an alignment parameter
> /home/pi/qt-dab/dab-mini/radio.cpp:220:65: note: use ‘-faligned-new’ to enable C++17 over-aligned new support
> /home/pi/qt-dab/dab-mini/radio.cpp: In member function ‘void RadioInterface::stopService()’:
> /home/pi/qt-dab/dab-mini/radio.cpp:837:64: error: no matching function for call to ‘dabProcessor::stopService(QString&)’
>      my_dabProcessor -> stopService (currentService. serviceName);
>                                                                 ^
> In file included from /home/pi/qt-dab/dab-mini/radio.h:37,
>                  from /home/pi/qt-dab/dab-mini/radio.cpp:38:
> /home/pi/qt-dab/dab-mini/../dab-processor.h:84:8: note: candidate: ‘void dabProcessor::stopService(descriptorType*)’
>   void  stopService  (descriptorType *);
>         ^~~~~~~~~~~
> /home/pi/qt-dab/dab-mini/../dab-processor.h:84:8: note:   no known conversion for argument 1 from ‘QString’ to ‘descriptorType*’
> /home/pi/qt-dab/dab-mini/radio.cpp: In member function ‘void RadioInterface::stop_secondService()’:
> /home/pi/qt-dab/dab-mini/radio.cpp:1258:60: error: no matching function for call to ‘dabProcessor::stopService(QString&)’
>   my_dabProcessor -> stopService (secondService. serviceName);
>                                                             ^
> In file included from /home/pi/qt-dab/dab-mini/radio.h:37,
>                  from /home/pi/qt-dab/dab-mini/radio.cpp:38:
> /home/pi/qt-dab/dab-mini/../dab-processor.h:84:8: note: candidate: ‘void dabProcessor::stopService(descriptorType*)’
>   void  stopService  (descriptorType *);
>         ^~~~~~~~~~~
> /home/pi/qt-dab/dab-mini/../dab-processor.h:84:8: note:   no known conversion for argument 1 from ‘QString’ to ‘descriptorType*’
> make[2]: *** [CMakeFiles/dabMini-1.0.dir/build.make:984: CMakeFiles/dabMini-1.0.dir/radio.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/dabMini-1.0.dir/all] Error 2
> make: *** [Makefile:130: all] Error 2
> 
>

JvanKatwijk, 2020-08-31:
> fibDecoder::insert(std::vector, serviceId, int)’:

you are right, I uploaded changes to dab-maxi without checking the impact
on dab-mini
Uploaded a version that compiled (and runs) nicely on my laptop
(I know there are warnings, slightly depending on the computers used, but
as far as I can see still harmless)

hsantens, 2020-08-31:
thanks, it's working now

closed 2020-08-31 by hsantens