JvanKatwijk/qt-dab #180: Multiplex received from more than one transmitter (Windows?)

rhollander73, 2020-09-16:
If qt-dab receives a multiplex from more than one transmitter, the reception falters, even when the signal seems strong enough. For example here at my home in the Netherlands the NPO 12C multiplex I can receive easily indoors with a normal DAB radio, but with qt-dab it somehow doesn't work (other multiplexes like Oost 6B, Zuid 7A, 9C and 11C have no problems in qt-dab).

The SNR of the NPO 12C multiplex is at 16 which should be enough. Sync FIC jumps up and down between 90% and 100%. Est switches between "1 5", "39 5" and "40 5". I do not know if this problem also occurs under Linux or Mac.

andimik, 2020-09-17:
You could try to change the gain settings.

Or you upload a dump to a file hoster. 30sec are enough to investigate.

JvanKatwijk, 2020-09-17:
What andimik says: upload a dump

closed 2020-09-27 by JvanKatwijk