JvanKatwijk/qt-dab #182: Loading a 2nd raw or sdr file does not close the 1st

andimik, 2020-09-21:
I'm very often using prerecorded files.

For several versions now, there is a strange behavior: When you open a raw or sdr file, and then want to open another one, the first one is not closed and both are played in parallel (with audio!).

[image]

The only thing is to close Qt-DAB and load the 2nd one again.

JvanKatwijk, 2020-09-21:
Yes
last week I noticed. But it does not seem to occur when the second file is
different type (i.e. the first one "raw", second one "sdr")
I'll have a look tomorrow

JvanKatwijk, 2020-09-22:
The problem was caused by an over-enthousiastic optimization,

If you want to address it in your sourcetree, then adapt the function
"doStart" in the file radio.cpp as follows
//      when doStart is called, a device is available and selected
bool    RadioInterface::doStart () {
        QString h       = dabSettings -> value ("channel", "12C").
toString();
        int k           = channelSelector -> findText (h);
        if (k != -1)
           channelSelector -> setCurrentIndex (k);     my_dabProcessor =
new dabProcessor  (this, inputDevice, &globals);
       my_dabProcessor = new dabProcessor  (this, inputDevice, &globals);

//      Some buttons should not be touched before we have a device
        connectGUI ();

//      we avoided up till now connecting the channel selector
//      to the slot since that function does a lot more that we
//      do not want here
        connect (presetSelector, SIGNAL (activated (const QString &)),
                 this, SLOT (handle_presetSelector (const QString &)));
        connect (channelSelector, SIGNAL (activated (const QString &)),
                 this, SLOT (selectChannel (const QString &)));
//
//      Just to be sure we disconnect here.
//      It would have been helpful to have a function
//      testing whether or not a connection exists, we need a kind
//      of "reset"
        disconnect (deviceSelector, SIGNAL (activated (const QString &)),
<<<<<< these 2 lines
                    this, SLOT (doStart (const QString
&)));                                 ,<<<<<<<<<<<
        disconnect (deviceSelector, SIGNAL (activated (const QString &)),
                    this,  SLOT (newDevice (const QString &)));

Of course you can use the 3.6 sources, but the documentation is not up to
date yet

andimik, 2020-11-28:
Confirmed, this works now.

closed 2020-11-28 by andimik