JvanKatwijk/qt-dab #183: DAB stick not found on 3.5.3 and 3.6

casual-listener, 2020-09-25:
Hello,
thanks for this nice program...it's fun to work with. Or at least, it was up to AppImage version 3.4.3.
I'm using it with a RTL2832U-based stick.
On newer releases (i tried 3.5.3 and 3.6) the DAB Stick won't be found.
When i start it from a console, i get "failed to open rtlsdr.dll (librtlsdr.so: cannot open shared object file: No such file or directory)".
Since 3.4.3 ist still running fine, is there a fault in the recent versions?
I tried this both on openSuse 15.1 and Devuan 3.0.

Thank you very much.

JvanKatwijk, 2020-09-25:
Thanks for the message. A similar message from an Ubuntu 20.04 user was
there last week.
The message as such is clear: the librtlsdr.so cannot be found and is not
in the loadpath
I checked and double checked, but the code to access librtlsdr.so is
unchanged since ages.

Did you change/update the system?
Did you install your own version of the linrtlsdr.so library or do you
depend on the one in the standard reposito

Last week I brought up a version of 20.04 in a VM and encountered at first
the same problem.
Then I installed the most recent osmocom version of te rtlsdr library, made
it visible by ensuring that the lib
was found by the loaded by adding a file /etc/ld.config.d/local.conf,
ensuring that an udev file
was written. Then after updating the loader cache by sudo ldconfig the
library was found.
Then I uninstalled the osmocom rtlsdr library, removed all files from the
system and still the library
was found. So, as far as I can see, the problem is that the directory the
library is in is not standard in the load path

I assume you know where the rtlsdr library is residing? and can you ensure
that that directory is in the load path

casual-listener, 2020-09-25:
Hello Jan,
thank you for the fast reply.

>  Did you change/update the system?

I did the usual security updates, but the librtlsdr.so hasn't changed for months.

> Did you install your own version of the linrtlsdr.so library or do you depend on the one in the standard reposito

I use the one provided by the distribution / repository.

> So, as far as I can see, the problem is that the directory the library is in is not standard in the load path I assume you know > where the rtlsdr library is residing? and can you ensure that that directory is in the load path

Yes, it's in /usr/lib64 and /usr/lib64 is still in the load path. 
There hasn't changed anything since months.
What bothers me is, why is the 3.4.1 AppImage still running fine w/o any problems and why won't the newer AppImages?
I thought, an AppImage contains all necessary libraries and since 3.4.1 is still running there might be a missing librtlsdr.so in the newer versions. I believed that no additional library from the "host system" is needed. That might not be true? 

Thank you for your efforts.

JvanKatwijk, 2020-09-25:
I ran some tests and indeed, on a 20.04 VM the 3.4 version accepts the
dabstick and the 3.6 version cannot find the driver library!
On my fedora box there is no problem whatsoever, but what I shall do (kind
of magic should happen I think)
is replace the rtlsdr driver part in Qt-dab by that of 3.4 and see what the
difference in behaviour is
The code looks as far as I can see the same (in the start up part, there
were changes in the callback handling)

I'll let you know later on this afternoon, this phenomenon bothers me very
much, it should not happen

casual-listener, 2020-09-25:
I just took a closer look on the mounted AppImages.
In 3.4.1, there's a librtlsdr in /tmp/.mount_Qt_DABxxxxx/usr/lib directory.
In 3.5 and 3.6 there's no librtlsdr in the /tmp/.mount_Qt_DABxxxxx/usr/lib directory.

I'll try to mount the AppImage r/w and copy the missing link into it. I'm curious, if this will work...

JvanKatwijk, 2020-09-25:
OK, I did some testing
a. on a 16.04 Ubuntu VM the appImage 3.6 works just fine, as with my fedora
box
I then modified the .travis file to include a librtlsdr.so in the /usr/lib
of the appImage
b. on a fresh install of Ubuntu 20.04 the appImage 3.6 does not find the
rtlsdr library
c. after installing the regular librtlsdr the appImage 3.6 finds the stick,
however, the installed
librtlsdr requires the kernel module to be blacklisted
d. after removing the installed librtlsdr the appImage still finds the
stick, again however
with the requirement to blacklist the kernel module

So, the problem is most likely in the loadpath. Note that in order to allow
running on systems with
no drivers installed, the qt-dab device handler loads the required
functions from the device library dynamically

In the very beginning the appImage would create udev files for the various
devices, however, this implies
that root permission is asked each and every time the appImage is started.
That is why later on
the user was made responsible for installing the device libraries.

JvanKatwijk, 2020-09-25:
you could try the current appImage for 3.6. I assume you blacklisted the
kernel module.
It is as in 3.4. However, as said, the udev file is not installed by the
software so it is assumed that you have/had
installed librtlsdr

casual-listener, 2020-09-26:
Hallo Jan,
this one now works as fine as the AppImage of 3.4.1 does. And even w/ or w/o a installed librtlsdr.
That's great work you've done. 
I do thank you very much.

closed 2020-09-26 by casual-listener