JvanKatwijk/qt-dab #184: How to build Qt-DAB on a Mac

JvanKatwijk, 2020-09-27:
I sometimes get requests about running the software on a Mac. While I admire the apple technology, I do not know anything about it. Any suggestion of pointer to a suggestion to build Qt-DAB on a Mac is welcome

richardh538, 2020-10-07:
I have managed to build and link dab-mini with my hackrf one, but unfortunately it fails to start because it can not find  libhackrf.so (hackrfHandler), The so-file doesn't exists in homebrew (the "apt" for mac), I tried to google the recipe how build the shared object but can't really find it. So I am stuck. Any idea?

JvanKatwijk, 2020-10-08:
OK,
The sources for hackrf support can be found on the net and the "host"
(sub)directory contains a CMakeLists.txt file.
I can send the source package if it helps

richardh538, 2022-07-15:
richardh@legazy ~/qt-dab % git push --set-upstream origin mac                                                                 mac
ERROR: Permission to JvanKatwijk/qt-dab.git denied to richardh538.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

closed 2022-08-12 by JvanKatwijk

MaplesThroughWindows, 2024-02-09:
Could you supply the link to the guide to installing/compiling Qt-DAB for macOS? There's nothing obvious that I can find either in the repo or with a web search.

JvanKatwijk, 2024-02-09:
I do not have a mac,
There is in the docs section a document "builder.pdf" that in quite some
detail tells what libraries are required and what settings there are
In the ".pro" file there are some sections marked with "mac" As fas as I ve
been told, with those settings (and the installed libraries)
a mac version can be build

richardh538, 2024-02-09:
Hello @MaplesThroughWindows 

I tried main-branch and qt-dab-s6 with no changes at all: you need homebrew, components needed are referenced in qt-dab-6.pro, PKGCONFIG zlib, faad, ... qwt needs to be registred in Qt (see my comments in the .pro file).
My homebrew installation is broken so I had to create softlinks for:
libhackrf.dylib -> /usr/local/Cellar/hackrf/2023.01.1/lib/libhackrf.dylib
libz.1.dylib -> /usr/local/Cellar/zlib/1.3.1/lib/libz.1.3.1.dylib
I am unsure if my binary will work out of the box on other Mac-systems. 
I tried once to move the binary to my old mac, and that, did not work.
Let me know if you want to try my binary anyway. :)

MaplesThroughWindows, 2024-02-14:
@richardh538: would you have the time to add your build process for the Mac to the official PDF? It's a novice area for me and a guide would be a useful springboard.

> There is by the way some kind of bug when the program select soundcard on a Mac. I guess some kind of enumeration error.

What impact does this have in practice?