JvanKatwijk/qt-dab #188: Audio skips/jitter in playback

Weppel, 2020-11-16:
Very nice work on qt-dab @JvanKatwijk! 

I've installed qt-dab (dab-maxi) on a Raspberry Pi 4 with the latest Raspberry Pi OS. I also have a rtl-sdr v3 stick. I'm receiving fine, but I'm experiencing audio skips/jitter (like a bad CD skipping over) every minute or so. 

My signal looks good. Sometimes I'm even picking up multiple transmitters:

[image]

I've tried swapping the Pi (v4 > v3), new OS install, new stick, new antenna, all still result in the same issue.

JvanKatwijk, 2020-11-17:
a. if you look at the processor load (e.g. by looking at htop), is the
processor load divided (more or less( equally over the cores or is there a
single core with most of the load
b. from the picture it is not clear what audio device you selected.
It might be a buffer issue then, in the ".ini" file there is a setting for
the size of the audio buffers

Weppel, 2020-11-17:
Processor load isn't equally divided over cores, but I also do not see a single core being overloaded/maxed:

[image]

As for buffers I've already set "latency=30", but this does not seem to solve the problem. Will raising this value even higher possibly fix this?

For output, I'm not using local playback but rather the record function.

JvanKatwijk, 2020-11-18:
> :
> OK, I'll have a look at the record function later today.

I just made a 10 minute recording on an RPI 3, and when playing the
recording with aplay on the RPI there was no jitter/skipping at all.
(I recorded with an SDRplay, not a stick).
Can you send me recording (well, a link to it)? then maybe I'm able to
detect a pattern or so
What device are you using to write the file on? an usb stick, or an ssd
card.

Given that it is not a performance issue, the possibilities seem to be
a. the input is distorted, and ithe decoding of sound fragments fails
b. writing the output gives problems

andimik, 2020-11-18:
Well, I am using RPI and Qt-DAB for recording, too.

But my recordings are error free.

Maybe you should specify which record function you are talking about:

RAW
UFF (XML)
AAC

zeitfalle, 2020-11-18:
I also use the recording function to record for 6-8 hours per day on a Raspberry Pi 4 - I have never had any problem whatsoever. Could it be possible that you might be using a bad SD-Card (if that is what you are actually using)?

Weppel, 2020-11-18:
Interesting! What settings did you guys use in .qt-dab.ini? 

I'm recording using the regular record function, this outputs a WAV file. I'm recording it to a mounted NFS volume to prevent overloading the SD card. I'll plug in a local USB stick to see if that helps. 

I'll try and get a sample file soon.

andimik, 2020-11-18:
No special settings.

From time to time I even delete this file. It will be created at the next start, the default settings are fine.

If you use raw, then you can save 50%. Wave (*.sdr) are 16bit, an RTLSDR dongle is 8bit only.

JvanKatwijk, 2020-11-19:
what AAC decoder did you include in the configuration ? fdkaac or faad

I seem to recall that when using fdkaac there were some glitches in the
audio output

closed 2022-08-28 by JvanKatwijk