JvanKatwijk/qt-dab #190: Widgets close at start up

andimik, 2020-11-28:
In the `.qt-dab.ini` the entry `device=dabstick` will chose the dabstick at the start, but since a couple of commits its widget closes after a second.

If you want to change the settings, you have to select dabstick again

[image]

Same with `device=file input (.raw)` or `device=file input (.sdr)`

The following widget either does not appear any more.

[image]

I guess it's the same with the other widgets.

JvanKatwijk, 2020-11-28:
there was some mixe up with a flag: in half of the software the flag up
meant "device widget is shown" and in the other half "do not show device
widget"
will be addressed

andimik, 2021-01-02:
In the meanwhile you fixed it. So I can close the issue.

closed 2021-01-02 by andimik