JvanKatwijk/qt-dab #191: MOT-Slides from EPG are not saved any more

andimik, 2020-11-28:
I've set all paths in `.qt-dab.ini`

```
epgPath=/tmp
filePath=/tmp
motSlides=1
pictures=/tmp
```
but the slides in the EPG are not saved in `/tmp`.

```
starting a backend for EPG ORS          (E0A0AD50)
handle_MOT: type 203 (2), name img/AC51_ENERGY_112x32.png dir = 1
handle_MOT: type 203 (2), name img/AD27_Rock_Antenne_112x32.png dir = 1
handle_MOT: type 203 (2), name img/AD27_Rock_Antenne_32x32.png dir = 1
handle_MOT: type 203 (2), name img/AD2A_Technikum_ONE_32x32.png dir = 1
handle_MOT: type 203 (2), name img/AD53_KLASSIK_RADIO_32x32.png dir = 1
handle_MOT: type 203 (2), name img/AD54_ARABELLA_RELAX_112x32.png dir = 1
```
They did in the past, but I cannot tell you the version. If you want, I could find out.

Hint: Maybe it makes sense to write the path instead of `dir = 1` ...

JvanKatwijk, 2020-11-28:
absolutely true

It turned out that the EPG mini slides were overriding the slides as passed
of the PAD in the audio signal.
You want the epg slides saved?

andimik, 2020-11-28:
Well it was possible before ...

BTW You could use them in the epg widget as icon ...

JvanKatwijk, 2020-11-28:
I have been thinking about that, but it is difficult (at least now) to
identify the slides/pictures that belong to a specific timetable for a
given service

I am making some changes nnow to save the slides - not too difficult, later
tonight or tomorrow I'll upload a version
with
a. the flag problems for showing the device widget
b. saving the pictures/slides for EPG

Right now I was adding one of the last items in my todo list: a skiptable
for scanning, so you can just specify - using a widget of course -
those channels that can be skipped when scanning.
In the current set up, this then also applies to selecting a "next" or
"previous" channel (of course you can select any channel you want,
but "next" is then skipping any channel in the skipList